nabaztag2018/pynab #1: Pb upgrade

BorisFR, 2019-05-03:
Bonjour. Souci avec l'upgrade actuelle.
via l'interface web : j'obtiens un "502 Bad Gateway" 
si je le fais via la ligne de commande :
```(...) d'autres lignes avant, sans souci
+ sudo systemctl stop nabd
+ cd /home/pi/pynab
+ [[ 1000 -ne 1000 ]]
+ git pull
Déjà à jour.
+ bash upgrade.sh newversion
+ trap 's=$?; echo "$0: Error on line "$LINENO": $BASH_COMMAND"; exit $s' ERR
+ IFS='
	'
++ sed -nE -e 's|WorkingDirectory=(.+)|\1|p'
+ root_dir=/home/pi/pynab
++ stat -c %U /home/pi/pynab
+ owner=pi
++ stat -c %u /home/pi/pynab
+ ownerid=1000
+ version=old
+ '[' 1 -eq 1 ']'
+ '[' newversion == newversion ']'
+ version=new
+ case $version in
+ cd /home/pi/pynab
+ venv/bin/pip install -r requirements.txt
Ignoring Cython: markers 'sys_platform == "linux" and "arm6l" in platform_machine' don't match your environment
Ignoring numpy: markers 'sys_platform == "linux" and "arm6l" in platform_machine' don't match your environment
Ignoring py-kaldi-asr: markers 'sys_platform == "linux" and "arm6l" in platform_machine' don't match your environment
Ignoring RPi.GPIO: markers 'sys_platform == "linux" and "arm6l" in platform_machine' don't match your environment
Ignoring rpi-ws281x: markers 'sys_platform == "linux" and "arm6l" in platform_machine' don't match your environment
Collecting git+https://github.com/pguyot/py-kaldi-asr (from -r requirements.txt (line 15))
  Cloning https://github.com/pguyot/py-kaldi-asr to /tmp/pip-434qmth9-build
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-434qmth9-build/setup.py", line 3, in <module>
        import numpy
    ImportError: No module named 'numpy'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-434qmth9-build/
++ s=1
++ echo 'upgrade.sh: Error on line 40: venv/bin/pip install -r requirements.txt'
upgrade.sh: Error on line 40: venv/bin/pip install -r requirements.txt
++ exit 1
++ s=1
++ echo 'upgrade.sh: Error on line 36: bash upgrade.sh "newversion"'
upgrade.sh: Error on line 36: bash upgrade.sh "newversion"
++ exit 1
```
Help me!

BorisFR, 2019-05-03:
Je tente
```
sudo apt update
sudo apt upgrade
    et la nouvelle phase 3 : 
wget https://github.com/pguyot/kaldi/releases/download/v5.4.1/kaldi-c3260f2-linux_armv6l-vfp.tgz
cd / && sudo tar xvf /home/pi/kaldi-c3260f2-linux_armv6l-vfp.tgz
```
Je dis quoi d'ici peu si c'est ok ou pas ;)

BorisFR, 2019-05-03:
L'url de la release a changée : [https://github.com/pguyot/kaldi/releases/download/v5.4.1/kaldi-c3260f2-linux_armv6l-vfp.tar.xz](https://github.com/pguyot/kaldi/releases/download/v5.4.1/kaldi-c3260f2-linux_armv6l-vfp.tar.xz)

pguyot, 2019-05-03:
Pour info, ce n'est pas nécessaire d'installer kaldi sur un tagtagtag avec la carte V1 car il n'y a pas de micro sur cette carte et du coup pas de reconnaissance vocale. Kaldi est désormais installé directement avec le script install.sh, avec la bonne URL, et uniquement pour les cartes V2.

Le bug était dans le fichier requirements.txt, normalement c'est corrigé avec le commit a86fc7b. En théorie, on peut mettre à jour directement depuis l'interface web, mais je pense qu'il faudrait davantage de robustesse. Si nécessaire, la commande ```bash upgrade.sh``` devrait fonctionner : elle arrête les services, fait un git pull puis appelle la nouvelle version de upgrade.sh.

N'hésitez pas à rouvrir le ticket si le problème persiste.

closed 2019-05-03 by pguyot

BorisFR, 2019-05-03:
J'ai fini par faire un mv pynab en pyold et faire un git pull puis install.sh qui est en cours :)