nabaztag2018/pynab #2: Pb upgrade

BorisFR, 2019-05-04:
Bonjour.
Je suis reparti d'une situation bien propre avec **bash install.sh --v1**
C'est mieux mais il subsiste encore des erreurs :
```
Collecting pycparser (from cffi!=1.11.3,>=1.8->cryptography>=1.6.0->Mastodon.py->-r requirements.txt (line 7))
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.19-py2.py3-none-any.whl
Collecting pyparsing>=2.0.2 (from packaging->deprecation<3,>=2->snips_nlu->-r requirements.txt (line 16))
  Using cached https://files.pythonhosted.org/packages/dd/d9/3ec19e966301a6e25769976999bd7bbe552016f0d32b577dc9d63d2e0c49/pyparsing-2.4.0-py2.py3-none-any.whl
Building wheels for collected packages: psycopg2-binary, cryptography, cffi, PyYAML
  Running setup.py bdist_wheel for psycopg2-binary ... error
  Complete output from command /home/pi/pynab/venv/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-502l53fg/psycopg2-binary/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpk3l9brzapip-wheel- --python-tag cp35:
  /usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'project_urls'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for psycopg2-binary
  Running setup.py clean for psycopg2-binary
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /home/pi/pynab/venv/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-502l53fg/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp93jbeglcpip-wheel- --python-tag cp35:
  c/_cffi_backend.c:2:20: fatal error: Python.h: Aucun fichier ou dossier de ce type
   #include <Python.h>
                      ^
  compilation terminated.
  Traceback (most recent call last):
    File "/usr/lib/python3.5/distutils/unixccompiler.py", line 118, in _compile
      extra_postargs)
    File "/usr/lib/python3.5/distutils/ccompiler.py", line 909, in spawn
      spawn(cmd, dry_run=self.dry_run)
    File "/usr/lib/python3.5/distutils/spawn.py", line 36, in spawn
      _spawn_posix(cmd, search_path, dry_run=dry_run)
    File "/usr/lib/python3.5/distutils/spawn.py", line 159, in _spawn_posix
      % (cmd, exit_status))
  distutils.errors.DistutilsExecError: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "/usr/lib/python3.5/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.5/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.5/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/home/pi/pynab/venv/lib/python3.5/site-packages/setuptools/command/bdist_egg.py", line 161, in run
      cmd = self.call_command('install_lib', warn_dir=0)
```
puis
Failed building wheel for cryptography
Failed building wheel for cffi
Failed building wheel for PyYAML
et enfin
```
Command "/home/pi/pynab/venv/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-502l53fg/psycopg2-binary/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-7b9ivgx6-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/pynab/venv/include/site/python3.5/psycopg2-binary" failed with error code 1 in /tmp/pip-build-502l53fg/psycopg2-binary/
++ s=1
++ echo 'install.sh: Error on line 87: venv/bin/pip install -r requirements.txt'
install.sh: Error on line 87: venv/bin/pip install -r requirements.txt
++ exit 1
```

pguyot, 2019-05-04:
Pourriez-vous tester :
```venv/bin/pip install wheel```

Puis
```bash install.sh --v1```

Si ça marche, je mettrais le script install à jour. Sinon, il manque d'autres paquets (python3-dev ?).

BorisFR, 2019-05-04:
```
pi@nabaztag:~/pynab $ venv/bin/pip install wheel
Requirement already satisfied: wheel in ./venv/lib/python3.5/site-packages
```
Effectivement, python3-dev n'est pas présent, du coup :
```
pi@nabaztag:~/pynab $ sudo apt-get install python3-dev
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
The following additional packages will be installed:
  libpython3-dev libpython3.5-dev python3.5-dev
Les NOUVEAUX paquets suivants seront installés :
  libpython3-dev libpython3.5-dev python3-dev python3.5-dev
0 mis à jour, 4 nouvellement installés, 0 à enlever et 0 non mis à jour.
Il est nécessaire de prendre 37,3 Mo dans les archives.
Après cette opération, 48,8 Mo d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] 
```
installation ok, j'ai relancé le script d'installation...
```
Building wheels for collected packages: psycopg2-binary, cryptography, cffi, PyYAML
  Running setup.py bdist_wheel for psycopg2-binary ... done
  Stored in directory: /home/pi/.cache/pip/wheels/b2/75/e5/c95c469028131b90868db0697c06531b503ce232904893b557
  Running setup.py bdist_wheel for cryptography ... done
  Stored in directory: /home/pi/.cache/pip/wheels/43/61/c8/0a4464601ce180d26e0a8dfdfa88c824e419dcc65bd43bda6e
  Running setup.py bdist_wheel for cffi ... \
```
Ca n'est pas terminé, mais ça a l'air de le faire ! Je cloture dès que c'est terminé et opérationnel.

BorisFR, 2019-05-04:
Bon bah nouvel échec :(
```
  Running setup.py bdist_wheel for PyYAML ... done
  Stored in directory: /home/pi/.cache/pip/wheels/ad/56/bc/1522f864feb2a358ea6f1a92b4798d69ac783a28e80567a18b
Successfully built psycopg2-binary cryptography cffi PyYAML
Installing collected packages: psycopg2-binary, pytest-django, lockfile, urllib3, certifi, chardet, idna, requests, asn1crypto, pycparser, cffi, cryptography, http-ece, python-dateutil, decorator, python-magic, Mastodon.py, Mpg123, gunicorn, Cython, numpy, py-kaldi-asr, soupsieve, beautifulsoup4, meteofrance, pathlib, future, snips-nlu-parsers, snips-nlu-utils, python-crfsuite, tabulate, tqdm, sklearn-crfsuite, scikit-learn, semantic-version, docopt, num2words, scipy, PyYAML, pyaml, pyparsing, packaging, deprecation, plac, snips-nlu, RPi.GPIO, rpi-ws281x, pyalsaaudio
Exception:
Traceback (most recent call last):
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/pip/commands/install.py", line 360, in run
    prefix=options.prefix_path,
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/pip/req/req_install.py", line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/pip/req/req_install.py", line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/pip/wheel.py", line 345, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/pip/wheel.py", line 323, in clobber
    shutil.copyfile(srcfile, destfile)
  File "/usr/lib/python3.5/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission non accordée: '/home/pi/pynab/venv/lib/python3.5/site-packages/numpy/__pycache__/dual.cpython-35.pyc'
++ s=2
++ echo 'install.sh: Error on line 87: venv/bin/pip install -r requirements.txt'
install.sh: Error on line 87: venv/bin/pip install -r requirements.txt
++ exit 2
```

pguyot, 2019-05-04:
Apparemment, des fichiers dans venv/lib appartiendrait à root et pas à l'utilisateur pi.
Je propose :
```sudo chmod -R pi:pi /home/pi/pynab/venv```

BorisFR, 2019-05-04:
Alors, après un clown (et pas chmod ;))

```
(...)
Requirement already satisfied: pyparsing>=2.0.2 in ./venv/lib/python3.5/site-packages (from packaging->deprecation<3,>=2->snips_nlu->-r requirements.txt (line 16))
+ '[' 1 -eq 0 ']'
++ grep -cE '^local +all +all +trust'
++ sudo grep local /etc/postgresql/9.6/main/pg_hba.conf
+ trust=1
+ '[' 1 -ne 1 ']'
+ '[' '!' -e /etc/nginx/sites-enabled/pynab ']'
+ psql -U pynab -c ''
+ venv/bin/python manage.py migrate
Traceback (most recent call last):
  File "manage.py", line 15, in <module>
    execute_from_command_line(sys.argv)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
(...)
  File "/home/pi/pynab/nabd/asr.py", line 2, in <module>
    import numpy as np
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
ImportError: cannot import name 'core'
++ s=1
++ echo 'install.sh: Error on line 147: venv/bin/python manage.py migrate'
install.sh: Error on line 147: venv/bin/python manage.py migrate
++ exit 1
```

pguyot, 2019-05-04:
Encore un problème avec numpy !
numpy est installé depuis la version précompilée, ce qui n'était peut-être pas une super idée, ça échoue peut-être de fait de petites différences entre votre système et celui que j'ai utilisé.
Je vous propose de désinstaller numpy à la main et de le réinstaller.
```
pip uninstall numpy
pip --no-cache-dir install numpy
```

BorisFR, 2019-05-05:
```
  Successfully uninstalled numpy-1.16.3
pi@nabaztag:~/pynab $ venv/bin/pip --no-cache-dir install numpy
Collecting numpy
  Downloading https://files.pythonhosted.org/packages/93/48/956b9dcdddfcedb1705839280e02cbfeb2861ed5d7f59241210530867d5b/numpy-1.16.3.zip (5.1MB)
    100% |████████████████████████████████| 5.1MB 987kB/s 
Installing collected packages: numpy
  Running setup.py install for numpy ... done
Successfully installed numpy-1.16.3
```

nouveau souci :)
```
Requirement already satisfied: pyparsing>=2.0.2 in ./venv/lib/python3.5/site-packages (from packaging->deprecation<3,>=2->snips_nlu->-r requirements.txt (line 16))
+ '[' 1 -eq 0 ']'
++ grep -cE '^local +all +all +trust'
++ sudo grep local /etc/postgresql/9.6/main/pg_hba.conf
+ trust=1
+ '[' 1 -ne 1 ']'
+ '[' '!' -e /etc/nginx/sites-enabled/pynab ']'
+ psql -U pynab -c ''
+ venv/bin/python manage.py migrate
Traceback (most recent call last):
  File "manage.py", line 15, in <module>
    execute_from_command_line(sys.argv)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/__init__.py", line 375, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/base.py", line 323, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/base.py", line 361, in execute
    self.check()
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/base.py", line 390, in check
    include_deployment_checks=include_deployment_checks,
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/commands/migrate.py", line 65, in _run_checks
    issues.extend(super()._run_checks(**kwargs))
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/management/base.py", line 377, in _run_checks
    return checks.run_checks(**kwargs)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/checks/registry.py", line 72, in run_checks
    new_errors = check(app_configs=app_configs)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/checks/urls.py", line 40, in check_url_namespaces_unique
    all_namespaces = _load_all_namespaces(resolver)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/core/checks/urls.py", line 57, in _load_all_namespaces
    url_patterns = getattr(resolver, 'url_patterns', [])
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/utils/functional.py", line 80, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/urls/resolvers.py", line 579, in url_patterns
    patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/utils/functional.py", line 80, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/urls/resolvers.py", line 572, in urlconf_module
    return import_module(self.urlconf_name)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/pynab/nabweb/urls.py", line 22, in <module>
    path('nab8balld/', include('nab8balld.urls')),
  File "/home/pi/pynab/venv/lib/python3.5/site-packages/django/urls/conf.py", line 34, in include
    urlconf_module = import_module(urlconf_module)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/pynab/nab8balld/urls.py", line 2, in <module>
    from .views import SettingsView
  File "/home/pi/pynab/nab8balld/views.py", line 5, in <module>
    from .nab8balld import Nab8Balld
  File "/home/pi/pynab/nab8balld/nab8balld.py", line 2, in <module>
    from nabcommon.nabservice import NabService
  File "/home/pi/pynab/nabcommon/nabservice.py", line 7, in <module>
    from nabd import nabd
  File "/home/pi/pynab/nabd/nabd.py", line 7, in <module>
    from .asr import ASR
  File "/home/pi/pynab/nabd/asr.py", line 5, in <module>
    from kaldiasr.nnet3 import KaldiNNet3OnlineModel, KaldiNNet3OnlineDecoder
ImportError: libgfortran.so.3: cannot open shared object file: No such file or directory
++ s=1
++ echo 'install.sh: Error on line 147: venv/bin/python manage.py migrate'
install.sh: Error on line 147: venv/bin/python manage.py migrate
++ exit 1
```

pguyot, 2019-05-10:
Je vais essayer de faire une nouvelle image pour les v1 avec les dernières fonctionnalités (météo et triple clic pour éteindre), d'autant qu'il n'est pas nécessaire de charger les modèles d'ASR sur cette première génération. Ceci permettra aussi de corriger l'incompatibilité de version de numpy.

En attendant, il vous manque libgfortran.so.3 (qui ne sert que pour la reconnaissance vocale).

Sur un v1 qui fonctionne ici :
````
$ dpkg -S /usr/lib/arm-linux-gnueabihf/libgfortran.so.3
libgfortran3:armhf: /usr/lib/arm-linux-gnueabihf/libgfortran.so.3
````

Du coup vous pouvez tenter :
````
sudo apt-get install libgfortran3
````

BorisFR, 2019-05-13:
Maintenant à nouveau complètent opérationnel avec la météo en plus :)
Merci !

closed 2019-05-13 by BorisFR