nabaztag2018/pynab #12: Make the Pi configurable with no SSH or SD access

pguyot, 2019-07-13:
Many users will be provided with a pre-loaded SD card. Make sure the Nabaztag can be configured with no SSH or SD card access, especially Wifi settings.

Possible solutions include AP mode and bluetooth.
Also make sure the Nabaztag functions properly in degraded mode with no network access at all.

srcshelton, 2020-06-13:
Crazy thought, but you could flash the Nabaztag LEDs to indicate state, and use the ears as input devices... potentially with a web-app running on a mobile device or a mobile app to interpret the flashes and output user-friendly status information.  Or, of course, with speech-synthesis or pre-recorded samples the Nabaztag could speak back the input on change.

I like the idea of rotating one ear to set individual digits of an IP address (assuming the motor encoders can have the resolution to detect 36º increaments) and a half-turn of the other ear (like a one-armed bandit machine!) to confirm the input!

If off-line speech recognition (even only of individual digits) could be added, then you could simply prompt the user to say the desired IP address a digit at a time.

f-laurens, 2021-05-30:
> Possible solutions include AP mode and bluetooth.

An elegant and easy (?) solution would be to package this: https://github.com/schollz/raspberry-pi-turnkey
on the Pynab images.

f-laurens, 2022-04-10:
Note also that recent version of the [Raspberry PI imager](https://www.raspberrypi.com/software/) allow:
- flashing custom (already downloaded) images, like Pynab images;
- configuring (at least for Paspberry Pi OS based images) image settings: hostname, Wifi setup, SSH setup, user/password.

pguyot, 2022-06-04:
Fixed with comitup.

closed 2022-06-04 by pguyot