nabaztag2018/pynab #15: Missing admin/info web interface

pguyot, 2019-07-13:
The web interface should display some admin information about the status of the Pi: installed version, network info, hostname, etc. and allow some basic operations avoiding SSH access.

See also #12

mehdichaouch, 2021-04-11:
One other project also in Python, for Raspberry, with a nice admin/info features is [OctoPrint](https://github.com/OctoPrint/OctoPrint), maybe part of code could be duplicate like the shutdown/restart menu.
What di you think ?

f-laurens, 2021-04-12:
> One other project also in Python, for Raspberry, with a nice admin/info features is [OctoPrint](https://github.com/OctoPrint/OctoPrint), maybe part of code could be duplicate like the shutdown/restart menu.
> What do you think ?
No need to duplicate here, since shutdown/restart was already implemented in PR #126.

mehdichaouch, 2021-04-27:
As I see in http://nabaztag.local/system-info/ missing issue elements are :
- network info
- hostname

That's right?

f-laurens, 2021-05-18:
Shouldn't this issue considered closed now by #126 #185 #190?

closed 2021-05-18 by nabaztag2018