nabaztag2018/pynab #16: Web interface lacks documentation/help

pguyot, 2019-07-13:
Typically, information led patterns (e.g. for the weather service) should be documented on the website.

sanderant, 2020-02-22:
The current help page just links to https://tagtagtag.fr/help.html which is pretty sparse. It makes more sense to host some of the documentation locally, maybe have a link to the the internet as fallback in the event things change again.

The advantage of this approach would be that the help would be relevant to that version of the software in the event you don't update or choose not to.

Let's start with the led patterns. :) I'm not sure if I should be worried with yellow dots and pink underneath!

f-laurens, 2021-05-18:
Weather forecast LEDs patterns help may presumably be 'stolen' from https://www.tagtagtag.fr/weather.html
But do we have somewhere some documentation for:

- rain alert patterns;
- air quality patterns.

mehdichaouch, 2021-05-18:
I asked the same question for air quality pattern, in [discussions#227](https://github.com/nabaztag2018/pynab/discussions/227), but it was deleted 🤔 😢

@pguyot what do you think about adding a folder `docs` folder for package reference documentation and moving weather documentation there?

nabaztag2018, 2021-05-18:
Here is the animation for rain alert (video), i'm going to ask Antoine for Air quality pattern.

https://user-images.githubusercontent.com/44699852/118674616-38591000-b7fa-11eb-938a-174ad9d7d99f.mp4

f-laurens, 2021-05-18:
> Here is the animation for rain alert (video)

Thanks. The same converted to GIF (to have same format as weather forecast patterns):
[image]

mehdichaouch, 2021-05-18:
> Thanks. The same converted to GIF (to have same format as weather forecast patterns):

Thanks, The same in square ratio
[image]

f-laurens, 2021-05-25:
> i'm going to ask Antoine for Air quality pattern.

Any progress on this?

PR #255 (2021-05-29): Revamp home page and add help page
Resolves #16.
Optimize screen real estate on nabweb home page by having services on 2 columns as on services page.
Slightly cleanup/amend presentation text, while keeping the original spirit.
Correct Mastodon reference to make it more generic (https://joinmastodon.org).

Add new help page (linked from `Help` menu) with local content and appropriate internet links.
Together with PRs #248 and #250, this should mostly resolve #16.
closed 2021-05-29 by pguyot