nabaztag2018/pynab #26: Better handle locations with weather forecast

pguyot, 2019-07-13:
User enters a location in weather forecast web interface, yet the service is based on the first result returned by meteofrance-py instead of displaying all results and allow user to pick one.

Revealed by one of unit tests on meteofrance-py.
Requires https://github.com/pguyot/meteofrance-py/commit/a1c0d4333b92dfc1d1024908a48f37ed1462ba1f and maybe more changes on meteofrance-py

srcshelton, 2020-06-13:
... additionally, the selection of cities in the Britain (and around the world?) available on meteofrance.com is very limited... it knows about Londonderry in Northern Ireland, but it doesn't know about the capital of the UK! 😅

(See https://www.tagtagtag.fr/forum/showthread.php?tid=210)

Would it be preferable to make the weather provider pluggable too?

nguillaumin, 2020-07-26:
I think it knows about London, but one has to use the French name for it: http://www.meteofrance.com/mf3-rpc-portlet/rest/lieu/facet/previsions/search/Londres

In any case, I think the logic of picking the first result of the list and updating the configuration with its name is a bit flawed. Here's what happens for me:
- I enter "Berne" in the weather configuration (Berne in Switzerland)
- The Météo France API first result is correct, but the display name is `Berne (Suisse)`. This is the name saved in the database
- Next time the service runs, it searches again but this time for `Berne (Suisse)`
- And this time the API returns a different result `Suisse (57340)` - Name of a French village in Moselle.
- Subsequent service runs use that last match, and not what I entered initially

srcshelton, 2020-07-26:
> I think it knows about London, but one has to use the French name for it: http://www.meteofrance.com/mf3-rpc-portlet/rest/lieu/facet/previsions/search/Londres

I should really have tried/guessed this 😅

> * Subsequent service runs use that last match, and not what I entered initially

So really, at the very least, the original search should be stored rather than the returned display-name?

f-laurens, 2021-05-17:
Isn't this issue fixed by PR #154?

omevel, 2021-05-18:
Yep

closed 2021-05-18 by omevel