nabaztag2018/pynab #37: Base pulsing color should reflect networking status

pguyot, 2019-11-30:
When idle, the base is currently pulsing in Fuchsia. It should pulse in a different color depending on Wifi status.

f-laurens, 2021-05-31:
Fine.
- How many network/WiFi 'states' do we need?
 I guess only two: _On_ (Wifi interface connected to SSID), and _Off_ (Wifi interface down or not connected to a SSID)
 This should not be too difficult to get, either from the `iwgetid` command or some Python module.
- But then where would be the hooks/places in nabd where to set the pulse according to this state?
- And finally, what color do we want for each state? (are there some historical colors, from Violet time?)

mehdichaouch, 2021-05-31:
I think we can have a third state, when connected to the Wi-Fi but with no internet.
What we have usually it's a periodical ping to 1.1.1.1 (exemple), if ping return timeout this mean no Internet.