nabaztag2018/pynab #48: Add interface to adjust sound volumes

pguyot, 2019-12-21:
Sound volumes can be adjusted by editing `/var/lib/tagtagtag-sound/mixer.conf`
A web interface should be provided.

pguyot, 2020-01-22:
After /var/lib/tagtagtag-sound/mixer.conf is modified, a USR1 signal should be sent to tagtagtag-mixerd daemon (/run/tagtagtag-mixerd.pid).
Cf : https://github.com/pguyot/wm8960/tree/tagtagtag-sound

srcshelton, 2020-06-13:
On the Nabaztag:tag, can the RPi read-back the state of the rear volume control, or is this device simply an in-line analogue scaler between the TagTagTag audio output and the speaker?

(The above link suggests that the volume control might be exposed on GPIOs 22 & 27, and https://www.tagtagtag.fr/files/tagtagtag_various_advanced_things.pdf that this only gives 4 possible states: off, low, medium, and high - if this is correct then is 'off' the lowest quarter of the wheel's travel, or is 'off' only when the wheel clicks, and then the remainder of it's travel is equally split three ways between low, medium, and high?)