nabaztag2018/pynab #59: Play a sound after the first boot

232enero, 2020-01-02:
Users say they don't really realize Nabaztag has started or is alive (there is just the violet led beneath). 

Here's a list of potential sounds (it could only be french sounds, as we don't know what the user language setting is) : http://www.tagtagtag.fr/various/Nabaztag_first_boot.zip

BoOsteous, 2020-02-28:
Hi, I proposed a PR for this, i did only include one sound for now as the others were not really matching in my opinion (some are talking of feature that does not exist yet, like the name of the rabbit to configure, radio streaming...) 
So iprefered to put this FR sound for now.

Also, the sound is played after each boot of the rabbit, not only the first one...
If it's not what we want please tell me, but i think for now it's better like that because it help to know that the boot is complete.

232enero, 2020-03-10:
Hello !
Thanks for the PR!
A long time ago, there was a sound played by the Nabaztag at boot, it was a very simple sound (MIDI I guess), a few notes. I'm going to try to find it, it would make a good boot sound. I keep you posted !

srcshelton, 2020-06-13:
Not sure of the licence these files should be under, or whether they're the original Violet sound samples, but for authenticity might some of the MP3s in this repo be helpful?

https://sourceforge.net/p/nabaztaglives/code/ci/master/tree/www/vl/

f-laurens, 2021-04-24:
> A long time ago, there was a sound played by the Nabaztag at boot, it was a very simple sound (MIDI I guess), a few notes. 

Any sign of this simple boot sound?

sgillott, 2021-04-24:
Was this it? Or was there a different one for the tag:tag?
https://www.youtube.com/watch?v=ewZSM0BvJSs

nabaztag2018, 2021-04-25:
Yes, exactly. Jean-Jacques Birgé, the Nabaztag sound designer could not found it. I'm going to try to get it from Sylvain Huet as it was embedded in the VM.

PR #180 (2021-04-27): Play random 'boot' chime  on startup or reconfig.
Resolves #59 .
This PR is a variant of PR #124 (thanks @BoOsteous!) on which it tries to improve by
- slightly cleaning up code;
- playing the 'boot' chime both on startup and reconfig (locale change);
- providing several boot chimes (in `nabd/sounds/boot`) , linked to existing "respiration" sounds.

It also fixes loss of bottom LED pulse after reconfig.
closed 2021-04-27 by nabaztag2018