nabaztag2018/pynab #72: Surprise / Add the possibility to play specific sounds on a day or a period

232enero, 2020-01-13:
For example, Valentine's day sounds (or NYE...).

232enero, 2020-02-08:
Valentine's day is in a week :-) 
Here are the sound (french only !).
[saint valentin.zip](https://github.com/nabaztag2018/pynab/files/4173978/saint.valentin.zip)

232enero, 2020-02-08:
Nouvel an et Noel (french)
[noel.zip](https://github.com/nabaztag2018/pynab/files/4174045/noel.zip)
[nouvel an.zip](https://github.com/nabaztag2018/pynab/files/4174046/nouvel.an.zip)

f-laurens, 2021-05-01:
> Nouvel an et Noel (french)

Integrated in https://github.com/nabaztag2018/pynab/commit/4bfc79390acaac5e74bb792c8692e8416311b61c.

f-laurens, 2021-05-01:
> Valentine's day is in a week :-)  Here are the sound (french only !).

Integrated in https://github.com/nabaztag2018/pynab/commit/c5a876f366b7aa24b6529f52ab2b7df1fa0e7ffc.

mehdichaouch, 2021-05-01:
J'ai tapé des barres de rires en écoutant un peu au hasard les samples 👏  TIP TOP.

f-laurens, 2021-05-21:
Is this issue relevant?
Users can customize this to their taste by adding directories `nabsurprised/sounds/nabsurprised/mm-dd` or `nabsurprised/sounds/fr_FR/nabsurprised/mm-dd` containg sound files, which will then be played on day _dd_ of month _mm_.

mrcroupotin, 2021-05-22:
I have done it for the birthday of my family. So great 😀.
I have posted some instructions in the forum. 
But, not everyone will have the ability to do it.
And probably others users will have their own idea of messages they want to play on a specific day. 
It would be much simpler thru a web interface, no?

f-laurens, 2021-05-22:
> It would be much simpler thru a web interface, no?

What would the specification for this interface be?

mrcroupotin, 2021-05-24:
An interface with a date selector and a File chooser item would do the trick. A folder with the mm-dd naming is then created and the files chosen by the user are uploaded onto it.

f-laurens, 2021-05-24:
> An interface with a date selector and a File chooser item would do the trick. A folder with the mm-dd naming is then created and the files chosen by the user are uploaded onto it.

Fine. A locale selector will be needed too, depending on whether the user wants to consider the file to upload as locale-specific or not.
Now what we need is a PR to implement this. Any volunteer?...

Also, what about the cleanup (if/when users want to remove files they have uploaded)?