nabaztag2018/pynab #77: Air Quality - requests flooding

Killator, 2020-01-21:
Bonjour,

Je constate que mes 2 lapins (Pynab v0.7.0) floodent continuellement le site "api.waqi.info" lorsque le service qualité de l'air est actif, peu importe le choix de l'indice [AQI] ou [PM 2.5]. Environ 1 requête toutes les 10sec.

Le flood s'arrête dès que je sélectionne [Aucun] puis [Enregistrer].

Constat réalisé à partir de la surveillance des requêtes DNS sur mon LAN via PiHole.
Comportement parfaitement reproductible même après reboot du lapin.

232enero, 2020-01-21:
Bonjour, merci !
Est ce que vous avez un accès SSH sur vos lapins ? (pour m'envoyer les logs)

Killator, 2020-01-21:
Oui, pas de soucis. Ca se trouve où ? ;)

Killator, 2020-01-21:
Tout simplement ^^'

Ca boucle sur ce pattern dans daemon.log:
`
Jan 21 13:17:03 keynote systemd[1]: Started Nabaztag Air quality daemon.
Jan 21 13:17:09 keynote python[4557]: Traceback (most recent call last):
Jan 21 13:17:09 keynote python[4557]:   File "/home/pi/pynab/nabairqualityd/aqicn.py", line 43, in _fetch_airquality_data
Jan 21 13:17:09 keynote python[4557]:     indice_pm25 = json_data["data"]["iaqi"]["pm25"]["v"]
Jan 21 13:17:09 keynote python[4557]: KeyError: 'pm25'
Jan 21 13:17:09 keynote python[4557]: During handling of the above exception, another exception occurred:
Jan 21 13:17:09 keynote python[4557]: Traceback (most recent call last):
Jan 21 13:17:09 keynote python[4557]:   File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
Jan 21 13:17:09 keynote python[4557]:     "__main__", mod_spec)
Jan 21 13:17:10 keynote python[4557]:   File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
Jan 21 13:17:10 keynote python[4557]:     exec(code, run_globals)
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabairqualityd/nabairqualityd.py", line 138, in <module>
Jan 21 13:17:10 keynote python[4557]:     NabAirqualityd.main(sys.argv[1:])
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 150, in main
Jan 21 13:17:10 keynote python[4557]:     service.run()
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 312, in run
Jan 21 13:17:10 keynote python[4557]:     raise ex
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 273, in service_loop
Jan 21 13:17:10 keynote python[4557]:     config,
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 429, in perform
Jan 21 13:17:10 keynote python[4557]:     info_data = await self._do_fetch_info_data(config)
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 499, in _do_fetch_info_data
Jan 21 13:17:10 keynote python[4557]:     new_info = await self.fetch_info_data(config)
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabairqualityd/nabairqualityd.py", line 89, in fetch_info_data
Jan 21 13:17:10 keynote python[4557]:     await sync_to_async(client.update)()
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/asgiref/sync.py", line 244, in __call__
Jan 21 13:17:10 keynote python[4557]:     return await asyncio.wait_for(future, timeout=None)
Jan 21 13:17:10 keynote python[4557]:   File "/usr/lib/python3.7/asyncio/tasks.py", line 388, in wait_for
Jan 21 13:17:10 keynote python[4557]:     return await fut
Jan 21 13:17:10 keynote python[4557]:   File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
Jan 21 13:17:10 keynote python[4557]:     result = self.fn(*self.args, **self.kwargs)
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/asgiref/sync.py", line 277, in thread_handler
Jan 21 13:17:10 keynote python[4557]:     return func(*args, **kwargs)
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabairqualityd/aqicn.py", line 33, in update
Jan 21 13:17:10 keynote python[4557]:     self._fetch_airquality_data()
Jan 21 13:17:10 keynote python[4557]:   File "/home/pi/pynab/nabairqualityd/aqicn.py", line 71, in _fetch_airquality_data
Jan 21 13:17:10 keynote python[4557]:     raise aqicnError(err)
Jan 21 13:17:10 keynote python[4557]: nabairqualityd.aqicn.aqicnError: 'pm25'
Jan 21 13:17:10 keynote systemd[1]: nabairqualityd.service: Main process exited, code=exited, status=1/FAILURE
Jan 21 13:17:10 keynote systemd[1]: nabairqualityd.service: Failed with result 'exit-code'.
Jan 21 13:17:11 keynote systemd[1]: nabairqualityd.service: Service RestartSec=1s expired, scheduling restart.
Jan 21 13:17:11 keynote systemd[1]: nabairqualityd.service: Scheduled restart job, restart counter is at 648.
Jan 21 13:17:11 keynote systemd[1]: Stopped Nabaztag Air quality daemon.
`

Et voici dans nabairqualityd.log
`
DEBUG 2020-01-21 13:10:06,322 Starting new HTTP connection (1): api.waqi.info:80
DEBUG 2020-01-21 13:10:06,980 http://api.waqi.info:80 "GET /feed/here/?token=4cf7f445134f3fb69a4c3f0e5001e507a6cc386f HTTP/1.1" 200 None
INFO 2020-01-21 13:10:12,049 nabairqualityd started
DEBUG 2020-01-21 13:10:14,337 Using selector: EpollSelector
DEBUG 2020-01-21 13:10:14,369 process nabd packet: {'type': 'state', 'state': 'idle'}
INFO 2020-01-21 13:10:14,446 compute_next saved_date=2020-01-21 11:32:48.279857+00:00
INFO 2020-01-21 13:10:14,448 compute_next saved_date < now
INFO 2020-01-21 13:10:14,452 fetch_info_data type = info
DEBUG 2020-01-21 13:10:14,454 index_airquality=0
`

N'hésitez pas s'il vous faut d'autres choses...

Killator, 2020-01-21:
Dsl pour la mise en forme... :-s

232enero, 2020-01-21:
Ok, je vois.
Vous habitez où ? Ou plutôt il habite où sur Internet ce lapin ?

Killator, 2020-01-21:
Dans son terrier: 109 . 190 . 125 . 82 (connexion OVH / Proche de Lille)

232enero, 2020-01-21:
Quand vous appelez cette url depuis un navigateur ça donne quoi ?
http://api.waqi.info/feed/here/?token=4cf7f445134f3fb69a4c3f0e5001e507a6cc386f

Killator, 2020-01-21:
La réponse semble conforme... Je précise qu'il n'y pas de restriction particulière sur mon réseau, pas plus pour le lapin que pour l'ordi qui a fait cette requête et j'ajoute que toutes les requêtes que font les lapins sont passantes dans PiHole.

`{"status":"ok","data":{"aqi":27,"idx":8613,"attributions":[{"url":"https://www.atmo-hdf.fr/","name":"Atmo en Hauts-de-France - Pas-de-Calais","logo":"France-AmtoNPC.png"},{"url":"https://waqi.info/","name":"World Air Quality Index Project"}],"city":{"geo":[50.67003055555521,3.0765610522280102],"name":"Marcq-en-Baroeul, Nord-Pas-de-Calais, France","url":"https://aqicn.org/city/france/nord-pas-de-calais/marcq-en-baroeul"},"dominentpol":"pm10","iaqi":{"co":{"v":0.1},"no2":{"v":27.4},"o3":{"v":7.8},"pm10":{"v":27},"so2":{"v":0.8},"wg":{"v":15.4}},"time":{"s":"2020-01-21 10:00:00","tz":"+01:00","v":1579600800},"debug":{"sync":"2020-01-21T20:17:44+09:00"}}}`

pguyot, 2020-01-21:
Tout simplement pas de PM25 à Marcq-en-Baroeul.

omevel, 2020-01-21:
Voilà. C'est aussi simple que ça.
Je corrige et je fais passer ça dans la prochaine release.

232enero, 2020-01-21:
ça vous interesse de passer sur la branche "master" et nous aider de temps en temps pour le debug ? faudra faire quelques git commandes, sinon pas de soucis !

pguyot, 2020-01-21:
Il faudrait idéalement permettre d'afficher chacun des indices disponibles individuellement.

Killator, 2020-01-21:
"Marcq en Baroeul, the place to be" ! ^^'

Merci pour votre réactivité en tous cas !

Je veux bien essayer de vous aider en basculant sur la master. Par contre, je suis pas super à l'aise avec GIT, je vais tâtonner au début je pense, mais ça sera l'occasion de me perfectionner justement :-)

Vous auriez un mod'op pour faire la bascule ?

pguyot, 2020-01-21:
Pour mémoire, il faut activer le SSH
Ensuite :

$ cd /home/pi/pynab
$ git checkout master

Puis aller sur l'interface web pour faire la mise à jour ou alors, toujours en ligne de commande : 
$ bash upgrade.sh

232enero, 2020-01-21:
Voilà, c'est bon, vous pouvez essayer :)
Merci !

closed 2020-01-21 by 232enero

Killator, 2020-01-21:
"Attention, apparemment, votre Nabaztag est monté sur la branche master : la mise à jour avec l'interface web peut échouer. Et là, que la force soit avec vous, il faudra terminer en SSH."

Mouarf... ^^

En tous cas, je confirme, le service "Air Quality" tourne bien maintenant !

Votre boulot est vraiment top en tous cas... Mention spéciale pour Nabblocky (poke @pguyot) qui est ultra ludique et pédagogique pour initier un enfant à la programmation. J'ai fais les premières "leçons" avec mon fils de 5 ans, il a accroché instantanément !

Bref, au plaisir de vous débusquer les trucs tordus ;-)

232enero, 2020-01-21:
Ah oui, je confirme, Nabblocky c'est la killer app !

pguyot, 2020-01-21:
Pour la programmation avec nabblockly, il faut quand même savoir lire. Ou alors il faudrait remplacer les textes par des images, un peu comme sur Scratch Jr ou Lego Powered Up.

Cette leçon sur les boucles venait après une première leçon faite sur un autre matériel pédagogique. Je connaissais le groupe, du coup j'ai mis un exercice assez dur à la fin pour un élève de CM1 qui se débrouillait bien.

J'ai essayé le premier chapitre de code.org sur des enfants qui ne savaient pas lire ou bien lire (mon cadet de 4 ans, des enfants en début de CP), c'est plutôt réussi : https://studio.code.org/s/course1

pguyot, 2020-01-21:
Je précise que dans nabblockly, les leçons sont très faciles à faire, c'est juste du HTML avec du Javascript pour vérifier la solution.

https://github.com/pguyot/nabblockly/blob/master/apps/nabblockly/priv/lessons/loops.html

commit 734e755f (2020-01-21): Fixed #77