nabaztag2018/pynab #80: [image] distribution avec paquet mal installé

nathanaelle, 2020-01-29:
Bonjour,

après un apt upgrade sur la derniere image, j'ai ce message :

> Error occurred during initialization of VM
> Server VM is only supported on ARMv7+ VFP
> E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.
> done.
> Errors were encountered while processing:
>  ca-certificates-java
>  openjdk-11-jre-headless:armhf
> E: Sub-process /usr/bin/dpkg returned an error code (1)

il peut être pertinent de reconstruire une nouvelle image sans ces paquets indisponible pour les raspberry π 0.

pguyot, 2020-01-29:
Nathanaelle, merci pour ce rapport. Effectivement, il y a un problème dans les paquets.

Le code qui fait ça est ici (`|| echo 0` ajouté au moment où ce paquet problématique est apparu dans les dépendances) :
https://github.com/nabaztag2018/pynab/blob/release/setup#L89

Une idée de comment faire ça proprement ? Pas eu le temps de chercher.
N'hésitez pas à proposer un PR sur la branche releng pour corriger le problème (le script n'est pas sur la branche master, c'est un artéfact/reste de la mise en place du déploiement continu).

nathanaelle, 2020-01-29:
`erlang` a pour dépendance recommandée `erlang-jinterface` qui nécessite `jre-headless` ou un autre pile java.

le plus propre serait de forcer la non installation des paquets recommandés et suggérés.

cela se fait dans le fichier `apt.conf` ou en ajoutant un fichier dans `apt.conf.d` : 

> APT::Install-Recommends 0;
> APT::Install-Suggests 0;
> APT::Get::Install-Recommends 0;
> APT::Get::Install-Suggests 0;

pguyot, 2020-01-29:
Merci pour l'analyse. Effectivement, on n'a pas besoin de jinterface pour nabblockly.
Si on désactive tous les paquets recommandés et suggérés, ça participera de la réduction de la taille de l'image, mais il faut vérifier qu'il n'y a pas des paquets recommandés ou suggérés dont on a en fait besoin (et qui devraient être indiqués).
On peut le faire (plus rapidement) juste pour erlang. Je teste.

pguyot, 2020-01-29:
Vu le log travis, seul erlang avait des dépendances recommandées.
Pour que ce soit le plus propre possible, j'ai ajouté `--no-install-recommends` aux deux autres lignes du script.

Ce sera donc corrigé avec la prochaine image.
Merci encore.

commit ae980ed5 (2020-01-29): Do not install recommended packages if any pops in (fix #80)

Among installed packages, only erlang had recommended dependencies.
To make sure we indeed list all required packages, explicitely do not
install recommended packages.
closed 2020-01-29 by pguyot