nabaztag2018/pynab #81: High CPU usage until I changed volume

Lyokovic, 2020-01-29:
I was testing some code on the nabaztag and it suddently became unresponsive (through ssh).
I saw that the CPU usage was very high, and `tagtagtag-mixerd` with `alsactl` and `rngd` were the most CPU demanding processes for a long time.
Even after reboot there still were this high CPU load. Other services could not work normally.

Just by moving the volume wheel the CPU load dropped and the system became responsive again.

It might be the volume wheel being in a weird position and sending many values overflowing the mixer daemon,

pguyot, 2020-01-29:
Uhuh. Anything in the logs? Probably /var/log/daemon.log or in dmesg?
Volume button’s driver changes alsa volume on rising GPIO interrupts. Wondering how this could happen while the button is not moved.

Lyokovic, 2020-01-29:
Yes that was strange. I moved the wheel because I saw the mixer seemed to be related (and I already rolled back my unrelated code changes without any effect).

I did not see any logs in the `dmesg` or in the journal (`journalctl`), apart from the `nabweb` service which could not run normally because of the CPU usage I guess.

BoOsteous, 2020-02-05:
Hi, i think i had the same issue today, about 6 hours after the upgrade to 7.2.
Not sure this is related to the update but before i did'nt had audio issues, and now i noticed that the sound was cutting a lot.
I saw this post and looked at the processes :
[image]

Then i tried to move the volume down a little down, and everything comes normal again.
Strange !

Will tell you if i can reproduce this issue.

f-laurens, 2021-06-05:
Has anybody come up with a reproducible test case for this?

pguyot, 2022-03-04:
Closing this as we failed to reproduce it. Feel free to reopen it if the problem still occurs.

closed 2022-03-04 by pguyot