nabaztag2018/pynab #84: Allow users to specify location for Air Quality

pguyot, 2020-02-01:
Automatic discovery of location can be wrong.
See: https://www.tagtagtag.fr/forum/showthread.php?tid=57

Create an interface, following Weather service.

fffab, 2020-02-09:
Hi @pguyot 
I did check the api used by this service. I found out that it can be called by latitude/longitude.
(the api return correct this on the nearest know city)

My actual skills in python are low, but i managed to make it work partially.
I'm stuck on migrating to add the two fields created.

```ssh
pi@nabaztag:~/pynab $ python manage.py migrate nabairqualityd
  File "manage.py", line 14
    ) from exc
         ^
SyntaxError: invalid syntax
```
i'm not sure of the syntax, but if i understood correctly, python manage.py should work without any arguments but give out the same error

If you, or someone wish to view and finish the job, or point me to the right direction, it would be cool.
https://github.com/fffab/pynab/commit/2237c18663bfd0eeccb7598e3a95bd39ab6cbb25

As this is my first message in this project, I must add a big thank you to all of the people/team who have made this nabaztag reborn possible !

pguyot, 2020-02-09:
Invalid syntax means you are using python 2, not python 3.
You probably would want to always use python from the virtual env, like this:

    $ ./venv/bin/python manage.py migrate nabairqualityd

Specifying by latitude and longitude is probably confusing. You probably should allow users to use either the default, IP-based geo-location, or allow them to perform a search.

fffab, 2020-02-09:
Thanks for the hint.
And, yes, best way will be to perform search.
I'm moving forward step by step ;)

PR #237 (2021-05-23): Air quality: switch from IP address based to true geolocalized AQICN API
Resolves #84.
IP address based geolocalization is often plain wrong.
For Air quality, switch to using geolocalized (latitude/longitude based) AQICN API, based on the location chosen for Weather, the default (IP address based) API being used only as a fallback.
closed 2021-05-23 by pguyot