nabaztag2018/pynab #85: Force the ear rotation and/or detection

Shadok, 2020-02-01:
Hello and thanks for all your work :)

My tag/tag has its left ear dead, so i am trying to repair it.
To access the motors, i had to disconnect the ears sensors, so they are both detected as broken in the status.

However, it would help to be able to rerun the detection or force the ear rotation even if they are detected as broken, to see if it's motor related or another issue.

Shadok, 2020-02-01:
Even a restart button would help if it would be quicker than an electrical reboot :)

pguyot, 2020-02-02:
You can already do this through SSH by unloading/reloading ears driver.

To enable SSH access, add a file named `ssh` to the visible FAT partition of the SD card.
Then log in with pi user (default password is raspberry, you can change it).

Then you can stop pynab with:

    cd /home/pi/pynab
    sudo venv/bin/python manage.py stop_all

Unload ears driver:

    sudo rmmod tagtagtag_ears

Reload it:

    sudo insmod /lib/modules/4.19.75+/kernel/input/misc/tagtagtag-ears.ko

Use `dmesg` to find out why driver is declaring the ear as broken.

Eventually, restart pynab with:

    cd /home/pi/pynab
    sudo venv/bin/python manage.py start_all

Or shutdown with:

    sudo halt

Or reboot with:

    sudo reboot

closed 2020-02-02 by pguyot