nabaztag2018/pynab #98: upgrade error 4/14 cr14 driver

Framb75, 2020-02-12:
+ echo 'Updating RFID driver - 4/14'
+ '[' -d /home/pi/cr14 ']'
+ cd /home/pi
+ git clone https://github.com/pguyot/cr14
Cloning into 'cr14'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 16 (delta 5), reused 16 (delta 5), pack-reused 0
Unpacking objects: 100% (16/16), done.
+ make
make: *** No targets specified and no makefile found.  Stop.
+ sudo touch /tmp/pynab.upgrade.reboot

After reboot Nabaztag is in loop rebooting again and again.

pguyot, 2020-02-12:
Thanks for the report!

pguyot, 2020-02-12:
Your system might be in a stuck state after failed upgrade.
You may want to do:

    cd /home/pi/cr14
    make && sudo make install

Framb75, 2020-02-12:
was already unstucked but thanks

pguyot, 2020-02-12:
"After reboot Nabaztag is in loop rebooting again and again." was another bug, which is now also fixed.

commit c26afe3a (2020-02-12): Fix install script (fix #98)
commit 355549e4 (2020-02-12): Fix crash when rfid driver is not loaded (fix #98)
closed 2020-02-12 by pguyot