nabaztag2018/pynab #101: Access 10543 from outside of the NabazTag

Ellerbach, 2020-02-13:
Is there a way to get connected to port 10543 from outside to be able to interact with the NabazTag and create services that will stand outside of the Pi?

pguyot, 2020-02-13:
nabd voluntarily only listens to the local interface.

This is done here (launched from command line):
https://github.com/nabaztag2018/pynab/blob/v0.7.4/nabd/nabd.py#L736
And here (launched from systemd):
https://github.com/nabaztag2018/pynab/blob/v0.7.4/nabd/nabd.socket#L2

ssh is disabled by default on release images, however you can easily setup a tunnel.

You can access a more limited and undocumented API from nabblockly, though, via HTTP on port 8080.

Ellerbach, 2020-02-14:
Thanks. 

> And here (launched from systemd):
> https://github.com/nabaztag2018/pynab/blob/v0.7.4/nabd/nabd.socket#L2

I've changed the 127.0.0.1 for 0.0.0.0 which should be any IP and rebooted but seems there is still no luck. I'd really like to intercept some of the events. So the nabblockly is not enough unfortunately.

pguyot, 2020-02-14:
You do not need to reboot the whole system, you only need to alter the systemd configuration file.
This is what I did, and it worked well:

    sudo sed -i -e 's|127.0.0.1|0.0.0.0|g' /lib/systemd/system/nabd.socket
    sudo systemctl daemon-reload
    sudo systemctl stop nabd.service
    sudo systemctl restart nabd.socket
    sudo systemctl start nabd.service

Let me know if it works for you.

Ellerbach, 2020-02-14:
> Let me know if it works for you.

Yes, worked! Found the trick in parallel looking at the install script. Thanks! Will close the Issue

closed 2020-02-14 by Ellerbach