nabaztag2018/pynab #102: Ears driver mise à jour web impossible

Framb75, 2020-02-14:
Ears driver indique 3 mises à jour  mais le bouton d'interface web reste grisé.

https://www.tagtagtag.fr/forum/showthread.php?tid=117

pguyot, 2020-02-16:
Even when there is no change on pynab (release or master branch), there can be changes on drivers, yet the upgrade button is (on purpose) not available.

Several users have been confused by this interface.

sanderant, 2020-02-22:
So is it correct that it only updates pynab and not the drivers?

pguyot, 2020-02-26:
Update button runs pynab's upgrade script which updates everything, but is only available when there is something to update for pynab.
We should get pynab's release branch to stick to specific commits or branches of the other projects.

pguyot, 2020-02-26:
Cf also #122

closed 2021-05-23 by omevel