nabaztag2018/pynab #108: nabd crashs in loop

cyril1929, 2020-02-16:
After update on the last commit on master and reboot, nabd crashs in loop.
log:
```
INFO 2020-02-16 09:00:30,951 nabd started
DEBUG 2020-02-16 09:00:36,261 Using selector: EpollSelector
DEBUG 2020-02-16 09:01:31,342 Traceback (most recent call last):
  File "/home/pi/pynab/nabd/nabd.py", line 150, in idle_worker_loop
    await self._do_transition_to_idle()
  File "/home/pi/pynab/nabd/nabd.py", line 134, in _do_transition_to_idle
    await self.nabio.move_ears_with_leds((255, 0, 255), left, right)
  File "/home/pi/pynab/nabd/nabio.py", line 40, in move_ears_with_leds
    current_left, current_right = await self.ears.get_positions()
TypeError: object tuple can't be used in 'await' expression
```

pguyot, 2020-02-16:
Oops. There was an inconsistency in the API interface, 419ac7f should fix this.

cyril1929, 2020-02-16:
It's not correct yet :
```
INFO 2020-02-16 09:13:26,195 nabd started
DEBUG 2020-02-16 09:13:30,601 Using selector: EpollSelector
INFO 2020-02-16 09:19:33,040 nabd started
DEBUG 2020-02-16 09:19:37,071 Using selector: EpollSelector
DEBUG 2020-02-16 09:21:01,884 Traceback (most recent call last):
  File "/home/pi/pynab/nabd/nabd.py", line 150, in idle_worker_loop
    await self._do_transition_to_idle()
  File "/home/pi/pynab/nabd/nabd.py", line 134, in _do_transition_to_idle
    await self.nabio.move_ears_with_leds((255, 0, 255), left, right)
  File "/home/pi/pynab/nabd/nabio.py", line 42, in move_ears_with_leds
    if not self.ears.is_broken(Ears.LEFT_EAR):
NameError: name 'Ears' is not defined
```

closed 2020-02-16 by pguyot

cyril1929, 2020-02-16:
It works.
Thank you for your reactivity!

commit 419ac7fd (2020-02-16): Fix inconsistency in get_positions interface (fix #108)
commit d44fd7ee (2020-02-16): Fix inconsistency in ears position cache. Also fix #108