nabaztag2018/pynab #110: quelle utilité à postgreSQL ?

nathanaelle, 2020-02-16:
l'OOM killer a été activé tuant dpkg et pgsql durant une mise à jour du systeme par apt.

postgreSQL est un excellent outil sur un serveur avec plusieurs Go de RAM, mais sur un raspberry pi il agit comme un bloatware si il n'est pas un minimum configuré.

sans vouloir lever un lièvre, est ce que SQLite ne serait pas la bonne solution sachant que c'est autrement plus adapté à un environnement embarqué comme le raspberry pi ?

j'ai la possibilité technique de déporter postgreSQL sur une machine avec plus de RAM, mais tout le monde n'a pas cette possibilité, et le fait que postgreSQL consomme à lui seul plus de 50% de la RAM du lapin me semble être un probleme à traiter puisqu'il empeche de faire un usage correct de la machine.

augmenter le swap n'est pas une solution, une partie de la solution me semble nécessiter entre autre une suppression du swap pour que les performances de cet innocent lagomorphe se trouvent améliorées.

pguyot, 2020-02-16:
Si PostgreSQL utilise plus de la moitié de la RAM de votre RaspberryPi, il y a un sérieux problème sur votre lapin. Normalement, PostgreSQL utilise très peu de RAM, environ 25 Mo, soit environ autant qu'un interprète Python.

Pourriez-vous étayer cette information par un log ou un top ?

nathanaelle, 2020-02-16:
immediatement apres le reboot :

```
postgres 485 0.0 4.4 199M 21M ? S 19:05 0:00 /usr/lib/postgresql/11/bin/postgres -D /var/lib/postgresql/11/main -c config_file=/etc/postgresql/11/main/postgresql.conf
postgres 488 0.0 1.3 199M 6M ? Ss 19:05 0:00 \_ postgres: 11/main: checkpointer
postgres 489 0.0 1.0 199M 5M ? Ss 19:05 0:00 \_ postgres: 11/main: background writer
postgres 490 0.0 1.6 199M 8M ? Ss 19:05 0:00 \_ postgres: 11/main: walwriter
postgres 491 0.0 1.2 199M 6M ? Ss 19:05 0:00 \_ postgres: 11/main: autovacuum launcher
postgres 492 0.0 0.7 59M 3M ? Ss 19:05 0:00 \_ postgres: 11/main: stats collector
postgres 493 0.0 1.0 199M 5M ? Ss 19:05 0:00 \_ postgres: 11/main: logical replication launcher
postgres 775 0.0 2.3 200M 11M ? Ss 19:05 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 776 0.0 2.3 200M 11M ? Ss 19:05 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 784 0.0 2.3 200M 11M ? Ss 19:05 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 809 0.0 2.3 200M 11M ? Ss 19:05 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 846 0.0 2.4 200M 11M ? Ss 19:06 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 849 0.0 2.4 200M 11M ? Ss 19:06 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 866 0.0 2.3 200M 11M ? Ss 19:06 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 871 0.0 2.4 200M 11M ? Ss 19:06 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 928 0.0 2.4 200M 11M ? Ss 19:06 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 931 0.0 2.4 200M 11M ? Ss 19:06 0:00 \_ postgres: 11/main: pynab pynab [local] idle
postgres 932 0.0 2.4 200M 11M ? Ss 19:06 0:00 \_ postgres: 11/main: pynab pynab [local] idle
```

21M + 6M + 5M + 8M + 6M + 3M + 5M + 11*11M = 175Mio de RSS

donc après un reboot, un lapin tout frais a un postgreSQL qui consomme 175 Mio de RSS c'est à dire 175Mio de RAM dédiée.

nathanaelle, 2020-02-16:
si vous ajoutez la dizaine de process python qui consomment au total un peu plus de 244Mio de RSS … cela fait un raspberry pi légèrement encombré.

```
root 508 0.3 5.1 65M 24M ? Ssl 19:05 0:05 /home/pi/pynab/venv/bin/python -m nabairqualityd.nabairqualityd
root 509 0.3 3.3 23M 16M ? Ss 19:05 0:04 /home/pi/pynab/venv/bin/python3.7 /home/pi/pynab/venv/bin/gunicorn --timeout 60 nabweb.wsgi
root 763 0.4 6.1 45M 29M ? R 19:05 0:06 \_ /home/pi/pynab/venv/bin/python3.7 /home/pi/pynab/venv/bin/gunicorn --timeout 60 nabweb.wsgi
root 510 0.2 4.6 45M 22M ? Ssl 19:05 0:04 /home/pi/pynab/venv/bin/python -m nabtaichid.nabtaichid
root 511 0.2 4.6 36M 22M ? Ss 19:05 0:04 /home/pi/pynab/venv/bin/python -m nabclockd.nabclockd
root 512 0.3 5.1 66M 24M ? Ssl 19:05 0:05 /home/pi/pynab/nabblockly/_build/default/rel/nabblockly/bin/nabblockly -Bd -K true -A30 -- -root /home/pi/pynab/nabblockly/_build/default/rel/nabblockly -progname home/pi/pynab/nabblockly/_build/default/rel/nabblockly/bin/nabblockly -- -home /root -- -noshell -noshell -noinput -boot /home/pi/pynab/nabblockly/_build/default/rel/nabblockly/releases/0.1.0/nabblockly -mode embedded -boot_var ERTS_LIB_DIR /usr/lib/erlang/lib -config /home/pi/pynab/nabblockly/_build/default/rel/nabblockly/releases/0.1.0/sys.config -sname nabblockly -setcookie nabblockly_cookie -heart -- foreground
root 808 0.0 0.2 1M 1M ? Ss 19:05 0:00 \_ erl_child_setup 1024
root 890 0.0 0.2 1M 1M ? Ss 19:06 0:00 \_ heart -pid 512
root 513 0.4 5.4 58M 26M ? Ssl 19:05 0:06 /home/pi/pynab/venv/bin/python -m nabmastodond.nabmastodond
root 516 0.2 4.6 45M 22M ? Ssl 19:05 0:04 /home/pi/pynab/venv/bin/python -m nabsurprised.nabsurprised
root 521 2.5 7.7 91M 37M ? Ssl 19:05 0:38 /home/pi/pynab/venv/bin/python -m nabweatherd.nabweatherd
root 524 0.2 4.5 45M 22M ? Ssl 19:05 0:03 /home/pi/pynab/venv/bin/python -m nab8balld.nab8balld

```

nathanaelle, 2020-02-16:
pour mieux comprendre les valeurs il me semble important de rappeler qu'il faut en général corriger les valeurs en arrondissant au multiple de 4Mio supérieur pour chaque process (c'est un peu plus compliqué mais l'idée est là).

les 4Mio correspondent à la taille de page allouée en mémoire, cette valeur peut varier et elle est donnée par :

```
root@nabaztag:~# getconf PAGE_SIZE
4096
```

pguyot, 2020-02-16:
Tout va bien sur votre lapin, je ne vois pas nabd qui prend l'essentiel de la RAM avec le modèle de reconnaissance vocale (surtout) et celui de NLU (en deuxième).

Les processus PostgreSQL partagent de la mémoire et ne prennent pas autant que vous le dites.
Sur un lapin, ici j'obtient 33 Mo avec la commande suivante :

    ps -u postgres o pid= | sed -E 's| +([0-9]+)|/proc/\1/smaps|' | xargs sudo grep ^Pss: | awk '{A+=$2} END{print A}'

Pour l'explication : https://www.depesz.com/2012/06/09/how-much-ram-is-postgresql-using/

La même commande (`ps -C python`) donne 282 Mo pour Python, 24 Mo pour gunicorn et 19 Mo pour Erlang (`ps -C beam.smp`), ce qui est l'impact de pynab dans son ensemble, on reste dans les limites de la RAM disponible sur le Pi Zero.

PostgreSQL utilise la configuration par défaut, n'hésitez pas à proposer un PR pour l'optimiser, même si de fait, il n'y a pas grand chose à gagner pour l'instant.

Quant à votre première question, PostgreSQL tourne très bien sur les Pi, SQLite ne répond pas à la possibilité d'être accédé par plusieurs processus (Python) en même temps.

closed 2020-02-16 by pguyot

nathanaelle, 2020-02-16:
merci pour votre réponse :) 

pour l'OOM killer qui a tué dpkg et postgreSQL, je suis certaine … exactement aussi certaine qu'un dpkg qui a planté et qu'il faille faire `dpkg --configure -a` pour finir la mise à jour des paquets.

vous indiquez que l'ensemble des applications python consomment 282 Mio sur un systeme avec 512 Mio … cela me semble de toute façon beaucoup de ressource.

A coté de cela, le fichier smaps est imprécis sur la valeur Pss puisqu'un arrondi par défaut est effectué.
Donc les 33 Mio que vous obtenez sont un minorant de la valeur réelle.

Voici un échantillon pour vous montrez l'erreur de calcul induite par le Pss.

```
Rss:                  60 kB
Pss:                   1 kB
Rss:                  12 kB
Pss:                   8 kB
Rss:                   4 kB
Pss:                   0 kB
Rss:                   4 kB
Pss:                   0 kB
Rss:                   8 kB
Pss:                   0 kB
```

Si vous me concédez l'odacieuse affirmation qu'aucune somme infinie de 0 permettra d'obtenir ni un 8 ni un 4, alors vous pouvez vous douter  du problème d'estimation qu'induit le Pss dans votre calcul, et qu'il est impossible que PostgreSQL ne consomme que 33Mio.

Dans le cas qui nous importe, il aurait été plus pertinent de disposer du mapcount en plus du Rss et Pss, mais le fichier smaps ne semble pas fournir l'information.

à la louche, l'essentiel des pages communes sont issues d'avant les forks, donc il pourrait etre possible de retrouver le mapcount en comptant le nombre de processus utilisant chaque plage mémoire commune.

Donc non PostgreSQL ne consomme pas que 33Mio, meme si effectivement, il ne consomme pas 175Mio non plus … mais une valeur entre les deux.

La configuration de PostgreSQL fait qu'il peut avoir 100 connexions au maximum chacune pouvant manipuler 4Mio de mémoire propre, le tout avec un shared_buffer de 128Mio, donc de toute façon, PostgreSQL n'est pas configuré pour une machine avec uniquement 512Mio.

Je n'ai pas encore regardé le code python, mais je m'étonne de la nécessité d'une bdd commune si il y a un processus central qui sert de hub à l'ensemble des services (le protocole JSON proposé n'est peut etre pas actif, ce qui fait que PostgreSQL fait office de queue pub/sub pour les services).

si vous utilisez PostgreSQL pour du pub/sub , mosquitto implémente MQTT qui est un standard dans le monde des IoT.
mosquitto ne consomme que 6Mio de RAM et peut aussi servir de bridge sur un autre serveur MQTT (domoticz, home assistant, jeedom …)

pguyot, 2020-02-16:
PostgreSQL est facilement accusé d'être responsable de la consommation de RAM. Cependant, j'ai tendance à penser qu'il s'agit d'une méconnaissance de PostgreSQL (quel dommage !) et d'une erreur d'interprétation de la mémoire utilisée par les différents processus.

Pour être très clair, PostgreSQL a été choisi parce que c'est la base par défaut pour Django et ça fait le job. De surcroît c'est un logiciel de très bonne qualité et il n'y a aucun doute sur l'intégrité des données dans le contexte de cartes SD et de lapins débranchés sauvagement.

PostgreSQL utilise une zone de mémoire partagée (shared_buffers) entre les différents processus, ce qui ne se voit pas avec top ou ps, et consomme au final beaucoup moins de RAM que l'on pense. C'est logique vu la taille de la base de données qu'on a ici. "shared_buffers" est effectivement plafonné par défaut à 128 Mo, mais dans les faits, il n'a pas besoin d'autant. Ce qu'on peut faire, c'est modifier la configuration de PostgreSQL pour s'assurer de ne jamais se rapprocher de ces 128 Mo, par exemple plafonner cette zone à 4 ou 8 Mo, voire tenter avec le minimum de 128 Ko.

Pour vérifier la quantité de RAM utilisée par PostgreSQL, si la ligne de commande que je vous propose ne vous convient pas, je vous invite à arrêter les clients (les processus python & gunicorn), puis à mesurer la quantité de RAM disponible, puis à arrêter PostgreSQL et voir combien de RAM disponible en plus vous avez. D'après l'article de blog que je vous ai mis, l'estimatif est assez juste et le delta vient du cache disque du système. Vous verrez que vous êtes beaucoup plus près de 33 Mo que de 175 Mo.

On peut sans doute optimiser, que ce soit l'utilisation de la RAM ou le CPU. C'est une bonne chose, et si vous arrivez à dégager 100 ou 150 Mo de RAM on peut même envisager d'utiliser un modèle de reconnaissance vocale plus performant ;-)

Typiquement, j'ai essayé de remplacer l'ORM de Django par SQLAlchemy, vu que le chargement de Django par les services est responsable d'une bonne partie du temps de démarrage du lapin. Mais c'était juste plus lent et la piste a été abandonnée.

On peut aussi envisager de mettre SQLite à la place de PostgreSQL, mais ce n'est pas si simple, parce que les données sont lues et écrites à la fois par le site web (django, gunicorn) et par les services. Il faudrait donc changer significativement l'architecture et je crains que vous ne gagniez pas grand chose au niveau de la RAM.

La plus grosse utilisation de RAM aujourd'hui est pour le modèle de reconnaissance vocale. Il y a eu un gros progrès de ce côté avec la release 0.7.2. N'hésitez pas à vous plonger dans kaldi pour aller plus loin, sachez cependant que c'est un projet qui n'est pas du tout conçu pour l'informatique embarquée. Vient ensuite le modèle de NLU, mais là pour le coup c'est du code conçu spécifiquement pour des systèmes embarqués, il n'y a peut-être pas grand chose à gagner.

Il me semble que la plus grosse utilisation inutile de RAM soit liée à la nécessité d'avoir des processus Python séparés (en gros 20 Mo par service). Du coup, on duplique l'interpréteur et les modules partagés. C'est cependant nécessaire pour permettre le développement de services tiers, les isoler (en utilisant systemd comme superviseur) et garantir une réactivité du système dans le contexte du global interpreter lock. On peut envisager d'intégrer nabd et les services "core" dans un seul processus Python, mais c'est risqué en termes de stabilité. On peut aussi envisager de porter tout ça dans un langage concurrent et robuste (tolérant aux pannes), mais du coup on limite sérieusement l'attrait du projet pour les développeurs, Python reste plus facile d'accès.

Enfin, comme vous pourrez le voir dans le script de mise à jour, c'est plus sage d'arrêter nabd et donc les différents services avant de faire des mises à jour. Vous aurez besoin des presque 200 Mo de RAM actuellement utilisés par la reconnaissance vocale. Et pour compiler certains programmes sur le Pi Zero, il vous faut de toute façon de la swap en plus, jusqu'à 1 Go pour compiler les bibliothèques de NLU.

nathanaelle, 2020-02-17:
Vous m'invitez à faire de la découpe sauvage dans la consommation de ressources ?

J'ai quelques onglets déjà ouvert depuis plusieurs jours pour voir comment faire un docker qui construit une image pour raspberry pi et qui puisse faire tourner cette image via qemu.
Après, j'envisage de voir comment simuler du hardware via qemu et/ou via le Device Tree.
Ensuite, j'envisage de voir comment gérer le netboot du pi (en gros, reboot du lapin = mise à jour systeme du lapin).
Cela permet de simuler en local via docker un lapin puis de mettre à jour à distance un lapin de dev.

L'autre possibilité (qui est complémentaire) est d'avoir des tests comportementaux des services afin de les valider loin du lapin (encore une fois, docker est un bon outil … meme sur des images linux/amd64).

Dès qu'il y a un environnement de tests, n'importe qui pourra développer.
Yakafokon, vous avez votre travail et votre planning et j'ai les mien.

En l'état, à l'aveugle, cela nécessite de comprendre ce que fait le code actuel pour réimplémenter … et cette analyse du code actuel pour une personne débarquant sur le projet est un possible facteur bloquant (en tout cas à titre personnel, cela induit un facteur temps dont je ne dispose pas).

A titre personnel, ne voyant toujours pas l'utilité de PostgreSQL dans l'histoire, je serai bien tenté par faire des binaires go par services avec une base embarquée par service (si nécessaire) et faire dialoguer le tout soit par un pub/sub ad-hoc (par exemple HTTP POST pour un ordre émis par un service au hub, puis HTTP GET SSE pour le broadcast) soit par du MQTT.

De toute façon, au final, c'est la personne qui livre le code à la fin qui a raison … donc tant que je ne fourni aucun code, j'ai bien conscience  que tout ce que je peux dire n'aura qu'une valeur très relative 😐