nabaztag2018/pynab #116: Add possibility to play a streamed mp3 / remote mp3 file.

kymodoke, 2020-02-20:
`{"type":"command","sequence":{"audio":"path/to/chanson.mp3"},"cancelable":true}` sequence > audio only allows to play a local mp3 files.

Would it be possible to add the possibility or support of a stream object to be able to play a remote mp3 file ?

For instance a call like that:
`{"type":"command","sequence":{"stream":"http://rf.proxycast.org/ede95900-a814-418b-b355-fe60e95dfbb6/15309-20.02.2020-ITEMA_22287850-0.mp3"},"cancelable":true}`

Would play the 2 minutes long "Agenda loisirs France Bleu Champagne-Ardenne" for the day 2020-02-20.

mpg123 is able to play mp3 streams (http only, unfortunately not https...). It would open the doors for many possibilities, remotes sound services, including podcasts *(most podcasting platforms do still use http urls for compatibly)*.

pguyot, 2020-02-26:
This might create a security risk. A pertinent use case, though, would be for radio (streaming "live" sounds). We do not really need to offload this to mpg123 and could implement it in python.

mehdichaouch, 2021-04-27:
@kymodoke did you investigate deeper to play sound?

kymodoke, 2021-05-16:
@mehdichaouch , no I didn't investigate any further, for several reasons.

1. Pyhton and Django framework is not my cup of tea, so the learning curve is a bit too long for me, for just that feature.
2. I had hardware troubles with the two tagtagtag boards I've ordered. Audio amplifier is deficient and makes the board crash electrically when playing any sound. So my tagtagtag boards were put in a drawer for a long time, and the rabiits were pluggued back with original board to alternative server openjabnab.fr _(I've fixed that problem recently by unplugin the internal speaker and using an external speaker on line-out port)_
3. I instead dedicated my time to make a lightweight port of Nabaztag for ESP8266 module. Retrofited into a vintage alarm clock https://imgur.com/a/tSWGP2E So no ear nor LED, but it display time, says the time, says surprises/moods, wake up/sleep and has user defined alarms with added sounds.

With openjabnab.fr I was used to many more features than the original Violet features, like alarms, voice memo, radio and podcasts playing. That's why my original request for playing external sounds.
I think this things will append with a modular architecture and the use of plugins.

PR #380 (2022-06-12): Stream MP3s over HTTP & HTTPS
Fix #116
closed 2022-06-12 by nabaztag2018