nabaztag2018/pynab #117: RFID PPIC non géré

Smaky07, 2020-02-21:
J'ai pas mal de tag RFID reconnus mais inutilisables. Voici un UID posant problème : d0:02:0c:c1:1e:cb:0d:02

Si je peux donner plus d'information (pas ce week end mais la semaine prochaine), je suis disponible.

En pièce jointe une photo de ce que à quoi ils ressemble.

Ils fonctionnais très bien avec le nabaztag original et permettait de lancer des MP3s.

Je n'ai pas encore eu le temps de vérifier si il était possible d'écrire dedans.

[image]

pguyot, 2020-02-21:
Il semble que ce soit un tag SRIX4K de Motorola si j'ai bien décodé le UID.
On a la datasheet, je peux ajouter le support et vous laisser essayer ?

Smaky07, 2020-02-21:
Je pourrais tester à partir de dimanche soir pas soucis. Je suis sur la branche Master.

closed 2020-02-25 by pguyot

Smaky07, 2020-02-26:
C'est bon pour moi. mes 10 tags SRIX4 sont OK.

pguyot, 2020-02-26:
Merci.

commit 50b3f46e (2020-02-25): Add support for more tags (fix #117)