nabaztag2018/pynab #118: RFID to URL

kaleofr, 2020-02-24:
Bonjour,
pour mettre en place l'appel d'url par les tags rfid, j'ai commencé quelques bouts de code (dans rfid_dev.py) 

D'abord un
import requests

Puis dans la fonction __init__  j'ai ajouté:

        self.__rfid_2_url_mode = 2
        self.__rfid_2_url_url = 'https://XXXXXXXX/core/api/jeeApi.php?apikey=XXXXXX&type=scenario&id=XX&action=start&tags=rfid="#RFID_TAG#"%20etat="#RFID_STATE#"%20flags="#RFID_FLAGS#"%20app="#RFID_APP#"'

Un peu plus loin j'ai créé la fonction _send_rfid_2_url:

def _send_rfid_2_url(self, flags):
        if self.__rfid_2_url_mode==0 : return # Never send url
        if (self.__rfid_2_url_mode==1 and ((flags & TagFlags.FORMATTED)==2) and (self.__current_app != 255)) : return # Send only unknown tags
        str_uid = self.__current_uid.hex()
        url_message = self.__rfid_2_url_url.replace("#RFID_TAG#",str_uid)
        url_message = url_message.replace("#RFID_STATE#",self.__state.name)
        url_message = url_message.replace("#RFID_FLAGS#",str(flags))
        url_message = url_message.replace("#RFID_APP#",str(self.__current_app))
        f = requests.get(url_message)

Enfin, un appel à cette fonction en dernière ligne de _process_read_blocks:
self._send_rfid_2_url(flags)

Cela fonctionne tel que je le souhaite, mais n'est pas très user-friendly puisque mon url et le choix du mode de fonctionnement se fait dans le code.

J'ai ajouté quelques lignes d'html dans l'index.html de rfid:
				<p>{% trans "Tags can be transmitted to a remote server by URL." %}</p>
				<p>{% trans "Setup below the mode that suits for you (always send the tag, only when it is unknown, never send the tag)" %}</p>
				<p>{% trans "Setup also the url of your server. You have to include #RFID_TAG# where you want your Rfid to be placed. You can also use #RFID_STATE#, #RFID_FLAGS#, #RFID_APP# to know about repetition, tag flags and app called by the tag" %}</p>
				
				<div class="form-group row">				
					<label for="rfid_2_url" class="col-6 col-form-label">{% trans "URL Send Option" %}</label>
                    <div class="col-6">
                      <select id="rfid_2_url_modeSel" name="rfid_2_url_mode">
                        <option value="0">{% trans "Never send the tag" %}</option>
                        <option value="1">{% trans "Only for unknown tag" %}</option>
                        <option value="2">{% trans "Always send the tag" %}</option>
                      </select>
                    </div>
				</div>
				<div class="input-group input-group mb-3">
					<div class="input-group-prepend">
						<span class="input-group-text">{% trans "URL of your server:" %}</span>
					</div>
					
					<input name="rfid_2_url_url" type="text" class="form-control" id="rfid_2_url_url" size="150" value="Ex: https://MY_SERVER/core/api/jeeApi.php?apikey=MY_API_KEY&type=scenario&id=MY_SCENARIO_ID&action=start&tags=rfid=&#34;#RFID_TAG#&#34;%20etat=&#34;#RFID_STATE#&#34;%20flags=#34;#RFID_FLAGS##34;%20app=#34;#RFID_APP##34;'">
					<div class="input-group-append">
						<button type="submit" class="btn btn-primary">{% trans "Save" %}</button>
					</div>
					</div>
			 </span>
 
J'ai commencé le python la semaine dernière et django vendredi soir. Bref je ne sais pas trop comment passer ces 2 valeurs à rfid_dev.py ( __rfid_2_url_mode et __rfid_2_url_url), ni comment faire les traductions correctement (mes modifications de django.po ne semblent pas suffir).

Donc, si quelqu'un de plus performant que moi en la matière peut termier ca ou m'indiquer les étapes qu'il me manque, je suis preneur :-)

Kaléo.

pguyot, 2020-02-24:
Il n'est pas vraiment prévu de s'interfacer à cet endroit de rfid_dev.py qui est l'implémentation de l'interface rfid via le device cr14. On pourrait imaginer un autre lecteur RFID avec une autre interface… Le risque avec cette approche est qu'à chaque mise à jour de pynab il faudra revoir votre code.

Si vous souhaitez que les deux valeurs "mode" et "url" puissent être changées par l'interface, il vous faut les stocker quelque part depuis la vue django (view.py) et relire depuis rfid_dev.py.

Cependant, ce qui serait plus dans l'esprit, ce serait un service ad hoc qui se connecterait sur le protocole nabd et qui écouterait les événements RFID. Aggelos sur le forum semble aller dans cette direction :
https://www.tagtagtag.fr/forum/showthread.php?tid=138

Dans cette hypothèse, vous avez le choix en termes d'usage entre :
- écrire le tag avec des informations spécifiques que vous pouvez utiliser derrière, et dans ce cas il vous faut ajouter un id dans rfid.py pour votre application et prévoir une interface pour configurer le tag, vous pouvez partir de nabweatherd ou de nabsurprised ;
- écouter tous les événements rfid en envoyant le paquet idle comme décrit sur le forum, et dans ce cas vous traiterez les uid des tags. Vous pouvez filtrer sur les tags déjà formattés selon votre mode 1 en regardant les slots "supported" et "app" de l'événement.

Enfin, pour votre questions sur la traduction dans les fichiers po, il faut appeler la commande `django-admin compilemessages` (typiquement : `cd /home/pi/pynab/service; ../venv/bin/django-admin compilemessages`) pour générer le fichier mo, puis redémarrer le serveur web pour qu'il récupère la traduction (`sudo service nabweb restart`).

closed 2020-02-24 by pguyot

kaleofr, 2020-02-25:
Ok merci.
Même si pour ma part je ne voyais pas ca comme un service mais une focntion optionnelle ajoutée pour tout le monde... d'ou l'écriture tel que le l'ai faite.
Je vais donc travailler à la compréhension de django, des services et aller dans ton sens.

Cordialement.
Kaléo.

kaleofr, 2020-02-25:
Bonsoir,
En plus d'apprendre le python, django, j'intègre maintenant github. Ca fait beaucoup en quelques jours.
En tous cas, mon service avance, j'ai une page de confirguration qui apparait, la traduction fonctionne, les valeurs de configuration sont bien sauvegardées mais... le code ne fonctionne pas.

Je n'ai malheureusement aucun moyen de comprendre ce qu'il se passe.

-> Y-a-t-il un moyen de débuggé/espionner/logger/ se que se passe dans mon module principal en ssh? (nabrfid2server.py)
-> A quoi les fonctions de tests des services (chatouille, aujourd'hui/demain(météo)) font-elle appel? Comment celà est-il récupéré dans le code python?

Désolé que vous solliciter ainsi mais je ne vois pas trop comment avancer maintenant.

Merci pour votre attention.

Mon code est disponible sur https://github.com/kaleofr/pynab/tree/master/nabrfid2server
https://github.com/kaleofr/pynab/blob/master/nabd/rfid.py
https://github.com/kaleofr/pynab/blob/master/nabweb/settings.py

pguyot, 2020-02-26:
Quelle piste choisissez-vous ? 
- réagir à tous les tags RFID selon un mode (0, 1, 2)
- ne réagir qu'aux tags RFID configurés pour votre service

# Réagir à tous les tags RFID, selon un mode (0, 1, 2) configurable dans une interface

Dans ce premier cas, vous n'avez pas besoin de cette ligne :

        nabrfid2server.NABAZTAG_RFID_APPLICATION_ID: "rfid/*",  # 10

ni de ces lignes :

     NABAZTAG_RFID_APPLICATION_ID = 10
     NABAZTAG_RFID_APPLICATION_NAME = "Rfid to Server"

À la place, il vous faut envoyer un paquet à nabd pour lui dire de vous envoyer tous les événements. Vous pouvez faire cela en surchargeant la méthode client_loop qui par défaut envoie un masque restreint (on pourrait envisager de faire une classe de base pour éviter le code dupliqué).

    async def client_loop(self):
        try:
            idle_packet = '{"type":"mode","mode":"idle","events":["rfid/*"]}\r\n'
            self.writer.write(idle_packet.encode())
            while self.running and not self.reader.at_eof():
                line = await self.reader.readline()
                if line != b"" and line != b"\r\n":
                    try:
                        packet = json.loads(line.decode("utf8"))
                        logging.debug(f"process nabd packet: {packet}")
                        await self.process_nabd_packet(packet)
                    except json.decoder.JSONDecodeError as e:
                        logging.error(
                            f"Invalid JSON packet from nabd: {line}\n{e}"
                        )
            self.writer.close()
            await self.writer.wait_closed()
        except KeyboardInterrupt:
            pass
        finally:
            if self.running:
                self.loop.stop()

Ensuite, il faut installer et lancer votre service !

    cd /home/pi/pynab/nabrfid2server
    sudo systemctl enable nabrfid2server.service
    sudo systemctl start nabrfid2server

Le log se trouve dans /var/log/daemon.log (les crashs en amont) et /var/log/nabrfid2server.log (les appels au module logging).

Vous pouvez ensuite redémarrer le service avec :

    sudo systemctl restart nabrfid2server

ou

    sudo service nabrfid2server restart

# Réagir aux tags RFID configurés pour votre service

Dans le deuxième cas (votre service ne fonctionne qu'avec les tags configurés pour lui), il vous faut effectivement modifier rfid.py pour qu'il décode vos applications.

Ces lignes sont très bien :

    NABAZTAG_RFID_APPLICATION_ID = 10
    NABAZTAG_RFID_APPLICATION_NAME = "Rfid to Server"

En revanche, ici, il faut mettre :

    nabrfid2server.NABAZTAG_RFID_APPLICATION_ID: "nabrfid2server",  # 10

La valeur ici ("nabrfid2server") est calculée automatiquement par NabService dans client_loop, il vous faut le même nom que votre module. Pas besoin de surcharger la méthode.

En revanche, vous pouvez jeter la logique `mode`, puisqu'elle ne s'applique pas, vous n'aurez que les tags spécifiquement pour votre service.

Il vous faut également une interface de configuration des tags RFID. Pour ce faire, vous avez besoin d'une nouvelle vue, pour /nabrfid2server/rfid-data/. Regardez celle du taichi (si vous ne voulez rien écrire dans les tags sinon votre application).

https://github.com/nabaztag2018/pynab/blob/master/nabtaichid/views.py#L34
https://github.com/nabaztag2018/pynab/blob/master/nabtaichid/templates/nabtaichid/rfid-data.html
https://github.com/nabaztag2018/pynab/blob/master/nabtaichid/urls.py#L6

# Comment fonctionnent les boutons surprise/météo ?

Cela n'a strictement rien à voir :-)

Ces services sont des services récurrents (`NabRecurrentService`) avec une date pour la prochaine performance (aléatoire et éventuellement programmée pour la météo même si cela n'est pas encore codé #18). Lorsque le bouton est cliqué, la date de la prochaine performance est fixée à tout de suite, et le service est réveillé avec un signal unix qui fait qu'il relit la configuration dans la base de données, se rend compte qu'il faut jouer la performance et la joue.

kaleofr, 2020-02-26:
Bonsoir et merci pour votre aide très préciseuse.
Mon service est partiquement opérationnel. 
Mes seuls problèmes maintenant sont:
-> la mise à jours de valeurs de la page settings vers nabrfid2server
     (je ne comprend pas tout à fait le fonctionnement, parfois ca marche... peut être après les reboot seulmement)
-> l'exécussion de mon bouton "Envoyer une url de test" sur la page settings. Surement pour les même raisons.

Enfin, comment partager cela une fois que cela sera terminé? Cela nécessite notamment de modifier nabweb/settings.py, c'est donc nécessairement par vous?

Bonne soirée,

Kaléo.

pguyot, 2020-02-26:
En relisant votre code, j'attire votre attention sur ces lignes :

https://github.com/kaleofr/pynab/blob/master/nabrfid2server/views.py#L28

    NabRfid2server.server_test = True
    NabRfid2server.signal_daemon()

La deuxième ligne appelle la méthode signal_daemon() de la classe NabRfid2server (héritée) qui envoie un signal USR1 au processus (unix) nabrfid2server lancé par systemd.

En revanche, la première ligne modifie la variable de la classe NabRfid2server, mais vous êtes toujours dans le processus du serveur web (gunicorn). Il y a ici (si cela fait sens) deux instances de la classe NabRfid2server : une dans le serveur web et une dans le processus nabrfid2server. Si vous modifiez la classe d'un côté, ça ne se verra pas de l'autre !

Du coup, ce test :

https://github.com/kaleofr/pynab/blob/master/nabrfid2server/nabrfid2server.py#L21

    if self.server_test : send_rfid_2_url("RFID_UID_TEST","EVENT_TEST","APP_TEST","SUPPORT_TEST")

sera toujours faux.

Si vous voulez communiquer du site web vers le processus du service, vous pouvez faire comme pour les autres services : un signal USR1 du serveur web vers le processus unix qui dit juste qu'il faut recharger la configuration (et du coup la fonction reload_config est appelée), et auparavant, une écriture dans la base de données. Mais ça veut dire stocker que vous voulez le test dans la base de données.

Concernant l'intégration au projet, il y a un ticket à ce sujet (#122). L'idée serait de permettre de le faire sans modifier le code (settings.py), sauf peut-être pour les IDs rfid puisque dans ce cas il faut une liste centralisée. Les utilisateurs pourraient ajouter des services via un repository Git. Je mets ça assez haut sur la liste pour accompagner votre service !

kaleofr, 2020-02-27:
Bonsoir,
J'ai résolu le problème du bouton de test, ajouté la possibilité d'envoyer le packet rfid_event complet, amélioré la lisibilité de l'interface etc.

Il ne me reste qu'un problème: ma fonction [reload_config](nabrfid2server.py) recoie bien les mises à jour de l'interface web, le [log](https://github.com/kaleofr/pynab/blob/master/nabrfid2server/nabrfid2server.py#L26) le confirme.

Mais ces valeurs  ne sont pas concervées pour les autres fonctions du service.  Du coup "enregistrer" les modifications ne marche pas, le test envoi à l'ancienne url...

Tout est réenregistré correctement après un restart du service.

Un dernier petit coup de main?

Bonne soirée,

Kaléo.

pguyot, 2020-02-27:
J’ai l’impression que la méthode `__config` n’est jamais appelée.

`__init__` stocke la configuration dans `self.config` et `process_nabd_packet` utilise `self.config` mais `reload_config` ne touche pas à `self.config` mais utilise une variable locale `config` qui est perdue.

https://github.com/kaleofr/pynab/blob/master/nabrfid2server/nabrfid2server.py#L24

kaleofr, 2020-02-27:
Merci!
J'avais effectivement compris cela comme ca mais je cherchais certainement trop compliqué.
-> __config supprimé
-> config remplacé par self.config dans la fonction reload_config

-> tout semble ok!
Maintenant, je vous laisse me dire pour la suite, nous pouvons attendre la résolution du ticket [#122](https://github.com/nabaztag2018/pynab/issues/122).

Encore merci pour votre aide, je vais laisser le lapin garder la maison la semaine prochain... il n'aime pas la neige!

Kaléo.