nabaztag2018/pynab #123: Sound with a lots of noises in interactive books with default voice.

supermomo31, 2020-02-27:
It sounds really good with the first book and the nabaztag voice.
It sounds horrible with a lot of skratches with the second book and default voice.

pguyot, 2020-02-27:
I guess this is a known issue with the sound driver and you need to upgrade it. I guess you are on master branch. Did you upgrade through the web page?

supermomo31, 2020-02-27:
yes

pguyot, 2020-02-27:
Could you please perform the following:

cd /home/pi/wm8960
git fetch
git status

supermomo31, 2020-02-27:
pi@nabaztag:~/wm8960 $ git status
On branch tagtagtag-sound
Your branch is up to date with 'origin/tagtagtag-sound'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        .tagtagtag-sound.dtbo.d.dtc.tmp
        .tagtagtag-sound.dtbo.d.pre.tmp
        .tagtagtag-sound.dtbo.dts.tmp

nothing added to commit but untracked files present (use "git add" to track)

pguyot, 2020-02-27:
Let's try to find out if latest module is installed.

    ls -l /lib/modules/*/kernel/sound/soc/codecs/snd-soc-wm8960.ko
    ls -l /home/pi/wm8960/wm8960.c
    ls -l /home/pi/wm8960/snd-soc-wm8960.ko
    md5sum /home/pi/wm8960/wm8960.c

supermomo31, 2020-02-27:
pi@nabaztag:~/wm8960 $ ls -l /lib/modules/*/kernel/sound/soc/codecs/snd-soc-wm8960.ko
-rw-r--r-- 1 root root 42956 Feb  4 16:06 /lib/modules/4.19.93+/kernel/sound/soc/codecs/snd-soc-wm8960.ko
pi@nabaztag:~/wm8960 $ ls -l /home/pi/wm8960/wm8960.c
-rw-r--r-- 1 pi pi 41506 Feb 19 18:00 /home/pi/wm8960/wm8960.c
pi@nabaztag:~/wm8960 $ ls -l /home/pi/wm8960/snd-soc-wm8960.ko
-rw-r--r-- 1 root root 42956 Feb  4 16:06 /home/pi/wm8960/snd-soc-wm8960.ko
pi@nabaztag:~/wm8960 $ md5sum /home/pi/wm8960/wm8960.c
3a7b79a11b309a01f7712b9cd571a4f0  /home/pi/wm8960/wm8960.c

supermomo31, 2020-02-27:
`pi@nabaztag:~/wm8960 $ ls -l /lib/modules/*/kernel/sound/soc/codecs/snd-soc-wm8960.ko
-rw-r--r-- 1 root root 42956 Feb  4 16:06 /lib/modules/4.19.93+/kernel/sound/soc/codecs/snd-soc-wm8960.ko
pi@nabaztag:~/wm8960 $ ls -l /home/pi/wm8960/wm8960.c
-rw-r--r-- 1 pi pi 41506 Feb 19 18:00 /home/pi/wm8960/wm8960.c
pi@nabaztag:~/wm8960 $ ls -l /home/pi/wm8960/snd-soc-wm8960.ko
-rw-r--r-- 1 root root 42956 Feb  4 16:06 /home/pi/wm8960/snd-soc-wm8960.ko
pi@nabaztag:~/wm8960 $ md5sum /home/pi/wm8960/wm8960.c
3a7b79a11b309a01f7712b9cd571a4f0  /home/pi/wm8960/wm8960.c
`

pguyot, 2020-02-27:
For some weird reason, the driver has not been updated.
In your specific case:

    cd /home/pi/wm8960
    make
    sudo make install
    sudo reboot

should fix the problem.

However, I am wondering why the code was updated but the driver was not updated.

closed 2020-02-27 by pguyot