nabaztag2018/pynab #128: Surprised: the frequency scale isn't natural

rbarrois, 2020-03-08:
Currently, the available settings translate, based on the `frequency` formula, to:
- Sometimes: every 103 to 513 minutes (1.7 to 8.5 hours), or 0.2 times per hour
- Often: every 65 to 326 minutes (1.1 to 5 hours), or 0.3 times per hour
- Very often: every 3 to 15 minutes, or 6.66 times per hour

I feel that there should be a regular progression between "sometimes" and "very often"; being based on a human perception, logarithmic scales should work fine:
```python
>>> P_max = 308; P_min = 9
>>> P_max = P_mid * ratio; P_mid = P_min * ratio
>>> ln(P_max) = ln(P_mid) + ln(ratio); ln(P_mid) = ln(P_min) + ln(ratio)
>>> P_mid = exp((ln(P_max) + ln(P_min)) / 2)
>>> P_mid == 52.65
```

Brought back to the formula (`delta = (256 - f_mid) * (127 + 64) / 128`), this would provide: `(256 - f_mid) * 191 / 128 = 52.65` ⇒ `f_mid = 221`; which brings the available intervals to `(6, 35, 206)`.

Putting it another way, I suggest switching "Often" to 221, with an explanation of this computation logic.

What do you think? Would you be interested in a pull-request in that direction?

rbarrois, 2020-03-08:
For quick reference, the relevant code is here: https://github.com/nabaztag2018/pynab/blob/master/nabsurprised/nabsurprised.py#L54-L55

pguyot, 2020-03-09:
Very often is indeed quite too much compared to other values and mostly useful when discovering the rabbit or testing.

I believe the computation follows the original code, at least for tai chi:

https://github.com/rngtng/NabaztagHackKit/blob/master/bytecode/_original/nominal-ping.mtl#L5468

rbarrois, 2020-03-09:
Maybe for taichi; however, I haven't been able to find the code for "surprised" in the source you've shared. Was it managed server-side?

pguyot, 2020-03-11:
Indeed, surprise was handled server side.
You can find the source code of (Mindscape's fork of) Violet source code.

From what I read, it seems that very often meant in rand(0, 1200) seconds, which is every 10 minutes on average.

https://github.com/nguillaumin/nabaztag-server/blob/master/net.violet.platform/src/main/java/net/violet/platform/schedulers/FrequencyHandler.java#L29
https://github.com/nguillaumin/nabaztag-server/blob/master/net.violet.platform/src/main/java/net/violet/platform/daemons/schedulers/RandomWithFrequencyScheduler.java#L26
https://github.com/nguillaumin/nabaztag-server/blob/master/net.violet.platform/src/main/java/net/violet/platform/daemons/schedulers/AbstractFrequencyScheduler.java#L112

So drawing from this analysis, surprise does not follow the original Nabaztag's frequency (since it uses taichi frequency computation which was different). And we would definitely accept a pull request that would implement very often/often/sometimes and rarely following the old algorithm.

PR #295 (2021-08-18): Change Surprise frequency computation, to match original Nabaztag scale
Resolves #128.
`nabsurprised` frequency computation changed to try and match original Violet frequency scale (see https://github.com/nabaztag2018/pynab/issues/128#issuecomment-597740918).
closed 2021-08-18 by nabaztag2018