nabaztag2018/pynab #135: Interactive mode doesn't send any event

Ellerbach, 2020-04-28:
Hi,
When my application is interactive mode, it does not receive any of the event. The nabd service doesn't send anything while it's supposed to send all the events. Version used is 0.7.4.

Steps to repro:

- send: ```{"type":"mode","request_id":"uniqueid","mode":"interactive","events":["asr","button","ears"]}```
- nothing at all is sent when pressing the button, turning the ears (asr is normal as the button is not handled anymore by the service)
```
Your Nabaztag is in interactive mode and will receive all events
Nabaztag status changed, new status is Interactive
Reset all events
```
- send:  ```{"type":"mode","request_id":"uniqueid","mode":"idle","events":["asr","button","ears"]}```
- you will receive all the events as expected
Same code to handle events, no IP traffic coming to the service in the "interactive" mode. 
```
Nabaztag status changed, new status is Idle
Your Nabaztag is in idle mode
Your Nabaztag is in interactive mode and will receive all events
New button event: Down, Time: 1588058768.7107267
New button event: Up, Time: 1588058768.9703965
New button event: Click, Time: 1588058769.1285439
New button event: Down, Time: 1588058769.7709365
New button event: Up, Time: 1588058769.9808636
New button event: Click, Time: 1588058770.1390917
New button event: Down, Time: 1588058770.844595
New button event: Up, Time: 1588058771.0850823
New button event: Click, Time: 1588058771.2434516
Reset all events
```

Ellerbach, 2020-04-28:
After switching manually to the master branch, it seems to work normally now.

pguyot, 2020-06-29:
Can we close this with 0.8.0?

Ellerbach, 2020-06-30:
> Can we close this with 0.8.0?

Sure

closed 2020-06-30 by Ellerbach