nabaztag2018/pynab #138: Add web UI option to enable & disable SSH (and set password?)

srcshelton, 2020-06-13:
Right now, the only way to enable SSH it to unscrew the Nabaztag shell with the special tri-pointed screwdriver, remove the microSD card from the PiZero, put it in an adapter and insert it into another machine capable of reading ext4 filesystems... and then figure out the magic needed to get systemd to enable a service on a different architecture.

It would be _so_ much easier if there were a UI option to (en|dis)able SSH without needing physical access to the bunny.

(N.B. The magic needed is documented here: https://unix.stackexchange.com/questions/157816/what-does-systemctl-enable-netctl-service-do ☺️)

ViennaMike, 2020-06-19:
I agree a way to do this from the web interface would be nice. And at a bare minimum, PLEASE TELL PEOPLE THIS IN THE INSTRUCTIONS BEFORE PUTTING EVERYTHING BACK TOGETHER AND THEN GETTING TO SOFTWARE DETAILS AND LEARNING THIS.

232enero, 2020-06-24:
Hello @ViennaMike GOOD IDEA! I JUST MODIFIED THE TUTORIAL!

closed 2020-06-24 by 232enero