nabaztag2018/pynab #141: Advertise rabbit on local network

f-laurens, 2020-06-24:
Suggestion: advertise rabbit interface on local network through Bonjour by installing a file **/etc/avahi/services/pynab.service** with contents similar to:
```
<?xml version="1.0" standalone='no'?><!--*-nxml-*-->
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<!-- See avahi.service(5) for more information about this configuration file -->

<service-group>
    <name replace-wildcards="yes">Nabaztag rabbit (%h)</name>
    <service>
        <type>_http._tcp</type>
        <port>80</port>
    </service>
</service-group>
```

232enero, 2020-06-24:
Thanks again!
I'll put that in the install.sh file too.

f-laurens, 2020-06-24:
And you might as well throw in NabBlocky too...: **/etc/avahi/services/nabblocky.service**  :
```
<?xml version="1.0" standalone='no'?><!--*-nxml-*-->
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<!-- See avahi.service(5) for more information about this configuration file -->

<service-group>
  <name replace-wildcards="yes">NabBlockly (%h)</name>
  <service>
    <type>_http._tcp</type>
    <port>8080</port>
  </service>
</service-group>
```

f-laurens, 2020-06-24:
A variant of **/etc/avahi/services/pynab.service** to make network administrators happy ;-) (end users won't care):
```
<?xml version="1.0" standalone='no'?><!--*-nxml-*-->
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<!-- See avahi.service(5) for more information about this configuration file -->

<service-group>
  <name replace-wildcards="yes">Nabaztag rabbit (%h)</name>
  <service>
    <type>_http._tcp</type>
    <port>80</port>
    <txt-record>vendor=enero</txt-record>
    <txt-record>model=tag:tag:tag v2.0</txt-record>
  </service>
</service-group>
```
Adapt to your liking...

closed 2020-06-24 by f-laurens