nabaztag2018/pynab #142: Autopromo sounds for non French-speaking rabbits?

f-laurens, 2020-06-25:
Unless they speak French, rabbits are desperately mute when asked what they can do...
And indeed:
```
pi@Nabaztag:~ $ find /home/pi/pynab/ -name '*utopromo*'
/home/pi/pynab/nabsurprised/sounds/fr_FR/nabsurprised/autopromo
/home/pi/pynab/nabsurprised/sounds/fr_FR/nabsurprised/autopromo/Autopromo4.mp3
/home/pi/pynab/nabsurprised/sounds/fr_FR/nabsurprised/autopromo/Autopromo5.mp3
/home/pi/pynab/nabsurprised/sounds/fr_FR/nabsurprised/autopromo/Autopromo2.mp3
/home/pi/pynab/nabsurprised/sounds/fr_FR/nabsurprised/autopromo/Autopromo1.mp3
/home/pi/pynab/nabsurprised/sounds/fr_FR/nabsurprised/autopromo/Autopromo3.mp3
```
Is it possible to add the missing sounds (if they ever existed)?

pguyot, 2020-06-29:
How do you ask your rabbit what it can do?
These sounds only exist in French, they were actually created from original sounds that served another purpose. Similar sounds could be created, but the autopromo intent only exists in French indeed.

f-laurens, 2020-06-29:
_> How do you ask your rabbit what it can do?_
Eh Eh... By showing it a RFID tag configured for 'Que sais-tu faire' on a French-speaking rabbit ;-)
So indeed this a moot point: non-French speakers will not miss a feature they don't know about...

I guess this closes the issue.

closed 2020-06-29 by pguyot