nabaztag2018/pynab #146: Cleanup macOS leftovers from Pynab image

f-laurens, 2020-06-29:
it would be nice to add a phase at the end of the Pynab ```create-image``` script to do some clean up of macOS leftovers from the image:
- remove ```._*``` files
- change ownership to```root:root```for files belonging to ```501:staff```

pguyot, 2020-06-29:
Confused about the issue. Are you talking about released images or images you build yourself on your mac?

f-laurens, 2020-06-29:
This is about released images.
I noticed this on the 0.7.4 image (but forgot about it...), and noticed it again on the 0.8.1 image.
The ```._*```files are mainly in the ```/opt/kaldi``` directory tree.
The files belonging to ```root``` in ```/home/pi``` are artefacts from builds of the drivers.

pguyot, 2021-05-02:
> Also maybe remove /usr/bin/qemu-arm-static*x86 binaries from image.

Fixed in #193 
Also the image being created on ubuntu, it should have no macOS' .DS_Store or resource forks.

f-laurens, 2021-05-23:
As of the current  (9.0.0 pre-release) image build process , there are still: some macOS leftovers in the images built on GitHub, which apparently come from the build of _Kaldi_:
- a number of `._*` files:
`/._etc /etc/ld.so.conf.d/._kaldi.conf /etc/._ld.so.conf.d`
`/usr/._lib /usr/lib/._pkgconfig /usr/lib/pkgconfig/._kaldi-asr.pc /._usr`
`/._opt`
plus files in `/opt/kaldi/`
- files belonging to `501:staff` whereas they should belong to `root:root` same files as above plus others, mainly in `/opt/kaldi`

PR #269 (2021-06-08): Cleanup macOS leftovers from Pynab image
Resolves #146.
Cleanup macOS leftovers at the end of `setup` script.
closed 2021-06-08 by omevel