nabaztag2018/pynab #148: nabd crashes regularly trying to load libmpg123

fcharlier, 2020-07-02:
Hello, issue similar to #133 happened again today on v0.8.1 (happens every 2~3 days)

Restarting `nabd.service` is sufficient to make it work again.

```
pi@nabaztag:~/pynab $ git show
commit c076e3434595f52862ba458581035fcb0fb6c069 (HEAD -> release, tag: v0.8.1, origin/releng, origin/release)
Merge: 2b429bf f1b7607
Author: Olivier Mevel <olivier@enero.fr>
Date:   Thu Jun 25 21:25:48 2020 +0200

    Merge branch 'master' into releng
```

```
Linux nabaztag 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Jun 30 12:29:47 2020 from 2a01:cb00:312:cf00:e2cb:4eff:fe01:68fd
pi@nabaztag:~ $ tail -n 100 /var/log/nabd.log
DEBUG 2020-07-01 16:04:58,306 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:04:58,330 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '0000ff', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}]}}DEBUG 2020-07-01 16:10:04,656 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:10:04,678 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '0000ff', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}]}}DEBUG 2020-07-01 16:11:59,509 packet from service: {'type': 'message', 'signature': {'audio': ['nabsurprised/respirations/*.mp3']}, 'body': [{'audio': ['nabsurprised/07-01/*.mp3;nabsurprised/*.mp3']}], 'expiration': '2020-07-01T14:12:59.370887+00:00'}
DEBUG 2020-07-01 16:14:49,790 packet from service: {'type': 'command', 'sequence': [{'choreography': 'nabtaichid/taichi.chor'}], 'expiration': '2020-07-01T14:15:49.708296+00:00'}
DEBUG 2020-07-01 16:15:11,429 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:15:11,451 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '0000ff', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}]}}DEBUG 2020-07-01 16:20:17,827 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:20:17,849 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '0000ff', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}]}}DEBUG 2020-07-01 16:21:27,159 packet from service: {'type': 'message', 'signature': {'audio': ['nabsurprised/respirations/*.mp3']}, 'body': [{'audio': ['nabsurprised/07-01/*.mp3;nabsurprised/*.mp3']}], 'expiration': '2020-07-01T14:22:27.017921+00:00'}
DEBUG 2020-07-01 16:22:00,266 packet from service: {'type': 'command', 'sequence': [{'choreography': 'nabtaichid/taichi.chor'}], 'expiration': '2020-07-01T14:23:00.197214+00:00'}
DEBUG 2020-07-01 16:25:24,188 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:25:24,211 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '0000ff', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}]}}DEBUG 2020-07-01 16:28:11,140 packet from service: {'type': 'message', 'signature': {'audio': ['nabsurprised/respirations/*.mp3']}, 'body': [{'audio': ['nabsurprised/07-01/*.mp3;nabsurprised/*.mp3']}], 'expiration': '2020-07-01T14:29:11.066853+00:00'}
DEBUG 2020-07-01 16:30:30,588 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:30:30,598 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '0000ff', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}]}}DEBUG 2020-07-01 16:33:38,850 packet from service: {'type': 'command', 'sequence': [{'choreography': 'nabtaichid/taichi.chor'}], 'expiration': '2020-07-01T14:34:38.756862+00:00'}
DEBUG 2020-07-01 16:34:02,669 packet from service: {'type': 'message', 'signature': {'audio': ['nabsurprised/respirations/*.mp3']}, 'body': [{'audio': ['nabsurprised/07-01/*.mp3;nabsurprised/*.mp3']}], 'expiration': '2020-07-01T14:35:02.612394+00:00'}
DEBUG 2020-07-01 16:35:36,902 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:35:36,914 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '0000ff', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}, {'left': '000000', 'center': '0000ff', 'right': '000000'}, {'left': '000000', 'center': '000000', 'right': '0000ff'}]}}DEBUG 2020-07-01 16:36:22,010 packet from service: {'type': 'info', 'info_id': 'nabairqualityd', 'animation': {'tempo': 42, 'colors': [{'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '000000', 'center': '000000', 'right': '000000'}]}}
DEBUG 2020-07-01 16:40:43,733 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2020-07-01 16:40:43,742 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 20, 'colors': [{'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center [... cut ...]
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found

pi@nabaztag:~ $
```

cyril1929, 2020-07-02:
I have exactly the same problem.

sanderant, 2020-07-29:
Is it possible installing the libmpg123 might solve this problem?

cyril1929, 2020-07-29:
> Is it possible installing the libmpg123 might solve this problem?

I tried, but it doesn't resolve the problem.

fcharlier, 2020-08-06:
> Is it possible installing the libmpg123 might solve this problem?

libmpg123 is installed, the library load failures appear after a few days of uptime bad this piece of code isolated works after a failure. Restarting the nabd service is sufficient to make it work again.

mehdichaouch, 2021-04-27:
Do you have this problem with the latest version [v0.8.4](https://github.com/nabaztag2018/pynab/releases/tag/v0.8.4)?

fcharlier, 2021-04-27:
@mehdichaouch Hi,

I need to verify, I had the Nabaztag reboot every morning to try to work this issue around.
I have just disabled it and will try to update this issue in a few days according to the results.

fcharlier, 2021-04-29:
So I confirm it still happens with the current version (here after ~36h uptime).

Log from nabd:
```
DEBUG 2021-04-28 15:51:29,749 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2021-04-28 15:51:29,773 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 125, 'colors': [{'left': '000000', 'center': 'ffff00', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}]}}
DEBUG 2021-04-28 15:53:58,715 packet from service: {'type': 'message', 'signature': {'audio': ['nabsurprised/respirations/*.mp3']}, 'body': [{'audio': ['nabsurprised/04-28/*.mp3;nabsurprised/*.mp3']}], 'expiration': '2021-04-28T13:54:58.577718+00:00'}
DEBUG 2021-04-28 15:56:30,189 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2021-04-28 15:56:30,212 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 125, 'colors': [{'left': '000000', 'center': 'ffff00', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}]}}
DEBUG 2021-04-28 15:57:46,742 packet from service: {'type': 'info', 'info_id': 'nabairqualityd', 'animation': {'tempo': 14, 'colors': [{'left': '000000', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '000000'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '000000', 'right': '00ffff'}, {'left': '000000', 'center': '000000', 'right': '00ffff'}, {'left': '000000', 'center': '000000', 'right': '000000'}, {'left': '000000', 'center': '00ffff', 'right': '000000'}, {'left': '000000', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '00ffff'}, {'left': '00ffff', 'center': '00ffff', 'right': '000000'}, {'left': '00ffff', 'center': '000000', 'right': '00ffff'}, {'left': '000000', 'center': '00ffff', 'right': '00ffff'}]}}
DEBUG 2021-04-28 16:00:00,918 packet from service: {'type': 'message', 'signature': {'audio': ['nabclockd/signature.mp3']}, 'body': [{'audio': ['nabclockd/16/*.mp3']}], 'expiration': '2021-04-28T16:03:00.914981'}
ERROR 2021-04-28 16:00:02,163 Future exception was never retrieved
future: <Future finished exception=LibInitializationException('libmpg123 not found')>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 287, in <lambda>
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found
DEBUG 2021-04-28 16:01:30,620 packet from service: {'type': 'info', 'info_id': 'nabweatherd_rain'}
DEBUG 2021-04-28 16:01:30,643 packet from service: {'type': 'info', 'info_id': 'nabweatherd', 'animation': {'tempo': 125, 'colors': [{'left': '000000', 'center': 'ffff00', 'right': '000000'}, {'left': '0000ff', 'center': '000000', 'right': '0000ff'}]}}
DEBUG 2021-04-28 16:02:26,101 packet from service: {'type': 'command', 'sequence': [{'choreography': 'nabtaichid/taichi.chor'}], 'expiration': '2021-04-28T14:03:25.990097+00:00'}
INFO 2021-04-28 16:02:36,083 Crash in choreography interpreter: Traceback (most recent call last):
  File "/home/pi/pynab/nabd/choreography.py", line 433, in play
    await self.play_binary(chor)
  File "/home/pi/pynab/nabd/choreography.py", line 312, in play_binary
    await self.do_play_binary(4, chor, opcodes, timescale)
  File "/home/pi/pynab/nabd/choreography.py", line 345, in do_play_binary
    index = await handler(index, chor)
  File "/home/pi/pynab/nabd/choreography.py", line 300, in attend
    await self.sound.wait_until_done(self.cancel_event)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 186, in wait_until_done
    await wait_with_cancel_event(self.future, event, self.stop_playing)
  File "/home/pi/pynab/nabd/cancel.py", line 20, in wait_with_cancel_event
    await task
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 287, in <lambda>
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found

[... cut ...]
future: <Task finished coro=<NabService.client_loop() done, defined at /home/pi/pynab/nabcommon/nabservice.py:44> exception=RuntimeError('This event loop is already running')>
Traceback (most recent call last):
  File "/home/pi/pynab/nabcommon/nabservice.py", line 84, in client_loop
    self.loop.run_until_complete(self.stop_service_loop())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 571, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 526, in run_forever
    raise RuntimeError('This event loop is already running')
RuntimeError: This event loop is already running
```

cyril1929, 2021-04-29:
For my part, since I did a complete re-installation of the image on the SD card, I no longer have the problem.
But maybe the problem appears after an "apt upgrade"...

f-laurens, 2022-02-05:
Does this issue still occur with latest [(v0.9.1)](https://github.com/nabaztag2018/pynab/releases/tag/v0.9.1) release, or should it be closed?

cyril1929, 2022-02-06:
Yes it's Ok for me to close it.
The problem no longer occurs for me.

fcharlier, 2022-02-08:
Let me verify, I had an automatic reboot everyday to workaround this problem. 

I've just disabled it, I'll update this issue in a few days to let you know how it goes.

fcharlier, 2022-02-11:
The issue is still present after a few days of uptime (almost 4 this time). Logs are:

```
pi@nabaztag:~ $ cat /var/log/nabd.log
INFO 2022-02-06 08:31:22,447 nabd started with log level INFO
INFO 2022-02-07 08:31:21,922 nabd started with log level INFO
ERROR 2022-02-10 22:03:17,361 Future exception was never retrieved
future: <Future finished exception=LibInitializationException('libmpg123 not found')>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 287, in <lambda>
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found
ERROR 2022-02-10 22:12:03,259 Future exception was never retrieved
future: <Future finished exception=LibInitializationException('libmpg123 not found')>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 287, in <lambda>
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found
ERROR 2022-02-10 22:26:46,934 Future exception was never retrieved
future: <Future finished exception=LibInitializationException('libmpg123 not found')>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 287, in <lambda>
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found

[… removing a few hundreds of occurrences of the same message …]

ERROR 2022-02-11 09:28:39,452 Future exception was never retrieved
future: <Future finished exception=LibInitializationException('libmpg123 not found')>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 287, in <lambda>
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found
ERROR 2022-02-11 09:28:39,745 Future exception was never retrieved
future: <Future finished exception=LibInitializationException('libmpg123 not found')>
Traceback (most recent call last):
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
[... 11 lines cut ...]
    await self.play_binary(chor)
  File "/home/pi/pynab/nabd/choreography.py", line 313, in play_binary
    await self.do_play_binary(4, chor, opcodes, timescale)
  File "/home/pi/pynab/nabd/choreography.py", line 346, in do_play_binary
    index = await handler(index, chor)
  File "/home/pi/pynab/nabd/choreography.py", line 301, in attend
    await self.sound.wait_until_done(self.cancel_event)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 186, in wait_until_done
    await wait_with_cancel_event(self.future, event, self.stop_playing)
  File "/home/pi/pynab/nabd/cancel.py", line 20, in wait_with_cancel_event
    await task
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 287, in <lambda>
    self.executor, lambda f=filename: self._play(f)
  File "/home/pi/pynab/nabd/sound_alsa.py", line 137, in _play
    mp3 = Mpg123(filename)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 110, in __init__
    self._lib = self.init_library(library_path)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/mpg123.py", line 99, in init_library
    raise self.LibInitializationException('libmpg123 not found')
mpg123.Mpg123.LibInitializationException: libmpg123 not found
```

If you need anything that could help diagnose the problem don't hesitate to ask me.

f-laurens, 2022-03-24:
It would be useful to:

1. Make a list of the Linux packages installed on the SD card with which problem occurs:
`dpkg-query --list > installed.lst1`
2. Check if the problem occurs with a "fresh" Pynab 0.9.1 image. If it does not, make a list of its installed packages:
 `dpkg-query --list > installed.lst0`

Then comparing the two lists might possibly give a clue about what package(s) cause the problem.

fcharlier, 2022-05-13:
I  finally took the time to flash a "fresh" Pynab 0.9.1 image, it's running now, I'll post the results in a few days.

fcharlier, 2022-05-30:
Hi.

Finally, with a fresh image, except a crash after 5 days due to usual Raspberry PI / SD card issues (`Task kworker […] blocked for more than [XXX] seconds` `Workqueue: events_freezable mmc_rescan`) my nabaztag doesn't exhibit the problem with libmpg123 mentioned above.

I attached the packages list anyway just in case:

- [installed.lst0](https://github.com/nabaztag2018/pynab/files/8797135/installed.lst0.log) has the packages from the "fresh" install
- [installed.lst1.log](https://github.com/nabaztag2018/pynab/files/8797136/installed.lst1.log) has the packages with problems occurring.

I also kept a dump of the image with problems for further analysis if needed.

I'll close the issue for now since it seems to be resolved with the latest images.

closed 2022-05-30 by fcharlier