nabaztag2018/pynab #152: Meteo no longer works

Nicosss, 2020-08-05:
Meteo no longer works since the update of the site http://meteofrance.com .

```
Aug 05 17:51:18 nabaztag systemd[1]: nabweatherd.service: Main process exited, code=exited, status=1/FAILURE
Aug 05 17:51:18 nabaztag systemd[1]: nabweatherd.service: Failed with result 'exit-code'.
Aug 05 17:51:19 nabaztag systemd[1]: nabweatherd.service: Service RestartSec=1s expired, scheduling restart.
Aug 05 17:51:19 nabaztag systemd[1]: nabweatherd.service: Scheduled restart job, restart counter is at 55.
Aug 05 17:51:19 nabaztag systemd[1]: Stopped Nabaztag weather daemon.
Aug 05 17:51:19 nabaztag systemd[1]: Started Nabaztag weather daemon.
Aug 05 17:51:27 nabaztag python[1517]: Traceback (most recent call last):
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/urllib3/connectionpool.py", line 672, in urlopen
Aug 05 17:51:27 nabaztag python[1517]:     chunked=chunked,
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/urllib3/connectionpool.py", line 376, in _make_request
Aug 05 17:51:27 nabaztag python[1517]:     self._validate_conn(conn)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/urllib3/connectionpool.py", line 994, in _validate_conn
Aug 05 17:51:27 nabaztag python[1517]:     conn.connect()
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/urllib3/connection.py", line 394, in connect
Aug 05 17:51:27 nabaztag python[1517]:     ssl_context=context,
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/urllib3/util/ssl_.py", line 370, in ssl_wrap_socket
Aug 05 17:51:27 nabaztag python[1517]:     return context.wrap_socket(sock, server_hostname=server_hostname)
Aug 05 17:51:27 nabaztag python[1517]:   File "/usr/lib/python3.7/ssl.py", line 412, in wrap_socket
Aug 05 17:51:27 nabaztag python[1517]:     session=session
Aug 05 17:51:27 nabaztag python[1517]:   File "/usr/lib/python3.7/ssl.py", line 853, in _create
Aug 05 17:51:27 nabaztag python[1517]:     self.do_handshake()
Aug 05 17:51:27 nabaztag python[1517]:   File "/usr/lib/python3.7/ssl.py", line 1117, in do_handshake
Aug 05 17:51:27 nabaztag python[1517]:     self._sslobj.do_handshake()
Aug 05 17:51:27 nabaztag python[1517]: ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)
Aug 05 17:51:27 nabaztag python[1517]: During handling of the above exception, another exception occurred:
Aug 05 17:51:27 nabaztag python[1517]: Traceback (most recent call last):
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
Aug 05 17:51:27 nabaztag python[1517]:     timeout=timeout
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/urllib3/connectionpool.py", line 720, in urlopen
Aug 05 17:51:27 nabaztag python[1517]:     method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/urllib3/util/retry.py", line 436, in increment
Aug 05 17:51:27 nabaztag python[1517]:     raise MaxRetryError(_pool, url, error or ResponseError(cause))
Aug 05 17:51:27 nabaztag python[1517]: urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='meteofrance.com', port=443): Max retries exceeded with url: / (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)')))
Aug 05 17:51:27 nabaztag python[1517]: During handling of the above exception, another exception occurred:
Aug 05 17:51:27 nabaztag python[1517]: Traceback (most recent call last):
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/meteofrance/client.py", line 51, in _init_codes
Aug 05 17:51:27 nabaztag python[1517]:     results = requests.get(url, timeout=10).json()
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/api.py", line 75, in get
Aug 05 17:51:27 nabaztag python[1517]:     return request('get', url, params=params, **kwargs)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/api.py", line 60, in request
Aug 05 17:51:27 nabaztag python[1517]:     return session.request(method=method, url=url, **kwargs)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
Aug 05 17:51:27 nabaztag python[1517]:     resp = self.send(prep, **send_kwargs)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/sessions.py", line 668, in send
Aug 05 17:51:27 nabaztag python[1517]:     history = [resp for resp in gen] if allow_redirects else []
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/sessions.py", line 668, in <listcomp>
Aug 05 17:51:27 nabaztag python[1517]:     history = [resp for resp in gen] if allow_redirects else []
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/sessions.py", line 247, in resolve_redirects
Aug 05 17:51:27 nabaztag python[1517]:     **adapter_kwargs
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
Aug 05 17:51:27 nabaztag python[1517]:     r = adapter.send(request, **kwargs)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/requests/adapters.py", line 514, in send
Aug 05 17:51:27 nabaztag python[1517]:     raise SSLError(e, request=request)
Aug 05 17:51:27 nabaztag python[1517]: requests.exceptions.SSLError: HTTPSConnectionPool(host='meteofrance.com', port=443): Max retries exceeded with url: / (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)')))
Aug 05 17:51:27 nabaztag python[1517]: During handling of the above exception, another exception occurred:
Aug 05 17:51:27 nabaztag python[1517]: Traceback (most recent call last):
Aug 05 17:51:27 nabaztag python[1517]:   File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
Aug 05 17:51:27 nabaztag python[1517]:     "__main__", mod_spec)
Aug 05 17:51:27 nabaztag python[1517]:   File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
Aug 05 17:51:27 nabaztag python[1517]:     exec(code, run_globals)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/nabweatherd/nabweatherd.py", line 585, in <module>
Aug 05 17:51:27 nabaztag python[1517]:     NabWeatherd.main(sys.argv[1:])
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 206, in main
Aug 05 17:51:27 nabaztag python[1517]:     service.run()
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 139, in run
Aug 05 17:51:27 nabaztag python[1517]:     raise ex
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/nabcommon/nabservice.py", line 327, in service_loop
[... 4 lines cut ...]
Aug 05 17:51:27 nabaztag python[1517]:     info_data = await self.fetch_info_data(config)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/nabweatherd/nabweatherd.py", line 429, in fetch_info_data
Aug 05 17:51:27 nabaztag python[1517]:     client = await sync_to_async(meteofranceClient)(location, True)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/asgiref/sync.py", line 244, in __call__
Aug 05 17:51:27 nabaztag python[1517]:     return await asyncio.wait_for(future, timeout=None)
Aug 05 17:51:27 nabaztag python[1517]:   File "/usr/lib/python3.7/asyncio/tasks.py", line 388, in wait_for
Aug 05 17:51:27 nabaztag python[1517]:     return await fut
Aug 05 17:51:27 nabaztag python[1517]:   File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
Aug 05 17:51:27 nabaztag python[1517]:     result = self.fn(*self.args, **self.kwargs)
Aug 05 17:51:27 nabaztag python[1517]:   File "/home/pi/pynab/venv/lib/python3.7/site-packages/asgiref/sync.py", line 277, in thread_handler
[... cut ...]
Aug 05 17:51:28 nabaztag systemd[1]: nabweatherd.service: Failed with result 'exit-code'.
Aug 05 17:51:29 nabaztag systemd[1]: nabweatherd.service: Service RestartSec=1s expired, scheduling restart.
Aug 05 17:51:29 nabaztag systemd[1]: nabweatherd.service: Scheduled restart job, restart counter is at 56.
Aug 05 17:51:29 nabaztag systemd[1]: Stopped Nabaztag weather daemon.
Aug 05 17:51:29 nabaztag systemd[1]: Started Nabaztag weather daemon.
```

That seems to be an error with SSL certificate verification?

nguillaumin, 2020-08-05:
It looks like the URLs used by the Python client no longer work:

```python
SEARCH_API = 'http://www.meteofrance.com/mf3-rpc-portlet/rest/lieu/facet/previsions/search/{}'
RAIN_FORECAST_API = 'http://www.meteofrance.com/mf3-rpc-portlet/rest/pluie/{}/'
WEATHER_FORECAST_URL = 'http://www.meteofrance.com/previsions-meteo-france/{}/{}'
WEATHER_FORECAST_WORLD_URL = 'http://www.meteofrance.com/previsions-meteo-monde/{}/{}'
```

Not sure where they were coming from, possibly reverse-engineered from the meteo widget that is offered by Météo France: http://meteofrance.com/widgets ? But on the current widgets the URLs are different: https://rpcache-aa.meteofrance.com/internet2018client/2.0/forecast?lat=48.859333&lon=2.340591 for example, but it requires a token.

There's also the official webservices: https://donneespubliques.meteofrance.fr/?fond=geoservices&id_dossier=14 although it also requires a token.

So I'm not sure there's an easy fix. It looks like the client needs to be re-implemented (or use another client on the webserices like https://github.com/hacf-fr/meteofrance-api ), and then the token needs to be dealt with somehow. I'm not sure we can get a single token for the whole project and distribute it, or if each user need to get their own token :fearful: 

Probably a good opportunity to abstract the weather provider but it's a fair amount of work...

f-laurens, 2020-08-05:
Indeed, the new version of the Météo France web site that went live on August 3, 2020 seems to have also "broken" the (unofficial?) API that was used by nabweatherd...
Indeed not an easy fix. Météo France webservices documentation looks a bit obscure... Do they provide the 'simple' weather forecast information we needed?

232enero, 2020-08-06:
Yep, the API is broken. https://github.com/hacf-fr/meteofrance-api works but OpenWeatherMap would probably be best. Checking this...

datagutt, 2020-08-26:
OpenWeatherMap probably has better data for other countries

232enero, 2020-10-07:
Fixed in 0.8.2

closed 2020-10-07 by 232enero