nabaztag2018/pynab #153: CLI tool + wakeup by systemd timer

guilhem, 2020-08-30:
Hi,
waiting for #125 to be merged, I wanted to use systemd timers to wakeup my rabbit later on weekend.
I watched how to do this… and `netcat` was too simple for someone sleepy and angry like me on Sunday morning.
So I worked on a go CLI tool to use the PROTOCOL.

Work is basic (for now) and only manage `nabcli sleep` and `nabcli wakeup` but I may improve this later if needed.

https://github.com/guilhem/nabcli

guilhem, 2020-08-30:
To be manipulated directly from pynab, It can be a good idea to use systemd timer through DBUS interface.
https://www.freedesktop.org/wiki/Software/systemd/dbus/

Here is an example of usage https://trstringer.com/python-systemd-dbus/

f-laurens, 2021-05-21:
To be closed since #125 is now merged?

closed 2021-05-23 by omevel