nabaztag2018/pynab #186: wrong ELF class: ELFCLASS64 when bash upgrade.sh

nabaztag2018, 2021-04-27:
There's an error when updating the NLU models.

+ echo 'Updating NLU models - 8/14'
+ '[' '!' -d venv/lib/python3.7/site-packages/snips_nlu_fr ']'
+ echo 'Downloading snips_nlu models for French'
Downloading snips_nlu models for French
+ venv/bin/python -m snips_nlu download fr
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.7/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu/__init__.py", line 4, in <module>
    from snips_nlu.nlu_engine import SnipsNLUEngine
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu/nlu_engine/__init__.py", line 1, in <module>
    from snips_nlu.nlu_engine.nlu_engine import SnipsNLUEngine
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu/nlu_engine/nlu_engine.py", line 29, in <module>
    from snips_nlu.intent_parser import IntentParser
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu/intent_parser/__init__.py", line 3, in <module>
    from .lookup_intent_parser import LookupIntentParser
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu/intent_parser/lookup_intent_parser.py", line 11, in <module>
    from snips_nlu_utils import normalize, hash_str
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu_utils/__init__.py", line 1, in <module>
    from snips_nlu_utils.string import (
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu_utils/string.py", line 4, in <module>
    from snips_nlu_utils.utils import lib, string_pointer, check_ffi_error
  File "/home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu_utils/utils.py", line 10, in <module>
    lib = cdll.LoadLibrary(str(dylib_path))
  File "/usr/lib/python3.7/ctypes/__init__.py", line 434, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /home/pi/pynab/venv/lib/python3.7/site-packages/snips_nlu_utils/dylib/libsnips_nlu_utils_rs.cpython-37m-x86_64-linux-gnu.so: wrong ELF class: ELFCLASS64
++ s=1
++ echo 'install.sh: Error on line 198: venv/bin/python -m snips_nlu download fr'
install.sh: Error on line 198: venv/bin/python -m snips_nlu download fr
++ exit 1
++ s=1
++ echo 'upgrade.sh: Error on line 45: sudo -u ${owner} bash install.sh --upgrade'
upgrade.sh: Error on line 45: sudo -u ${owner} bash install.sh --upgrade
++ exit 1
++ s=1
++ echo 'upgrade.sh: Error on line 41: bash upgrade.sh "newversion"'
upgrade.sh: Error on line 41: bash upgrade.sh "newversion"
++ exit 1

f-laurens, 2021-04-27:
This regression when running Pynab upgrade/install scripts natively seems introduced by PR #181 (commit https://github.com/nabaztag2018/pynab/pull/181/commits/28863549d196073247cb84a53e4d9e15eb976815):
When i rollback `requirements.txt` file to previous version, I'm again able to successfully run these scripts on my test Rabbit...

mehdichaouch, 2021-04-30:
>... on my test Rabbit...
@f-laurens how many rabbit you have? 🤔

f-laurens, 2021-04-30:
> how many rabbit you have?

3 (Nabaztag:tag )rabbits at the moment.
I'd say 2 is a reasonable minimum: 1 production rabbit and 1 test rabbit. It is also useful for testing ear pairing (and this: https://github.com/nabaztag2018/pynab/pull/157).

mehdichaouch, 2021-04-30:
Oh 3, Great! Lucky you!
What is ear pairing? to sync Nabaztag ?

f-laurens, 2021-04-30:
> What is ear pairing? to sync Nabaztag ?

Ear movement sync (between 2 rabbits married through Mastodon).

f-laurens, 2021-05-01:
@nabaztag2018 @pguyot:
after noticing the new snips releases published here: https://github.com/pguyot/snips-nlu-utils/releases/tag/v0.9.1 and here: https://github.com/pguyot/snips-nlu-parsers/releases/tag/v0.4.3, I gave a new try at running natively the Pynab upgrade script with the current (PR #181) version of `requirements.txt`:

I then get the following errors:
```
+ venv/bin/pip install -r requirements.txt
...
`Collecting snips-nlu-parsers==0.4.3 from https://github.com/pguyot/snips-nlu-parsers/releases/download/0.4.3/snips_nlu_parsers-0.4.3-cp37-cp37m-linux_armv6l.whl (from -r requirements.txt (line 23))`
  HTTP error 404 while getting https://github.com/pguyot/snips-nlu-parsers/releases/download/0.4.3/snips_nlu_parsers-0.4.3-cp37-cp37m-linux_armv6l.whl
  Could not install requirement snips-nlu-parsers==0.4.3 from https://github.com/pguyot/snips-nlu-parsers/releases/download/0.4.3/snips_nlu_parsers-0.4.3-cp37-cp37m-linux_armv6l.whl (from -r requirements.txt (line 23)) because of error 404 Client Error: Not Found for url: https://github.com/pguyot/snips-nlu-parsers/releases/download/0.4.3/snips_nlu_parsers-0.4.3-cp37-cp37m-linux_armv6l.whl
install.sh: Error on line 188: venv/bin/pip install -r requirements.txt
...
Collecting snips-nlu-utils==0.9.1 from https://github.com/pguyot/snips-nlu-utils/releases/download/0.9.1/snips_nlu_utils-0.9.1-cp37-cp37m-linux_armv6l.whl (from -r requirements.txt (line 25))
  HTTP error 404 while getting https://github.com/pguyot/snips-nlu-utils/releases/download/0.9.1/snips_nlu_utils-0.9.1-cp37-cp37m-linux_armv6l.whl
  Could not install requirement snips-nlu-utils==0.9.1 from https://github.com/pguyot/snips-nlu-utils/releases/download/0.9.1/snips_nlu_utils-0.9.1-cp37-cp37m-linux_armv6l.whl (from -r requirements.txt (line 25)) because of error 404 Client Error: Not Found for url: https://github.com/pguyot/snips-nlu-utils/releases/download/0.9.1/snips_nlu_utils-0.9.1-cp37-cp37m-linux_armv6l.whl
install.sh: Error on line 188: venv/bin/pip install -r requirements.txt
...
```
These HTTP 404 errors on fetching the wheel packages are really weird, since the packages are present on GitHub !!!
Downloading them by hand and unziping them into `./venv/lib/python3.7/site-packages`fixes the problem:
no more requirements related problems, and the Pynab upgrade script then runs correctly to completion.

BTW, this issue should maybe be renamed to a more generic title like '"**requirements related Pynab upgrade errors**".

pguyot, 2021-05-02:
The URL changed on purpose as original archives were faulty. I have yet to test them on a real Nabaztag before fixing requirements.txt, but apparently you did :-)
https://github.com/pguyot/snips-nlu-parsers/releases/download/0.4.3/snips_nlu_parsers-0.4.3-cp37-cp37m-linux_armv6l.whl
should be:
https://github.com/pguyot/snips-nlu-parsers/releases/download/v0.4.3/snips_nlu_parsers-0.4.3-cp37-cp37m-linux_armv6l.whl

Notice the v in the tag name.

f-laurens, 2021-05-02:
> Notice the v in the tag name.

Oooops... Sorry. I failed to spot the missing 'v' from the URLs in the script.

pguyot, 2021-05-02:
Fixed in #193

closed 2021-05-02 by pguyot