nabaztag2018/pynab #208: Visual information is not enabled by default for air quality & weather

pguyot, 2021-05-08:
As a result, rabbit is dark on first boot.

f-laurens, 2021-05-13:
According to to `makemigrations`, one migration seems to be missing from PR #221:
```
pi@Bunny:~/pynab $ /home/pi/pynab/venv/bin/python manage.py makemigrations
Migrations for 'nabweatherd':
  nabweatherd/migrations/0012_auto_20210512_1412.py
    - Alter field weather_animation_type on config
```
```
pi@Bunny:~/pynab $ cat nabweatherd/migrations/0012_auto_20210512_1412.py
# Generated by Django 3.0.14 on 2021-05-12 14:12

from django.db import migrations, models

class Migration(migrations.Migration):

    dependencies = [
        ('nabweatherd', '0011_auto_20210510_0944'),
    ]

    operations = [
        migrations.AlterField(
            model_name='config',
            name='weather_animation_type',
            field=models.TextField(default='weather_and_rain', null=True),
        ),
    ]
```

mehdichaouch, 2021-05-13:
In my case, I have no trouble with visual info about weather or air quality.

PR #234 (2021-05-20): Add data model migration missing from PR#221.
Resolves #208 Resolves #209 (cont'd from PR #221).
See https://github.com/nabaztag2018/pynab/issues/208#issuecomment-840409353.
closed 2021-05-20 by nabaztag2018