nabaztag2018/pynab #223: Logs are too verbose

omevel, 2021-05-11:
The current release has a DEBUG logging level and as SD card are not really that suited for heavy lifting it would be better to minimize the numer of write (i have received a couple of email at support of people whose SD card needed to be reflashed).

The easiest step is to lower logging to ERROR but are they other things we could do ?

f-laurens, 2021-05-11:
> The current release has a DEBUG logging level and as SD card are not really that suited for heavy lifting it would be better to minimize the numer of write (i have received a couple of email at support of people whose SD card needed to be reflashed).

The Pynab daemons currently don't do too much logging, even in DEBUG mode. Certainly not enough to stress a SD card to the point of damaging it...
If people had to reformat cards, the root cause is probably elsewhere:

- other software stressing the card?
- user error?
-  low quality card? (I personally never had a problem with SanDisk cards, even on a 'test' Rabbit with a lot of activity)

> The easiest step is to lower logging to ERROR but are they other things we could do ?

This would mean basically no logging...  :-(

omevel, 2021-05-12:
> Why not adding a select form in UI to set the verbosity of log? and setting it by default on nothing or critical?

Good idea !

> Adding a voice recognition for something like "bye bye", which switch off the Nabaztag properly and play sound.

Oh ! Very very good idea !

pguyot, 2021-05-13:
The corruption of SD cards with RaspberryPis is a real design issue and we probably should reduce or disable logging by default.

f-laurens, 2021-05-13:
> The corruption of SD cards with RaspberryPis is a real design issue

Agreed...
> and we probably should reduce or disable logging by default.

... but reducing Pynab logging (which is quite reasonable IMHO) certainly won't help much.
As I mentionned above, with a good brand of SD cards, I never had a corruption problem even on a 'test' Rabbit with a lot of activity/stress on the card.

kymodoke, 2021-05-16:
It also mainly depends on the distribution used as base system... Raspbian/RaspberryPi OS is good for general purpose but not optimized for a specific purpose.
There are distributions made for Pi which handle specific usages way better than Raspian, either optimized for RAM usage or where most of the OS is read-only, with also the usage of ramdisk. For instance [Dietpi](https://dietpi.com/) and [PiCore](http://www.tinycorelinux.net/ports.html).

_For instance I'm using [picoreplayer](https://www.picoreplayer.org/) on a Raspi Zero as a multiroom audio client, I can power it down electrically without any risk damaging the SD card. And for a Wireguard VPN + Mosquitto server, I'm using Armbian on another board with **overlayroot** (read-only system) to get an unbreakable filesystem._

f-laurens, 2021-05-29:
PR #255 should slightly improve the situation by prominently documenting the clean shutdown methods on the help page.

omevel, 2021-06-01:
Everyone OK to lower the logging level to ERROR to minimize the risk of damaging the SD when someone brutally (and despite everything we're written all over the place) switch off its rabbit ?

 logger.setLevel(logging.ERROR)

f-laurens, 2021-06-01:
If you wish... (although I'm skeptical about the effective outcome).
But **before** doing this, it might be wise to review in the  code all calls `to logging.<level>` to make sure each call is made with the adequate level:
 - assuming the (few) `error` and `critical` logs have the right level...,
- some `debug` logs might deserve to be promoted to `info` level;
- and reversely some `info` logs might need to be demoted to `debug` level.

f-laurens, 2021-06-02:
I'm working on a PR reviewing/amending the `logging` calls.
Will submit it shortly...

f-laurens, 2021-06-03:
Tracebacks for unhandled service exceptions (for example connection errors in `nabairqualityd` and `nabweatherd`) may also generate a lot of logging in system logs (`/var/log/syslog` `/var/log/daemon.log`).
PR #268 handles this.

f-laurens, 2021-06-05:
Another cause of uncontrolled logging are `git` error messages (from `nabweb` Update page), which apparently show up when using a purely local (no upstream) branch for Pynab.
To be investigated...

f-laurens, 2021-06-08:
What else can we do?
Shall we close this issue (for now)?

f-laurens, 2021-06-21:
> There are distributions made for Pi which handle specific usages way better than Raspbian, either optimized for RAM usage or where most of the OS is read-only, with also the usage of ramdisk. For instance [Dietpi](https://dietpi.com/) and [PiCore](http://www.tinycorelinux.net/ports.html).

DietPi support for Pynab is now available: see PR #283.

closed 2022-03-14 by nabaztag2018