nabaztag2018/pynab #224: Web interface is not available on May 11th releng

omevel, 2021-05-11:
pi@releng:~ $  sudo netstat -tulpn | grep LISTEN
tcp        0      0 127.0.0.1:10543         0.0.0.0:*               LISTEN      1/init              
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      511/nabblockly      
tcp        0      0 0.0.0.0:42833           0.0.0.0:*               LISTEN      511/nabblockly      
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      477/sshd            
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN      482/postgres        
tcp        0      0 127.0.0.1:8000          0.0.0.0:*               LISTEN      508/python3.7       
tcp6       0      0 :::4369                 :::*                    LISTEN      1/init              
tcp6       0      0 :::22                   :::*                    LISTEN      477/sshd            
tcp6       0      0 ::1:5432                :::*                    LISTEN      482/postgres        
pi@releng:~ $

f-laurens, 2021-05-12:
Cannot reproduce this behaviour with an existing (v8.0.4) image upgraded to to `latest` state of releng branch:
nginx is running/listening on port 80, nabweb behaves normally.

mehdichaouch, 2021-05-12:
I get this :
```
pi@nabaztag:~ $ sudo netstat -tulpn | grep LISTEN
tcp        0      0 127.0.0.1:5432          0.0.0.0:*               LISTEN      602/postgres
tcp        0      0 127.0.0.1:8000          0.0.0.0:*               LISTEN      631/python3.7
tcp        0      0 0.0.0.0:45675           0.0.0.0:*               LISTEN      627/nabblockly
tcp        0      0 127.0.0.1:10543         0.0.0.0:*               LISTEN      1/init
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      627/nabblockly
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      546/sshd
tcp6       0      0 ::1:5432                :::*                    LISTEN      602/postgres
tcp6       0      0 :::4369                 :::*                    LISTEN      1/init
tcp6       0      0 :::22                   :::*                    LISTEN      546/sshd
```
and access to http://nabaztag.local won't work.

f-laurens, 2021-05-12:
Does `/var/log/nginx `exist? (see #207 #212)

mehdichaouch, 2021-05-12:
> Does `/var/log/nginx `exist on your images? (see #207 #212)

No folder `/var/log/nginx`, I created this one manually and after reboot I can access to the web interface. Thanks.
I continue my test tomorrow...

f-laurens, 2021-05-12:
The test for existence/creation of `/var/log/nginx` should probably be moved from the `setup` script to (the nginx related section of) the `install.sh` script, for safety.

pguyot, 2021-05-16:
The issue comes from pishrink that empties /var/log among other things.

https://github.com/Drewsif/PiShrink/blob/9d9c0dd32cd988f9c3bd9953ca245d33a52fc9e3/pishrink.sh#L311

pguyot, 2021-05-18:
Fixed in #235

closed 2021-05-18 by pguyot

mehdichaouch, 2021-06-06:
I installed `pynab-nightly-2021-06-05.img.xz`and I get again trouble with missing ` /var/log/nginx`.
Here are the logs files I have :
```
pi@nabaztag:~ $ ls -lsha /var/log/
total 464K
4.0K drwxr-xr-x  4 root root     4.0K Mar  5 00:07 .
4.0K drwxr-xr-x 12 root root     4.0K Mar  5 00:05 ..
4.0K -rw-r-----  1 root adm      3.1K Jun  6 02:45 auth.log
   0 -rw-rw----  1 root utmp        0 Mar  5 00:05 btmp
 84K -rw-r-----  1 root adm       79K Jun  6 02:45 daemon.log
8.0K -rw-r-----  1 root adm      4.1K Jun  6 02:40 debug
 76K -rw-r-----  1 root adm       75K Jun  6 02:42 kern.log
4.0K -rw-rw-r--  1 root utmp     286K Jun  6 02:45 lastlog
 68K -rw-r-----  1 root adm       68K Jun  6 02:42 messages
4.0K -rw-r--r--  1 root root      134 Jun  6 02:41 nab8balld.log
4.0K -rw-r--r--  1 root root      752 Jun  6 02:42 nabairqualityd.log
4.0K -rw-r--r--  1 root root      132 Jun  6 02:41 nabbookd.log
4.0K -rw-r--r--  1 root root      300 Jun  6 02:42 nabclockd.log
4.0K -rw-r--r--  1 root root      124 Jun  6 02:41 nabd.log
4.0K -rw-r--r--  1 root root      140 Jun  6 02:42 nabmastodond.log
4.0K -rw-r--r--  1 root root      140 Jun  6 02:41 nabsurprised.log
4.0K -rw-r--r--  1 root root      136 Jun  6 02:41 nabtaichid.log
4.0K -rw-r--r--  1 root root      138 Jun  6 02:42 nabweatherd.log
4.0K drwxrwxr-t  2 root postgres 4.0K Mar  5 00:06 postgresql
4.0K drwx------  2 root root     4.0K Mar  5 00:05 private
156K -rw-r-----  1 root adm      155K Jun  6 02:45 syslog
4.0K -rw-r-----  1 root adm       167 Jun  6 02:42 user.log
8.0K -rw-rw-r--  1 root utmp     5.7K Jun  6 02:45 wtmp
```
Can someone else test the lastest nigthly build? And if someone have any idea of this bug?

f-laurens, 2021-06-06:
> Can someone else test the lastest nigthly build? And if someone have any idea of this bug?

No need to test here. This is due to a bug in the `nightly.yml` GitHub workflow, which still called PiShrink with the -p option.
This is now fixed in PR #271.
But this reveals a flaw in `nightly.yml`, which cannot ensure it builds in the same way as `arm-runner.yml` since it duplicates its functions. A way should be found to have these two workflows share the same code/commands for their common steps.

f-laurens, 2021-06-06:
To go back to the original issue #207, a better solution than #235 might be to add a 'pre' step to `nabweb.service:`
`ExecStartPre=mkdir -p /var/log/nginx`
This would allow reintroducing the `-p` option of PiShrink which would (if it doesn't not have other nasty side effects..) help reduce image size.

f-laurens, 2021-06-06:
> ... add a 'pre' step to `nabweb.service:`

Hmmm... More tricky than that: it is `nginx.service` that would need this 'pre' step, and it is not ours...