nabaztag2018/pynab #228: Boot chime may be a nuisance if boot occurs while rabbit is asleep.

f-laurens, 2021-05-12:
Boot chime (introduced in #180) may be an audio nuisance if boot occurs (typically following a nightly reboot of the OS) during a period where rabbit is in asleep state.
This chime should not be played in this case.

mehdichaouch, 2021-05-12:
I confirm the sound during asleep state.

f-laurens, 2021-05-13:
I am currently unable to find a fix in  `nabd/nabd.py` for this :-(
It is not enough to suppress the call to `self.boot_playsound()` in `sleep_setup()` (call which by the way must be useless since one is done in `_do_transition_to_idle()`.
The issue is that nabd starts in `IDLE` state (which is normal), before going possibly (if this is a 'sleep period') to `ASLEEP` state.

One could think of deferring playing the boot chime till a `wakeup` packed is received. But this would introduce an implicit dependency on nabbclockd: if nabclockd is disabled/stopped, there will not be such a packet.

Am I missing something here?

Is there technically a possible fix that would allow implementing the 'boot chime' feature without violating the _"rabbit should keep silent when asleep"_ golden rule?
If not the only solution seems to remove this feature altogether (suggesting to the users to workaround the problem by manually removing the boot chime files does not seem acceptable).

pguyot, 2021-05-13:
Can we move the feature to nabclockd?

f-laurens, 2021-05-13:
> Can we move the feature to nabclockd?

Why not, if it makes sense WRT the original request (#59), and if users are aware the feature is available only if nabclockd is enabled?
But then my fix above (_deferring playing the boot chime till a wakeup packed is received_) would also be acceptable.

PR #231 (2021-05-16): Remove 'boot chime' feature.
Resolves #228.
This removes the 'boot chime feature' altogether, pending a new implementation that would keep the rabbit silent while asleep.
closed 2021-05-16 by pguyot