nabaztag2018/pynab #245: Airquality location does not refresh after modification in the Weather forecast

omevel, 2021-05-23:
As mentionned in the PR by @f-laurens , the Airquality location is now based on the Weather location and it's not taken into account (until we do a manual service restart)

f-laurens, 2021-05-23:
> (until we do a manual service restart)

Not quite true: **no service restart is required:** as soon as the Weather _location_ is changed, it (its latitude/longitude) will be effectvely used by Air Quality in its following calls to the AQICN API, which will itself return the _location of the station_ used to get the air quality index.
As I mentionned here: https://github.com/nabaztag2018/pynab/pull/237#issuecomment-846176969, the only little glitch is that this station location is not immediately updated in the Air Quality widget in the web interface.

omevel, 2021-05-23:
Ok, i meant for it to be seen on the web interface. It is very confusing for the user to see the old location for Air quality even after a refresh of the page. A call to NabAirquality.signal_daemon() at least ?

f-laurens, 2021-05-23:
> A call to NabAirquality.signal_daemon() at least ?

I think I tried this, and it didn't work for some reason...

Until someone finds the appropriate fix, a workaround (again: **no service restart** needed) is, after changing the _Location_ in the Weather widget, to change the _Index_ in the Air Quality widget and back.

omevel, 2021-05-24:
Or maybe until we find a better solution we can just not display the city in the Airquality widget, this will be less confusing I think.

f-laurens, 2021-05-24:
> Or maybe until we find a better solution we can just not display the _location_ in the Airquality widget,

Hmm... not sure about this:

- After all, this is **not a new problem**: it was already present when the IP address based API was used (the widget refresh logic has not changed): a location change due to IP address change was not immediately reflected in the widget.
- this station location, which is different from the city location in the Weather widget, is valuable information.
- if we don't display this location, we might as well close this issue:  there will be no more motivation for anyone to fix it...

omevel, 2021-05-24:
Ok !

closed 2021-05-24 by omevel

f-laurens, 2021-05-24:
@omevel: Why close this issue?
We should rather keep it open, so that someone finds a fix...

omevel, 2021-05-24:
I think it would probably be better to either have a separate location selection for the two services (some users set a weather forecast location different from the one they live in) or, if we wanted to have a unique setting, put the location in the "general" setting so that it's easier to understand for the user.

but let's keep your PR ! (as it fixes a bug with the Airquality location geoloc)

f-laurens, 2021-05-24:
> (some users set a weather forecast location different from the one they live in)

while still wanting air quality for their home location??? Your users are really weird!
(but after all, anyone who wants to use a rabbit to get this kind of info must be somehow brain damaged ;-)

> I think it would probably be better to either have a separate location selection for the two services

You are kind of opening Pandora's box here... Then the generic solution would be to have 2..N selections to have weather/air quality info for 2..N different locations: do we want this?

> or, if we wanted to have a unique setting, put the location in the "general" setting so that it's easier to understand for the user.

Hmmm.... Moving it from 'Weather' to 'General' would not in any way help to solve our initial problem with the Air Quality widget refresh (which might be simple to dix...)

> but let's keep your PR ! (as it fixes a bug with the Airquality location geoloc)

Obviously!

But in any case, **please** re-open this issue.