nabaztag2018/pynab #247: Having trouble making the upgrade through the web interface

omevel, 2021-05-24:
With a clean v0.8.4 distribution : 

- git checkout release
- git show (to get the latest SHA)
- git checkout releng
- git reset --hard SHA

I get "Vous avez des commits locaux non publiés" on the web interface.

git status ->
On branch releng
Your branch and 'origin/releng' have diverged,
and have 3 and 110 different commits each, respectively.
  (use "git pull" to merge the remote branch into yours)

[image]

f-laurens, 2021-05-24:
Not a bug:
what you get here is the expected result from your `git reset --hard SHA` on `releng` (see:https://git-scm.com/docs/git-reset), which resets `releng` to the state of `release`.

omevel, 2021-05-24:
I followed the procedure described here by @pguyot to test upgrade through the web interface: https://github.com/nabaztag2018/pynab/wiki/Processus-de-release and don't remember I had the same issue. 

Can you upgrade from the previous release through the web interface on your side ?

mehdichaouch, 2021-05-24:
This allow to see [the difference](https://github.com/nabaztag2018/pynab/compare/master...release), these three commits missing in master.
The question is why those commits are not on `master`?

f-laurens, 2021-05-24:
> The question is why those commits are not on `master`?

Or rather (since the contents of `master` are presumably correct) why does `release` contain these extra 'garbage' commits that break the release process?
Looks like the last (`v0.8.4`) release was not quite done in accordance to the documented process...

omevel, 2021-05-25:
I may have commited directly on releng by mistake, can't remember.
How do we get out of this mess ? Shall we merge the 3 commits into master ?

omevel, 2021-05-25:
Le problem is that this release is the one everyone has, it's the v0.8.4 release.
And don't worry, i'm not going to do anything with Git without Paul's approval :-)

mehdichaouch, 2021-05-25:
@omevel I can also help with GIT.
In your case you just have to checkout `release`, run this 3 times `git reset HEAD^1 --hard` then `git push origin release -f`.

f-laurens, 2021-05-25:
> `git reset HEAD~3 --hard`

beware that the 3 garbage commits are **not** the last 3 ones on `release`...

mehdichaouch, 2021-05-25:
:warning: commit `ea6e79d` is the one of release `v0.8.4`. The tag was set on `release` branch and it's not on `master`.
I recommandation is to merge `release` in `master` to keep this tag, moreover this merge will have no effect because [the difference](https://github.com/nabaztag2018/pynab/compare/master...release)
> Showing 0 changed files with 0 additions and 0 deletions.

f-laurens, 2021-05-25:
> moreover this merge will have no effect

You are right!
I didn't notice the difference was in fact empty.
No problem with the garbage then if the (garbage) bags are empty... And not need to fool around with the `v0.8.4` tag.
So the merge should indeed be safe. Better to experiment on local branches though, before pushing to the origin repo.

omevel, 2021-05-25:
So the idea is to do the following things if I understand correctly : 

from a clean v0.8.4 release
- git checkout master
- git merge release
- git push
- git checkout releng
- git merge master
- git push

And then pray and 😰 ? :-) ping @mehdichaouch and @pguyot !

f-laurens, 2021-05-25:
Yes, now that the mess has bien cleaned up in the repository thanks to @mehdichaouch:
- `release` purely behind `master`
- `releng` even with `master`

you should be able to proceed normally with the test according to https://github.com/nabaztag2018/pynab/wiki/Processus-de-release .
By the way, maybe you should remove the **bug** label from this issue (not really a Pynab bug!).

omevel, 2021-05-25:
For the record, here's what I did exactly (thanks @pguyot ! thanks @mehdichaouch !)

- git fetch origin
- git checkout origin/master
- git merge origin/release
- git push origin HEAD:master
- git push origin HEAD:releng

Let's test this new releng image :)

omevel, 2021-05-25:
Hum. Ok. I've flashed a SD card with the new releng image. 

- it seems that there's only the releng branch on it : 

```
pi@nabaztag:~/pynab $ git checkout release
error: pathspec 'release' did not match any file(s) known to git

```
- when I try to reset to the last commit of the current release (to test the upgrade), I get that : 

```
pi@nabaztag:~/pynab $ git reset --hard ea6e79d3ea8d8a9e92e6d37707a1cf26c900ee25
fatal: Could not parse object 'ea6e79d3ea8d8a9e92e6d37707a1cf26c900ee25'.
```

We're not done yet but almost there :)

mehdichaouch, 2021-05-25:
@omevel there was more simple, open [diff link](https://github.com/nabaztag2018/pynab/compare/master...release)
then set the branches [image]
then [image]
No code, direct. After that just checkout & pull in local 😉 

What you have done was also correct 👏 👍

omevel, 2021-05-25:
yes but how can I test the upgrade through the web interface ?

closed 2021-05-26 by omevel