nabaztag2018/pynab #253: Nightly Builder workflow should not fail when there are no old builds to remove

f-laurens, 2021-05-27:
(see https://github.com/nabaztag2018/pynab/runs/2680683144?check_suite_focus=true)
The case where there are less than _keep_ existing builds should be handled properly.

f-laurens, 2021-05-29:
Also the `hub release` command used in the 'remove' step does not appear to work correctly in a clone  repository:
it does not operate on the releases in the origin (clone) repository, but on the ones in the upstream (nabaztag2018) repository.

mehdichaouch, 2021-05-31:
> Also the `hub release` command used in the 'remove' step does not appear to work correctly in a clone repository:
> it does not operate on the releases in the origin (clone) repository, but on the ones in the upstream (nabaztag2018) repository.

If I refer to your fork, the delete [worked](https://github.com/f-laurens/pynab/runs/2698939662?check_suite_focus=true#step:3:8) ?

f-laurens, 2021-05-31:
> If I refer to your fork, the delete [worked](https://github.com/f-laurens/pynab/runs/2698939662?check_suite_focus=true#step:3:8) ?
The behaviour of the `hub release` command seems weird: it looks like `hub release `(list) operates on the upstream repository, but `hub release delete` operates on the origin repository.

mehdichaouch, 2021-06-01:
> Have you tested the behavior in your own fork?

Yes and it worked well.

To test, you can create some fake releases with the name pattern: `nightly-` and see tomorrow.

f-laurens, 2021-06-12:
This issue may be closed since both issues (error when no old builds to remove, and non removal of associated tags) are now fixed in PR #271.

PR #271 (2021-06-21): Use raspios_lite:2021-05-07 base image
Resolves #258
Resolves #253.

- Updates GitHub workflows to use **raspios_lite:2021-05-07** base image.
  **Note**: this **depends** on update of GitHub action: https://github.com/pguyot/arm-runner-action/pull/2.
- ~~Also uses `nabooot.service` to ensure `/var/log/nginx `directory exists before `nginx.service` is started.
  This allows to reintroduce -p option to PiShrink when building images.~~
closed 2021-06-12 by f-laurens