nabaztag2018/pynab #256: Regression : raspberry pi zero yellow led is blinking

SonnyT72, 2021-05-28:
**Version of pynab:** v0.9.0 (updated from previous available release)

**Version of Nabaztag:**

- [ ] Nabaztag (v1)
- [X] Nabaztag:Tag (v2)

**Version of TagTagTag:**

- [ ] Maker Faire 2018
- [X] Ulule 05/2019

**The problem**
Regression : the raspberry pi zero internal yellow led, disabled in a previous release of pynab, is back to life on my rabbit following 0.9.0 pynab update.
Update was succesfully done through the web interface.

**Expected behavior**
The led shall be shut off after startup.

f-laurens, 2021-05-28:
What exact version of the Raspberry Pi kernel are you running? (see **Version** information on the **System Information** page).

SonnyT72, 2021-05-28:
I am running     Raspbian GNU/Linux 10 (buster) - Kernel 4.19.118+ armv6l

f-laurens, 2021-05-28:
> Kernel 4.19.118+ armv6l

This is not really a Pynab bug, but rather a Raspberry Pi kernel related glitch: the logic to toggle the Pi Zero LED on/off was inverted in kernel 5.x :-(
The solution for you is therefore to update your kernel from 4.x to 5.x:
- either through the  apt update / apt upgrade mechanism;
- or by reflashing your SD card with the pynab-v0.9.0 image: https://github.com/nabaztag2018/pynab/releases/download/v0.9.0/pynab-v0.9.0.img.xz

SonnyT72, 2021-05-28:
Thank you for your answer! 
I will try the first solution now and give the result here for others to know.
If I don't succeed I will then switch to your second solution.

Thanks again to you and the team working hard to bring our rabbits back to life!

f-laurens, 2021-05-28:
In the meanwhile, can you remove the **bug** label from this issue?

SonnyT72, 2021-05-28:
If I find how to remove it, I will :)

mehdichaouch, 2021-05-28:
> In the meanwhile, can you remove the **bug** label from this issue?

As I know labels can be managed only by @nabaztag2018

pguyot, 2021-05-29:
I beg to differ: this is a bug of pynab. We released 4.x images and allowed @SonnyT72 and others to upgrade our code without upgrading the kernel. We definitely
should check kernel version and handle the led accordingly.

f-laurens, 2021-05-29:
> We definitely should check kernel version and handle the led accordingly.

Indeed; PR #158 for Issue #156 should not have been marged as-is.

PR #257 (2021-05-31): Turn off Pi Zero system LED according to kernel version
Resolves #256.
closed 2021-05-31 by omevel