nabaztag2018/pynab #258: Update GitHub workflows that build/test image to use latest Raspberry Pi OS Lite base image

f-laurens, 2021-05-30:
See https://github.com/pguyot/arm-runner-action/issues/1.

mehdichaouch, 2021-05-31:
Nice idea, but one thing to see before changing is the final size of the pynab image, because GitHub [release max size is 2.0 GB](https://docs.github.com/en/github/administering-a-repository/releasing-projects-on-github/about-releases#storage-and-bandwidth-quotas).

f-laurens, 2021-06-04:
> because GitHub release max size is 2.0 GB

This is something we will have to continue coping with anyway...

> but one thing to see before changing is the final size of the pynab image

Well, the only way to know is to try, and check **after** changing if it still fits the 2.0GB limit...

Another benefit for us of using the latest base image is that it would somewhat reduce build times when creating the Pynab image from this base image, since the `apt update/install ` steps would have less packages to upgrade and run faster.

PR #271 (2021-06-21): Use raspios_lite:2021-05-07 base image
Resolves #258
Resolves #253.

- Updates GitHub workflows to use **raspios_lite:2021-05-07** base image.
  **Note**: this **depends** on update of GitHub action: https://github.com/pguyot/arm-runner-action/pull/2.
- ~~Also uses `nabooot.service` to ensure `/var/log/nginx `directory exists before `nginx.service` is started.
  This allows to reintroduce -p option to PiShrink when building images.~~
closed 2021-06-21 by pguyot