nabaztag2018/pynab #267: waqi.info feed seems wrong for Paris

omevel, 2021-06-03:
Wether using the geoloc from the weather forecast or the waqi IP based (http://api.waqi.info/feed/here/?token=4cf7f445134f3fb69a4c3f0e5001e507a6cc386f) the airquality is given for : 

**Faiencerie Creil, Picardie, France**

Data are available though : http://aqicn.org/city/paris/

Probably a bug with aqicn API ....

f-laurens, 2021-06-03:
> Probably a bug with aqicn API ....

Indeed. The selection of the "closest" station (from the given IP or given lat/lon) by the AQICN API seems sometimes fuzzy.
Maybe we are a bit too far from Beijing ;-)

I dont think we can do much about this, except try to find another provider for air quality info.

f-laurens, 2021-07-17:
After monitoring this for a couple of months, it seems that station selection by the AQICN API is most of the time reasonably right (i.e. getting `Faiencerie Creil, Picardie, France` when living in/near Paris is the exception rather than the norm).

I therefore suggest closing this issue, or at least removing its bug label.

closed 2022-03-14 by nabaztag2018