nabaztag2018/pynab #282: Error setting up the Docker-based environment

f-laurens, 2021-06-14:
I get the error below when setting up the Docker-based virtual Pynab environment, during the `docker-compose` build step.

Note: host system is a Raspberry Pi 4 running Raspbian, with the following Docker packages:
- docker-ce                            5:20.10.7\~3-0~raspbian-buster
- docker-compose                       1.21.0-3

f-laurens, 2021-06-14:
```
Step 19/24 : RUN ${VIRTUAL_ENV}/bin/pip install -r /tmp/requirements_docker.txt
 ---> Running in fc2a24116ce8
Ignoring rpi-ws281x: markers 'sys_platform == "linux" and "armv6l" in platform_machine' don't match your environment
Ignoring RPi.GPIO: markers 'sys_platform == "linux" and "armv6l" in platform_machine' don't match your environment
Ignoring snips-nlu-parsers: markers 'sys_platform == "linux" and "armv6l" in platform_machine and python_version == "3.7"' don't match your environment
Ignoring snips-nlu-utils: markers 'sys_platform == "linux" and "armv6l" in platform_machine and python_version == "3.7"' don't match your environment
Collecting wheel
  Downloading wheel-0.36.2-py2.py3-none-any.whl (35 kB)
Collecting black==21.4b2
  Downloading black-21.4b2-py3-none-any.whl (130 kB)
Collecting Django==3.0.14
  Downloading Django-3.0.14-py3-none-any.whl (7.5 MB)
Collecting flake8==3.9.1
  Downloading flake8-3.9.1-py2.py3-none-any.whl (73 kB)
Collecting gunicorn==20.0.4
  Downloading gunicorn-20.0.4-py2.py3-none-any.whl (77 kB)
Collecting isort==5.8.0
  Downloading isort-5.8.0-py3-none-any.whl (103 kB)
Collecting lockfile==0.12.2
  Downloading lockfile-0.12.2-py2.py3-none-any.whl (13 kB)
Collecting Mastodon.py==1.5.0
  Downloading Mastodon.py-1.5.0-py2.py3-none-any.whl (35 kB)
Collecting mpg123==0.4
  Downloading mpg123-0.4.tar.gz (2.2 kB)
Collecting psycopg2-binary==2.8.4
  Downloading psycopg2-binary-2.8.4.tar.gz (378 kB)
Collecting pytest-django==3.7.0
  Downloading pytest_django-3.7.0-py2.py3-none-any.whl (18 kB)
Collecting pytest-pep8==1.0.6
  Downloading pytest-pep8-1.0.6.tar.gz (7.3 kB)
Collecting pytest==5.3.1
  Downloading pytest-5.3.1-py3-none-any.whl (233 kB)
Collecting asynctest==0.13.0
  Downloading asynctest-0.13.0-py3-none-any.whl (26 kB)
Collecting coverage==5.0.3
  Downloading coverage-5.0.3.tar.gz (679 kB)
Collecting pre_commit==2.12.1
  Downloading pre_commit-2.12.1-py2.py3-none-any.whl (189 kB)
Collecting pyalsaaudio==0.8.4
  Downloading pyalsaaudio-0.8.4.tar.gz (315 kB)
Collecting snips-nlu==0.20.2
  Downloading https://github.com/pguyot/snips-nlu/releases/download/0.20.2/snips_nlu-0.20.2-py3-none-any.whl (135 kB)
Collecting snips-nlu-parsers==0.4.3
  Downloading snips_nlu_parsers-0.4.3.tar.gz (6.9 kB)
    ERROR: Command errored out with exit status 1:
     command: /home/pi/venv/bin/python3.7 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-nvehw40h/snips-nlu-parsers/setup.py'"'"'; __file__='"'"'/tmp/pip-install-nvehw40h/snips-nlu-parsers/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-uq_unzy1
         cwd: /tmp/pip-install-nvehw40h/snips-nlu-parsers/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-nvehw40h/snips-nlu-parsers/setup.py", line 8, in <module>
        from setuptools_rust import Binding, RustExtension
    ModuleNotFoundError: No module named 'setuptools_rust'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 20.1.1; however, version 21.1.2 is available.
You should consider upgrading via the '/home/pi/venv/bin/python3.7 -m pip install --upgrade pip' command.
ERROR: Service 'migrate' failed to build: The command '/bin/sh -c ${VIRTUAL_ENV}/bin/pip install -r /tmp/requirements_docker.txt' returned a non-zero code: 1
```

f-laurens, 2021-06-25:
Any hint about this issue?
Also, shouldn't `Docker/nab/requirements.txt` be identical (symlinked?) to `requirements.txt`?

nguillaumin, 2021-06-27:
It works on x86, so it's something to do with running it on the Raspberry Pi. Docker on Raspberry use ARM images, there's something in the ARM environment or Debian image that is missing, to do with Rust apparently. Maybe there are pre-compiled packages for NLU Parsers on x86, but not for ARM, so it tries to compile them but fail due to a missing dependency? I don't have a RP4 so I can't easily check unfortunately.

For the `requirements.txt`, symlinks are not supported by Docker I believe (it will still see the file is "outside" of the build context:

```
$ ls -la nab/requirements.txt 
lrwxrwxrwx 1 nicolas users 22 Jun 27 11:41 nab/requirements.txt -> ../../requirements.txt
$ docker-compose build
...
Step 17/24 : COPY requirements.txt /tmp/requirements.txt
COPY failed: forbidden path outside the build context: requirements.txt ()
ERROR: Service 'migrate' failed to build : Build failed
```

Plus it wouldn't work on Windows...

f-laurens, 2021-06-27:
Thanks a lot for your answer.
It might be useful to mention in the `Docker/README.md` file that is works only on X86.

> For the requirements.txt, symlinks are not supported by Docker I believe (it will still see the file is "outside" of the build context)

Indeed. Then care should be taken to keep the `Docker/nab/requirements.txt` up to date wrt `requirements.txt`.

pguyot, 2021-06-28:
My understanding is the docker file should not be limited to x86. It makes most sense for platforms where binaries packages are available. It also makes most sense on desktop platforms as the whole point is to develop services.

snips-nlu-parsers and snips-nlu-utils are apparently not available as binaries for the Pi4, including on piwheels
https://www.piwheels.org/project/snips-nlu-utils/
https://www.piwheels.org/project/snips-nlu-parsers/

They do not seem to be available on macOS-aarch64 either as their latest release predates the release of this platform.
https://pypi.org/project/snips-nlu-parsers/#files
https://pypi.org/project/snips-nlu-utils/#files

In your case, pip tries to compile these packages and fails because we are missing a build dependency on setuptools-rust which we may add conditionally (ideally making sure we only install it when building snips packages from source).

Another workaround consists in providing binaries for these packages. Compiling them is quite a pain, especially for the Raspberry Pi zero. A native build on a Pi3 with Rust is doable, and I guess the same applies for Pi4. If you want to go the cross build route, you might probably want to start from these GitHub action files that I used for Raspberry Pi zero binaries:
https://github.com/pguyot/snips-nlu-parsers/blob/develop/.github/workflows/crossenv.yml
https://github.com/pguyot/snips-nlu-utils/blob/develop/.github/workflows/crossenv.yml

f-laurens, 2021-06-28:
> If you want to go the cross build route, you might probably want to start from these GitHub action files that I used for Raspberry Pi zero binaries:

Thanks for the tip. I'll give it a look.

f-laurens, 2021-06-29:
This allows the `docker-compose` build to progress further, but it still fails with the error below.
Any hint?

f-laurens, 2021-06-29:
```
Collecting scipy==1.6.2
  Downloading scipy-1.6.2.tar.gz (27.2 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /home/pi/venv/bin/python3.7 /home/pi/venv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpsu2p4_l6
         cwd: /tmp/pip-install-wwg4afe5/scipy
    Complete output (159 lines):
    lapack_opt_info:
    lapack_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/home/pi/venv/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
      NOT AVAILABLE

    openblas_lapack_info:
    customize UnixCCompiler
    customize UnixCCompiler
      libraries openblas not found in ['/home/pi/venv/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
      NOT AVAILABLE

    openblas_clapack_info:
    customize UnixCCompiler
    customize UnixCCompiler
      libraries openblas,lapack not found in ['/home/pi/venv/lib', '/usr/local/lib', '/usr/lib', '/usr/lib/arm-linux-gnueabihf']
      NOT AVAILABLE

    atlas_3_10_threads_info:
    Setting PTATLAS=ATLAS
    customize UnixCCompiler
      libraries lapack_atlas not found in /home/pi/venv/lib
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /home/pi/venv/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
    customize UnixCCompiler
      libraries tatlas,tatlas not found in /usr/lib/arm-linux-gnueabihf
    <class 'numpy.distutils.system_info.atlas_3_10_threads_info'>
      NOT AVAILABLE

    atlas_3_10_info:
    customize UnixCCompiler
      libraries lapack_atlas not found in /home/pi/venv/lib
    customize UnixCCompiler
      libraries satlas,satlas not found in /home/pi/venv/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/local/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/lib
    customize UnixCCompiler
      libraries lapack_atlas not found in /usr/lib/arm-linux-gnueabihf
    customize UnixCCompiler
      libraries satlas,satlas not found in /usr/lib/arm-linux-gnueabihf
    <class 'numpy.distutils.system_info.atlas_3_10_info'>
[... 83 lines cut ...]
        return hook(metadata_directory, config_settings)
      File "/tmp/pip-build-env-niouf509/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 161, in prepare_metadata_for_build_wheel
        self.run_setup()
      File "/tmp/pip-build-env-niouf509/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 254, in run_setup
        self).run_setup(setup_script=setup_script)
      File "/tmp/pip-build-env-niouf509/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 145, in run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 598, in <module>
        setup_package()
      File "setup.py", line 594, in setup_package
        setup(**metadata)
      File "/tmp/pip-build-env-niouf509/overlay/lib/python3.7/site-packages/numpy/distutils/core.py", line 137, in setup
        config = configuration()
      File "setup.py", line 500, in configuration
        raise NotFoundError(msg)
    numpy.distutils.system_info.NotFoundError: No BLAS/LAPACK libraries found.
    To build Scipy from sources, BLAS & LAPACK libraries need to be installed.
    See site.cfg.example in the Scipy source directory and
    https://docs.scipy.org/doc/scipy/reference/building/index.html for details.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/venv/bin/python3.7 /home/pi/venv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpsu2p4_l6 Check the logs for full command output.
WARNING: You are using pip version 20.1.1; however, version 21.1.3 is available.
You should consider upgrading via the '/home/pi/venv/bin/python3.7 -m pip install --upgrade pip' command.
ERROR: Service 'migrate' failed to build: The command '/bin/sh -c ${VIRTUAL_ENV}/bin/pip install -r /tmp/requirements_docker.txt' returned a non-zero code: 1
```

pguyot, 2021-06-29:
scipy 1.6.2 is available as a binary wheel from piwheels, I wonder why it misses it.

https://www.piwheels.org/project/scipy/

Is pip trying from piwheels when building a Docker image?

f-laurens, 2021-06-30:
> Is pip trying from piwheels when building a Docker image?

This is weird indeed. I tried a couple of times and this is deterministic: not a network glitch or something of the kind.

pguyot, 2021-06-30:
Could you please try adding verbose option `-v` to `pip install` and post here the full output?

f-laurens, 2021-06-30:
Here it is: [Pi4_Docker_pip_install.zip](https://github.com/nabaztag2018/pynab/files/6738489/Pi4_Docker_pip_install.zip).

pguyot, 2021-06-30:
You need to enable piwheels, for example by editing pip.conf, to get binaries for the Raspberry Pi4.
See: https://www.piwheels.org

f-laurens, 2021-06-30:
Indeed, adding in `Docker/nab/Dockerfile` the line:
  `RUN /bin/echo -e "[global]\nextra-index-url=https://www.piwheels.org/simple\n" > /etc/pip.conf`
before the line:
  `RUN ${VIRTUAL_ENV}/bin/pip install -r /tmp/requirements_docker.txt`
does the trick. The build then goes smoothly on Pi 4.

Would you consider adding this in the official `Docker/nab/Dockerfile`? I suppose it wouldn't hurt on X86.

f-laurens, 2021-06-30:
Now I just have to understand why PostgreSQL crashes on statup....:
```
pi@Pi-4:/srv/pynab $ docker-compose logs -f
Attaching to pynab_nab8balld_1, pynab_nabmastodond_1, pynab_nabclockd_1, pynab_nabweatherd_1, pynab_nabairqualityd_1, pynab_nabweb_1, pynab_nabtaichid_1, pynab_nabbookd_1, pynab_nabsurprised_1, pynab_migrate_1, pynab_nabd_1, pynab_db_1
nabmastodond_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nabmastodond_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nabmastodond_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nabmastodond_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nabmastodond_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
nab8balld_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nab8balld_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nab8balld_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nab8balld_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nab8balld_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
pynab_nabmastodond_1 exited with code 1
nabweatherd_1     | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nabweatherd_1     | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nabweatherd_1     | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nabweatherd_1     | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nabweatherd_1     | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
pynab_nab8balld_1 exited with code 1
nabclockd_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nabclockd_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nabclockd_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nabclockd_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nabclockd_1       | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
nabairqualityd_1  | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nabairqualityd_1  | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nabairqualityd_1  | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nabairqualityd_1  | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nabairqualityd_1  | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
pynab_nabweatherd_1 exited with code 1
nabbookd_1        | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nabbookd_1        | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nabbookd_1        | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nabbookd_1        | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nabbookd_1        | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
pynab_nabairqualityd_1 exited with code 1
pynab_nabbookd_1 exited with code 1
pynab_nabclockd_1 exited with code 1
nabsurprised_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nabsurprised_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nabsurprised_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nabsurprised_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nabsurprised_1    | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
pynab_nabsurprised_1 exited with code 1
nabtaichid_1      | Waiting for /var/run/postgresql/migrate.completed to appear, 5 remaining attempts...
nabtaichid_1      | Waiting for /var/run/postgresql/migrate.completed to appear, 4 remaining attempts...
nabtaichid_1      | Waiting for /var/run/postgresql/migrate.completed to appear, 3 remaining attempts...
nabtaichid_1      | Waiting for /var/run/postgresql/migrate.completed to appear, 2 remaining attempts...
nabtaichid_1      | Waiting for /var/run/postgresql/migrate.completed to appear, 1 remaining attempts...
db_1              | The files belonging to this database system will be owned by user "postgres".
db_1              | This user must also own the server process.
db_1              |
db_1              | The database cluster will be initialized with locale "en_US.utf8".
db_1              | The default database encoding has accordingly been set to "UTF8".
db_1              | The default text search configuration will be set to "english".
db_1              |
db_1              | Data page checksums are disabled.
db_1              |
db_1              | fixing permissions on existing directory /var/lib/postgresql/data ... ok
db_1              | creating subdirectories ... ok
[... cut ...]
nabweb_1          | [2021-06-30 09:06:31 +0000] [30] [INFO] Worker exiting (pid: 30)
nabweb_1          | [2021-06-30 09:06:31 +0000] [1] [INFO] Shutting down: Master
nabweb_1          | [2021-06-30 09:06:31 +0000] [1] [INFO] Reason: Worker failed to boot.
nabd_1            | /home/pi/venv/bin/python3: Error while finding module specification for 'nabd.nabd' (ModuleNotFoundError: No module named 'nabd')
pynab_nabtaichid_1 exited with code 1
pynab_db_1 exited with code 1
pynab_nabd_1 exited with code 1
pynab_migrate_1 exited with code 2
pynab_nabweb_1 exited with code 3
``

pguyot, 2021-07-02:
It might be related to issue I encountered with qemu chroot. Can you try somehow to restrict postgreSQL to a single core? For example can you limit docker to a single core with `docker run --cpuset-cpus="0"`?

f-laurens, 2021-07-02:
These docker / docker-compose options are a bit of a mess...
Since I use `docker-compose`, I don't know how to set the `docker run` options, so I tried fiddling with the Compose file:
to use these options in the Compose file, it has to be downgraded to version 2.4 :-(, and then:

- `cpu_count: 1` has no effect (it's apparently a X86-only option anyway).
- `cpuset: "0"` has no effect neither.

Same SIGSEGV when starting PostgreSQL.

f-laurens, 2021-07-04:
Fixing `nabd` so that NabIO is non-virtual only on armv6l apparently solves the `problem` with nabd/nabweb startup.
The only remaing problem seems to be the PostgreSQL startup crash.

I have gathered my fixes so far in PR #286.

f-laurens, 2021-07-04:
Now what is left is to understand is what happens with nabweb:
```
nabweb_1          | [2021-07-04 16:07:40 +0200] [1] [INFO] Starting gunicorn 20.0.4
nabweb_1          | [2021-07-04 16:07:40 +0200] [1] [INFO] Listening at: http://0.0.0.0:8000 (1)
nabweb_1          | [2021-07-04 16:07:40 +0200] [1] [INFO] Using worker: sync
nabweb_1          | [2021-07-04 16:07:40 +0200] [31] [INFO] Booting worker with pid: 31
nabweb_1          | [2021-07-04 16:10:34 +0200] [1] [CRITICAL] WORKER TIMEOUT (pid:31)
nabweb_1          | [2021-07-04 14:10:34 +0000] [31] [INFO] Worker exiting (pid: 31)
nabweb_1          | [2021-07-04 16:10:34 +0200] [32] [INFO] Booting worker with pid: 32
nabweb_1          | [2021-07-04 16:11:36 +0200] [1] [CRITICAL] WORKER TIMEOUT (pid:32)
nabweb_1          | [2021-07-04 14:11:36 +0000] [32] [INFO] Worker exiting (pid: 32)
nabweb_1          | [2021-07-04 16:11:36 +0200] [33] [INFO] Booting worker with pid: 33
nabweb_1          | [2021-07-04 16:12:37 +0200] [1] [CRITICAL] WORKER TIMEOUT (pid:33)
nabweb_1          | [2021-07-04 14:12:37 +0000] [33] [INFO] Worker exiting (pid: 33)
nabweb_1          | [2021-07-04 16:12:37 +0200] [34] [INFO] Booting worker with pid: 34
nabweb_1          | [2021-07-04 16:13:38 +0200] [1] [CRITICAL] WORKER TIMEOUT (pid:34)
nabweb_1          | [2021-07-04 14:13:38 +0000] [34] [INFO] Worker exiting (pid: 34)
nabweb_1          | [2021-07-04 16:13:39 +0200] [35] [INFO] Booting worker with pid: 35
nabweb_1          | [2021-07-04 16:14:40 +0200] [1] [CRITICAL] WORKER TIMEOUT (pid:35)
nabweb_1          | [2021-07-04 14:14:40 +0000] [35] [INFO] Worker exiting (pid: 35)
nabweb_1          | [2021-07-04 16:14:40 +0200] [36] [INFO] Booting worker with pid: 36
nabweb_1          | [2021-07-04 16:15:41 +0200] [1] [CRITICAL] WORKER TIMEOUT (pid:36)
nabweb_1          | [2021-07-04 14:15:41 +0000] [36] [INFO] Worker exiting (pid: 36)
nabweb_1          | [2021-07-04 16:15:41 +0200] [37] [INFO] Booting worker with pid: 37
```

f-laurens, 2022-02-27:
Residual problem with gunicorn/nabweb is now solved on PI 4 platform,

- using latest version of Docker: `5:20.10.12~3-0~raspbian-buster`
- and also probably by using fixed version of` libseccomp2` library (see https://docs.linuxserver.io/faq#libseccomp)

closed 2022-02-27 by f-laurens