nabaztag2018/pynab #287: Hourly announcement for 16:00 is played at 17:00

srcshelton, 2021-07-05:
**Version of pynab:** v0.9.1

**Version of Nabaztag:**

- [ ] Nabaztag (v1)
- [ ] Nabaztag:Tag (v2)

**Version of TagTagTag:**

- [ ] Maker Faire 2018
- [ ] Ulule 05/2019

**The problem**

Since upgrading from v0.8.4 to v0.9.1, I’ve noticed that the hourly announcement played at 17:00 is the 16:00 one.  Other hours (certainly 18:00) seem to be correct (or, at least, I’ve not noticed any issue).

**Expected behavior**

17:00 announcement played at 17:00, instead of 16:00 announcement.

srcshelton, 2021-07-05:
(I assumed my time-zone was wrong - but all other hourly announcements do seem to be correct!)

mrcroupotin, 2021-07-05:
In which language speak your rabbit?

f-laurens, 2021-07-05:
And what is your timezone?

pguyot, 2021-07-05:
Well spotted!
I am wondering if this file was wrong with the original server code. Anyway, I don't think it is new to release 0.9.1.

commit e4d0f838 (2021-07-06): Fix location of 'It's only 4 o-clock, what a very very long day' (fix #287)
closed 2021-07-06 by pguyot