nabaztag2018/pynab #315: RFID tag handling broken on Pi Zero 2

f-laurens, 2022-03-05:
Reading or writing a tag doesn't seems to work on Pi Zero 2:

- `nabd` gets stuck (no visible exceptions in `/var/log/nabd.log`) and cannot be stopped .
- Eventually the whole system gets stuck and has to be power-cycled.

This looks like a a low-level problem with the `cr14` driver.

pguyot, 2022-03-05:
If there is a problem with cr14 driver, there might be some log in dmesg or kernel.log. Did you get anything?

Also you can try and stop nab*d and interact directly with the cr14 driver by opening /dev/rfid0 or by using [example scripts](https://github.com/pguyot/cr14/tree/master/examples).

f-laurens, 2022-03-05:
> Also you can try and stop nab*d and interact directly with the cr14 driver by opening /dev/rfid0 or by using [example scripts](https://github.com/pguyot/cr14/tree/master/examples).

This was to be my next step.

> If there is a problem with cr14 driver, there might be some log in dmesg or kernel.log. Did you get anything?

Not easy to get , since the whole system hangs.
I have  the following in `dmesg`, but it appears (without apparently causing any harm?) even if `cr14` driver is not used:
```
[11012.810983] WARNING: CPU: 0 PID: 0 at kernel/time/timer.c:1372 del_timer_sync+0x50/0x64
[11012.810991] Modules linked in: snd_soc_wm8960(O) regmap_i2c cr14(O) brcmfmac brcmutil sha256_generic cfg80211 rfkill snd_soc_simple_card tagtagtag_ears(O) snd_soc_simple_card_utils snd_soc_volume_gpio(O) raspberrypi_hwmon i2c_bcm2835 snd_soc_bcm2835_i2s snd_soc_max9759(O) snd_soc_pcm5102a snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd uio_pdrv_genirq uio fixed zram zsmalloc fuse ip_tables x_tables ipv6
[11012.811205] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W  O      5.10.92-v7+ #1514
[11012.811213] Hardware name: BCM2835
[11012.811221] Backtrace:
[11012.811249] [<809f4a58>] (dump_backtrace) from [<809f4de8>] (show_stack+0x20/0x24)
[11012.811265]  r7:ffffffff r6:00000000 r5:60000193 r4:80fe5ef0
[11012.811283] [<809f4dc8>] (show_stack) from [<809f8ffc>] (dump_stack+0xcc/0xf8)
[11012.811303] [<809f8f30>] (dump_stack) from [<8011eea0>] (__warn+0xfc/0x114)
[11012.811320]  r10:80f01da8 r9:00000009 r8:801a5e6c r7:0000055c r6:00000009 r5:801a5e6c
[11012.811331]  r4:80d02910 r3:80f05050
[11012.811349] [<8011eda4>] (__warn) from [<809f5438>] (warn_slowpath_fmt+0x70/0xd8)
[11012.811363]  r7:0000055c r6:80d02910 r5:80f05008 r4:00000000
[11012.811381] [<809f53cc>] (warn_slowpath_fmt) from [<801a5e6c>] (del_timer_sync+0x50/0x64)
[11012.811397]  r9:ffffe000 r8:000000c7 r7:82796200 r6:82fff890 r5:00000000 r4:82fff890
[11012.811423] [<801a5e1c>] (del_timer_sync) from [<7f0d5ff4>] (tagtagtagear_irq_handler+0xe4/0x690 [tagtagtag_ears])
[11012.811434]  r5:00000000 r4:82fff848
[11012.811459] [<7f0d5f10>] (tagtagtagear_irq_handler [tagtagtag_ears]) from [<80185fc0>] (__handle_irq_event_percpu+0xa8/0x268)
[11012.811476]  r9:ffffe000 r8:000000c7 r7:82796200 r6:00000000 r5:8279626c r4:82772cc0
[11012.811496] [<80185f18>] (__handle_irq_event_percpu) from [<801861bc>] (handle_irq_event_percpu+0x3c/0x90)
[11012.811513]  r10:80cfcf84 r9:80f00000 r8:814de000 r7:80f05008 r6:82796200 r5:8279626c
[11012.811522]  r4:80f05008
[11012.811542] [<80186180>] (handle_irq_event_percpu) from [<80186264>] (handle_irq_event+0x54/0x78)
[11012.811554]  r6:8170bc40 r5:8279626c r4:82796200
[11012.811573] [<80186210>] (handle_irq_event) from [<8018b1a4>] (handle_edge_irq+0xcc/0x1fc)
[11012.811587]  r7:80f05008 r6:8170bc40 r5:00000000 r4:82796200
[11012.811605] [<8018b0d8>] (handle_edge_irq) from [<80184f7c>] (generic_handle_irq+0x44/0x54)
[11012.811616]  r5:00000000 r4:00000018
[11012.811637] [<80184f38>] (generic_handle_irq) from [<80680de8>] (bcm2835_gpio_irq_handle_bank+0x94/0xcc)
[11012.811656] [<80680d54>] (bcm2835_gpio_irq_handle_bank) from [<80680eb0>] (bcm2835_gpio_irq_handler+0x90/0x124)
[11012.811670]  r7:8170bc40 r6:00000000 r5:80fe5fb4 r4:814e9618
[11012.811688] [<80680e20>] (bcm2835_gpio_irq_handler) from [<80184f7c>] (generic_handle_irq+0x44/0x54)
[11012.811701]  r7:00000001 r6:00000000 r5:00000000 r4:80f05aec
[11012.811722] [<80184f38>] (generic_handle_irq) from [<806786bc>] (bcm2836_chained_handle_irq+0x48/0x4c)
[11012.811742] [<80678674>] (bcm2836_chained_handle_irq) from [<80184f7c>] (generic_handle_irq+0x44/0x54)
[11012.811752]  r5:00000000 r4:80e91d90
[11012.811771] [<80184f38>] (generic_handle_irq) from [<801856dc>] (__handle_domain_irq+0x6c/0xc4)
[11012.811791] [<80185670>] (__handle_domain_irq) from [<801012c8>] (bcm2836_arm_irqchip_handle_irq+0x60/0x64)
[11012.811808]  r9:80f00000 r8:00000001 r7:80f01f04 r6:ffffffff r5:60000013 r4:80108874
[11012.811826] [<80101268>] (bcm2836_arm_irqchip_handle_irq) from [<80100abc>] (__irq_svc+0x5c/0x7c)
[11012.811836] Exception stack(0x80f01ed0 to 0x80f01f18)
[11012.811849] 1ec0:                                     00000000 00507278 9ef76344 80119040
[11012.811865] 1ee0: ffffe000 80f05058 80f050a0 00000001 00000001 8103193f 80cfcf84 80f01f2c
[11012.811879] 1f00: 80f01f30 80f01f20 80108870 80108874 60000013 ffffffff
[11012.811901] [<8010882c>] (arch_cpu_idle) from [<80a03958>] (default_idle_call+0x4c/0x118)
[11012.811920] [<80a0390c>] (default_idle_call) from [<8015664c>] (do_idle+0x118/0x168)
[11012.811937] [<80156534>] (do_idle) from [<80156970>] (cpu_startup_entry+0x28/0x30)
[11012.811954]  r10:9efffac0 r9:80e52a58 r8:00000000 r7:80e52a58 r6:80f05000 r5:80f05008
[11012.811965]  r4:000000d9 r3:80e87284
[11012.811981] [<80156948>] (cpu_startup_entry) from [<809fc990>] (rest_init+0xbc/0xc4)
[11012.812000] [<809fc8d4>] (rest_init) from [<80e00aa0>] (arch_call_rest_init+0x18/0x1c)
[11012.812012]  r5:80f05008 r4:81049068
[11012.812030] [<80e00a88>] (arch_call_rest_init) from [<80e01068>] (start_kernel+0x54c/0x580)
[11012.812045] [<80e00b1c>] (start_kernel) from [<00000000>] (0x0)
[11012.812054] ---[ end trace 585c9e0bce35d3c2 ]---

```

pguyot, 2022-03-05:
> I have  the following in `dmesg`, but it appears (without apparently causing any harm?) even if `cr14` driver is not used:
> 

It's apparently ears-related.
Still the DietPi armv7l image?

f-laurens, 2022-03-05:
> It's apparently ears-related.

Ears seem to behave fine, though.
> Still the DietPi armv7l image?

Yes. I didn't have a chance yet to check the behaviour with the `zero2_raspios` image.

f-laurens, 2022-03-05:
I switched from  the `zero2_dietpi` to the `zero2_raspios` image.
Behaviour is the same:
- `nabd` gets stuck (no visible exceptions in `/var/log/nabd.log`) and cannot be stopped .
- Eventually the whole system gets stuck and has to be power-cycled.

Kernel log shows this:
```
Mar  5 21:04:38 Hazel kernel: [ 2295.957165] rcu: INFO: rcu_sched self-detected stall on CPU
Mar  5 21:04:38 Hazel kernel: [ 2295.957196] rcu: 	0-....: (2099 ticks this GP) idle=fb2/1/0x40000002 softirq=23066/23066 fqs=1049
Mar  5 21:04:38 Hazel kernel: [ 2295.957208] 	(t=2100 jiffies g=43757 q=110)
Mar  5 21:04:38 Hazel kernel: [ 2295.957219] NMI backtrace for cpu 0
Mar  5 21:04:38 Hazel kernel: [ 2295.957236] CPU: 0 PID: 4977 Comm: python Tainted: G        WC O      5.10.92-v7+ #1514
Mar  5 21:04:38 Hazel kernel: [ 2295.957244] Hardware name: BCM2835
Mar  5 21:04:38 Hazel kernel: [ 2295.957252] Backtrace:
Mar  5 21:04:38 Hazel kernel: [ 2295.957287] [<809f4a58>] (dump_backtrace) from [<809f4de8>] (show_stack+0x20/0x24)
Mar  5 21:04:38 Hazel kernel: [ 2295.957303]  r7:ffffffff r6:00000000 r5:60000193 r4:80fe5ef0
Mar  5 21:04:38 Hazel kernel: [ 2295.957322] [<809f4dc8>] (show_stack) from [<809f8ffc>] (dump_stack+0xcc/0xf8)
Mar  5 21:04:38 Hazel kernel: [ 2295.957344] [<809f8f30>] (dump_stack) from [<80666e94>] (nmi_cpu_backtrace+0xc8/0xf4)
Mar  5 21:04:38 Hazel kernel: [ 2295.957361]  r10:80000193 r9:80d0120c r8:80b02060 r7:80b02064 r6:00000000 r5:00000000
Mar  5 21:04:38 Hazel kernel: [ 2295.957372]  r4:00000000 r3:9f188448
Mar  5 21:04:38 Hazel kernel: [ 2295.957392] [<80666dcc>] (nmi_cpu_backtrace) from [<80666fe8>] (nmi_trigger_cpumask_backtrace+0x128/0x140)
Mar  5 21:04:38 Hazel kernel: [ 2295.957404]  r5:80f05ad4 r4:8010e094
Mar  5 21:04:38 Hazel kernel: [ 2295.957426] [<80666ec0>] (nmi_trigger_cpumask_backtrace) from [<8010ef8c>] (arch_trigger_cpumask_backtrace+0x20/0x24)
Mar  5 21:04:38 Hazel kernel: [ 2295.957440]  r7:80f76280 r6:80f05060 r5:80f05104 r4:00000000
Mar  5 21:04:38 Hazel kernel: [ 2295.957459] [<8010ef6c>] (arch_trigger_cpumask_backtrace) from [<809f69fc>] (rcu_dump_cpu_stacks+0x10c/0x144)
Mar  5 21:04:38 Hazel kernel: [ 2295.957479] [<809f68f0>] (rcu_dump_cpu_stacks) from [<8019cb7c>] (rcu_sched_clock_irq+0x7c0/0xa38)
Mar  5 21:04:38 Hazel kernel: [ 2295.957496]  r10:80f03d00 r9:1e083000 r8:00000000 r7:80e932c0 r6:80e932c0 r5:9ef162c0
Mar  5 21:04:38 Hazel kernel: [ 2295.957505]  r4:80f76280
Mar  5 21:04:38 Hazel kernel: [ 2295.957525] [<8019c3bc>] (rcu_sched_clock_irq) from [<801a85ac>] (update_process_times+0x70/0x9c)
Mar  5 21:04:38 Hazel kernel: [ 2295.957542]  r10:801bc3c8 r9:9ef0f640 r8:ffffe000 r7:00000216 r6:3e8e1124 r5:00000000
Mar  5 21:04:38 Hazel kernel: [ 2295.957551]  r4:ffffe000
Mar  5 21:04:38 Hazel kernel: [ 2295.957569] [<801a853c>] (update_process_times) from [<801bbcb4>] (tick_sched_handle+0x64/0x70)
Mar  5 21:04:38 Hazel kernel: [ 2295.957584]  r7:00000216 r6:91e1623b r5:83013e10 r4:9ef0f978
Mar  5 21:04:38 Hazel kernel: [ 2295.957599] [<801bbc50>] (tick_sched_handle) from [<801bc424>] (tick_sched_timer+0x5c/0xb8)
Mar  5 21:04:38 Hazel kernel: [ 2295.957616] [<801bc3c8>] (tick_sched_timer) from [<801a9104>] (__hrtimer_run_queues+0xe8/0x384)
Mar  5 21:04:38 Hazel kernel: [ 2295.957630]  r7:9ef0f680 r6:80f05058 r5:9ef0f640 r4:9ef0f978
Mar  5 21:04:38 Hazel kernel: [ 2295.957649] [<801a901c>] (__hrtimer_run_queues) from [<801aa040>] (hrtimer_interrupt+0x130/0x2b8)
Mar  5 21:04:38 Hazel kernel: [ 2295.957666]  r10:9ef0f790 r9:9ef0f7c0 r8:ffffffff r7:7fffffff r6:00000003 r5:20000193
Mar  5 21:04:38 Hazel kernel: [ 2295.957674]  r4:9ef0f640
Mar  5 21:04:38 Hazel kernel: [ 2295.957695] [<801a9f10>] (hrtimer_interrupt) from [<808303fc>] (arch_timer_handler_phys+0x38/0x40)
Mar  5 21:04:38 Hazel kernel: [ 2295.957712]  r10:7f3172a8 r9:83012000 r8:814de000 r7:000000c3 r6:814e19c0 r5:80fe6284
Mar  5 21:04:38 Hazel kernel: [ 2295.957720]  r4:814eef00
Mar  5 21:04:38 Hazel kernel: [ 2295.957739] [<808303c4>] (arch_timer_handler_phys) from [<8018bd50>] (handle_percpu_devid_irq+0x8c/0x21c)
Mar  5 21:04:38 Hazel kernel: [ 2295.957758] [<8018bcc4>] (handle_percpu_devid_irq) from [<80184f7c>] (generic_handle_irq+0x44/0x54)
Mar  5 21:04:38 Hazel kernel: [ 2295.957772]  r7:00000001 r6:00000000 r5:00000000 r4:80e91d90
Mar  5 21:04:38 Hazel kernel: [ 2295.957791] [<80184f38>] (generic_handle_irq) from [<801856dc>] (__handle_domain_irq+0x6c/0xc4)
Mar  5 21:04:38 Hazel kernel: [ 2295.957811] [<80185670>] (__handle_domain_irq) from [<801012c8>] (bcm2836_arm_irqchip_handle_irq+0x60/0x64)
Mar  5 21:04:38 Hazel kernel: [ 2295.957828]  r9:83012000 r8:83013f58 r7:83013e44 r6:ffffffff r5:80000013 r4:80a0404c
Mar  5 21:04:38 Hazel kernel: [ 2295.957845] [<80101268>] (bcm2836_arm_irqchip_handle_irq) from [<80100abc>] (__irq_svc+0x5c/0x7c)
Mar  5 21:04:38 Hazel kernel: [ 2295.957855] Exception stack(0x83013e10 to 0x83013e58)
Mar  5 21:04:38 Hazel kernel: [ 2295.957868] 3e00:                                     836f65e4 00000000 00000fb8 00000fb7
Mar  5 21:04:38 Hazel kernel: [ 2295.957885] 3e20: 00000014 836f6040 50baf3a0 83013f58 83013f58 836f4040 7f3172a8 83013e6c
Mar  5 21:04:38 Hazel kernel: [ 2295.957899] 3e40: 83013e70 83013e60 7f31731c 80a0404c 80000013 ffffffff
Mar  5 21:04:38 Hazel kernel: [ 2295.957929] [<80a0400c>] (_raw_spin_lock) from [<7f31731c>] (cr14_write+0x74/0x67c [cr14])
Mar  5 21:04:38 Hazel kernel: [ 2295.957956] [<7f3172a8>] (cr14_write [cr14]) from [<80335150>] (vfs_write+0xd4/0x464)
[... cut ...]
Mar  5 21:04:38 Hazel kernel: [ 2295.958092] [<803356bc>] (sys_write) from [<80100040>] (ret_fast_syscall+0x0/0x28)
Mar  5 21:04:38 Hazel kernel: [ 2295.958102] Exception stack(0x83013fa8 to 0x83013ff0)
Mar  5 21:04:38 Hazel kernel: [ 2295.958116] 3fa0:                   0000000b 76f07e40 0000000b 50baf3a0 00000014 00000000
Mar  5 21:04:38 Hazel kernel: [ 2295.958132] 3fc0: 0000000b 76f07e40 00000014 00000004 0000000b 50baf3a0 00000000 01f68908
Mar  5 21:04:38 Hazel kernel: [ 2295.958144] 3fe0: 00000000 7e816d28 00000000 76eb4ec0
```

f-laurens, 2022-03-06:
Problem occurs on Pi Zero 2 both with "`zero2`" and `"zero`" Pynab images.
This confirms the problem is with the _armv7l_ version of the `cr14` driver / kernel.

pguyot, 2022-03-07:
Fixed. We just need to rebuild images to pick up latest version of CR14 driver.

closed 2022-03-07 by pguyot

f-laurens, 2022-03-07:
Thanks. I validated the fix on my current image by updating/rebuilding driver after switching to `cr14` _master_ branch.
All appears to be fine now, both for configuring and reading tags.

Can you please please push the fix to `cr14` _release_ branch, so that can I rebuild my images?