nabaztag2018/pynab #316: ASR flaky on Pi Zero 2

f-laurens, 2022-03-05:
ASR seems flaky on Pi Zero 2:
- most of the time, recognition fails,
- and eventually `nabd` gets stuck and has to be restarted.

pguyot, 2022-03-05:
ASR seemed to work here, didn't test RFID.

Which image did you try with the Zero 2:
- the rapios zero (supposedly compatible with both zero and zero 2)
- the raspios zero 2 (optimized for armv7l)
- a dietpi?

f-laurens, 2022-03-05:
So far, I tested with a `zero2_dietpi` image. I will try with a `zero2_raspios` image.
I didn't check at all the behaviour of zero images on Pi Zero 2.

pguyot, 2022-03-05:
I don't think releasing the zero2_raspios image really makes sense as it might add confusion and the zero_raspios image should work with Zero 2.

f-laurens, 2022-03-05:
> I don't think releasing the zero2_raspios image really makes sense as it might add confusion and the zero_raspios image should work with Zero 2.

Quite true, especially since we have not claimed official Pi Zero 2 support at this point.
I would then vote for not building the zero2 images: this would make builds simpler by reducing the number of images.

f-laurens, 2022-03-05:
> ... the zero_raspios image should work with Zero 2.

Are you sure? I do not quite figure out how the `zero` Pynab install ( with `armv6l` Python wheels) will work on an `armv7l` system... 
If you are right, I would then vote for making our builds simpler by not building the `zero2` images.

pguyot, 2022-03-05:
pip will try and fetch armv7l wheels if the (real or emulated) arch is armv7l, but these wheels install compiled files at the same location as armv6l wheels.

However, on the first upgrade, pip will replace code of these wheels, following requirements.txt logic. This means that after an upgrade, the SD will no longer work on a Zero. Likewise, our upgrade script only recompiles drivers for the native kernel. Not sure about the best behavior here.

f-laurens, 2022-03-05:
> pip will try and fetch armv7l wheels if the (real or emulated) arch is armv7l, but these wheels install compiled files at the same location as armv6l wheels.

Indeed, but pip is only invoked thru Pynab's `install.sh` (right?),  so a zero (`armv6l`) image will not work (Pynab wise) 'out of the box' on a Zero 2... (unless we force an upgrade on first boot?)

pguyot, 2022-03-05:
A Pynab image created with setup on armv6l will (a) install drivers compiled for armv6l for all three kernels (kernel.img, kernel7.img and kernel7l.img) and (b) install armv6l wheels.

It will boot on Zero 2 which will happily execute this code optimized for armv6l.

On update on Zero 2, pip may replace armv6l code with armv7l code and install.sh may recompile drivers for the current kernel only (kernel7l).

When installed again in a Zero, the native code installed by pip will not work and the drivers will still be the old code compiled for armv6l, but upgrade script will erroneously consider it up to date.

So our image should work in a Zero 2 but putting the SD later in a Zero might not work.

Same with Raspberry Pi OS if you install some native code with pip anyway.

f-laurens, 2022-03-05:
> A Pynab image created with setup on armv6l will (a) install drivers compiled for armv6l for all three kernels (kernel.img, kernel7.img and kernel7l.img) and (b) install armv6l wheels.

Indeed. I am fully aware of this (https://github.com/nabaztag2018/pynab/pull/305/commits/4a366941d0ccc963ac5493a7650e9270a546942b and https://github.com/nabaztag2018/pynab/pull/305/commits/6aac9b33562adceabb19f7df43a054107153e92f ;-)

> It will boot on Zero 2 which will happily execute this code optimized for armv6l.

Fine. I stand corrected then.
In that case, I still fully second your proposal to build only `zero` images. It will keep things simpler....

> So our image should work in a Zero 2 but putting the SD later in a Zero might not work.

... given that we should not care about this borderline use case.

multiplie-fr, 2022-03-05:
ASR is working perfectly here. It's faster too.

> ASR seemed to work here, didn't test RFID.
> 
> Which image did you try with the Zero 2:
> 
>     * the rapios zero (supposedly compatible with both zero and zero 2)
> 
>     * the raspios zero 2 (optimized for armv7l)
> 
>     * a dietpi?

f-laurens, 2022-03-05:
> ASR is working perfectly here. It's faster too.

**_here_** being what? (what image on what hardware?)

f-laurens, 2022-03-05:
I switched from  the `zero2_dietpi` to the `zero2_raspios` image.
Behaviour is roughly the same:
- most of the time, recognition fails (seemed better initially than on DietPi, but not really)
- and in any case eventually `nabd` gets stuck and has to be restarted.

f-laurens, 2022-03-06:
I switched from a `zero2` image to a `zero` image.
And... behaviour is still the same:

most of the time, recognition is flaky (seemed better initially than with `zero2` image - experimenter's wishful thinking? - but in fact not);
and quickly, after a handful of attempts, nabd gets stuck and has to be restarted.

f-laurens, 2022-03-06:
Also, despite all my attempts, it couldn't get ASR to fail with a `zero` image running on Pi Zero hardware:
it works flawlessly (or at least as well as it did with release 0.9.1).

f-laurens, 2022-03-06:
Concerning this:
> However, on the first upgrade, pip will replace code of these wheels, following requirements.txt logic. This means that after an upgrade, the SD will no longer work on a Zero. Likewise, our upgrade script only recompiles drivers for the native kernel. Not sure about the best behavior here.

I have, for the sake of it, played the scenario:

1) Use SD card with `zero` image on Pi Zero 2  (_bitch about ASR / RFID not working..._).
2) Run `upgrade.sh` script to replace armv6l wheels by armv7l ones  (_bitch again..._).
3) Replace PI Zero 2 by Pi Zero, using same SD card.
4) Run `upgrade.sh` script again to replace armv7l wheels by armv6l ones  (_enjoy working ASR / RFID..._).

To make this a little more foolproof in case there was an upgrade to drivers between 1) and 3), we might have the `install.sh` script build the drivers for all kernels present on the image, as is done in `setup`.
Then running `upgrade.sh` would be the only requirement in all cases when swapping a `zero` SD card between Pi Zero and Pi Zero 2.

pguyot, 2022-03-08:
No significant difference was found for ASR here between a Pi Zero with 0.9.1 and a Pi Zero 2 with latest image + two fixes on cr14 and tagtagtag-ears drivers.

f-laurens, 2022-03-08:
Thanks for the ears driver fix.
At first it seemed to alleviate a bit the ASR problem on Pi Zero 2 (I'm currently using a `zero` image) here: apparently more successful ASR attempts (taking indeed #36 into account) before failure.
But in the end the problem remains: after a few times, `nabd` gets stuck during the recognition phase (node LED remains lit), `htop` shows it consuming all CPU resources, and it has to be restarted.

f-laurens, 2022-03-08:
Looks like the issue is linked to the the complexity of the spoken phrase to recognize...
I.e. (in French mode): a single word phrase like "**_Météo_**" or "**_Carotte_**" will go through nicely, but a phrase like "**_Qualité de l'air_**" or "**_Que sais tu faire?_**" will fry `nabd` immediately.

f-laurens, 2022-06-12:
ASR behaviour on Pi Zero 2 will need full re-testing with the Pynab 1.0.0 images, when these are available.