nabaztag2018/pynab #323: Drivers build broken with kernel 5.15.30

f-laurens, 2022-04-03:
Upgrading Raspberry Pi kernel to version `5.15.30 `breaks build of drivers:
see https://github.com/pguyot/tagtagtag-ears/issues/7, https://github.com/pguyot/cr14/issues/6 and https://github.com/pguyot/wm8960/issues/18.

Commit-La-Grenouille, 2022-04-05:
The audio driver has been fixed and the changes are up for review: https://github.com/pguyot/wm8960/pull/19

Commit-La-Grenouille, 2022-04-05:
The RFID driver has been fixed and the changes are up for review: https://github.com/pguyot/cr14/pull/7

Commit-La-Grenouille, 2022-04-05:
The ears driver has been fixed and the change is up for review: https://github.com/pguyot/tagtagtag-ears/pull/8

Commit-La-Grenouille, 2022-04-05:
For the curious people, here is some information on the root cause: when building the drivers on kernel 5.10.x, there was a deprecation warning that always should be replaced by always-y.
Apparently, the kernel 5.15 got completely rid of the always support and left us with the overlay no longer being generated. A simple syntax update to register the overlay with always-y and we are back in business :)

f-laurens, 2022-04-17:
https://github.com/pguyot/tagtagtag-ears/issues/7, https://github.com/pguyot/cr14/issues/6 and https://github.com/pguyot/wm8960/issues/18 now fixed. Closing.

f-laurens, 2022-04-18:
Fix for [pguyot/wm8960#18](https://github.com/pguyot/wm8960/issues/18) needs to be pushed to `tagtagtag-sound` branch before closing this issue.

closed 2022-04-18 by f-laurens