nabaztag2018/pynab #337: Use of commitup to setup Wifi with Wifi ?

nabaztag2018, 2022-05-04:
Hello hello everyone,

Is there someone here that would be interested in trying to install and use the comitup package of @davesteele :
https://github.com/davesteele/comitup

Ping @f-laurens @Commit-La-Grenouille @mehdichaouch :-)

That would be very helpful for a lot of people.

(Alternative solution could also be implemented but this one seems great).

Thanks !

Olivier

f-laurens, 2022-05-04:
See https://github.com/nabaztag2018/pynab/issues/12#issuecomment-1094349482

davesteele, 2022-05-04:
Comitup is available via a [custom image](https://davesteele.github.io/comitup/) based on Raspberry Pi, or [installation instructions](https://github.com/davesteele/comitup/wiki/Installing-Comitup) exist.

Or, send me a Nazbatag and I'll see what I can do :-)

nabaztag2018, 2022-05-09:
Hello Dave,
Paul manage to make it work so we'll be using it with Pynab.
Thanks a lot !!!!
Olivier

closed 2022-05-09 by nabaztag2018

nabaztag2018, 2022-05-09:
> Comitup is available via a [custom image](https://davesteele.github.io/comitup/) based on Raspberry Pi, or [installation instructions](https://github.com/davesteele/comitup/wiki/Installing-Comitup) exist.
> 
> Or, send me a Nabaztag and I'll see what I can do :-)

And, @davesteele, can you contact me on nabaztag@tagtagtag.fr for a TagTagTag board ? :-)

Thanks !

Olivier